bblanchon/ArduinoJson #555: parseObject failed

ewaldc, 2017-07-21:
First of all, many thanks for this wonderful library.  I am using it with great success in a class library to create a mini OpenHAB2.0 server.  I am not a great C++ programmer, but I have learned a lot from just reading your code!
Recently I upgraded to Arduino IDE 1.8.3 with ESP8266 boardmanager 2.3.0 and ArduinoJSON 5.10.1 and noticed that a number of JSON strings that used to parse successfully, now fail.
Since my code is quite complex, I replicated the problem using the ArduinoJSON assistant and the code it produces.  If I take out either of the bold objects (Item or linkedPage) the parse succeeds.  The (dynamic) buffer size is a little over 2K but I have successfully parsed JSON strings over 10K with my ESP8266 12E (nodemcu 1.2). 

I verified all suggestions from the faq and "why parse fails", but no good.  Since I parsed far more complex structures before, I get the feeling it's not ArduinoJSON related, but how can I be sure?

Also tried to change "const char *json" to "char json[]" but nothing changes.

JSON object:
{
  "name": "Simple",
  "label": "My home automation",
  "link": "/rest/sitemaps/Simple",
  "homepage": {
    "id": "Simple",
    "title": "My home automation",
    "link": "/rest/sitemaps/Simple/Simple",
    "leaf": false,
    "widgets": [
      {
        "widgetId": "Simple_1",
        "type": "Frame",
        "label": "Demo",
        "icon": "frame",
        "mappings": [],
        "widgets": [
          {
            "widgetId": "Simple_1_0",
            "type": "Group",
            "label": "No. of Active Heatings (1)",
            "icon": "heating",
            "mappings": [],
            "**item**": {
              "members": [],
              "groupType": "Switch",
              "function": {
                "name": "OR",
                "params": [
                  "ON",
                  "OFF"
                ]
              },
              "link": "/rest/items/Heating",
              "state": "ON",
              "stateDescription": {
                "pattern": "(%d)",
                "readOnly": false,
                "options": []
              },
              "type": "Group",
              "name": "Heating",
              "label": "No. of Active Heatings",
              "category": "heating",
              "tags": [],
              "groupNames": []
            },
            "**linkedPage**": {
              "id": "0100",
              "title": "No. of Active Heatings (1)",
              "icon": "heating",
              "link": "/rest/sitemaps/Simple/0100",
              "leaf": true,
              "widgets": [
                {
                  "widgetId": "0100_0",
                  "type": "Switch",
                  "label": "Corridor",
                  "icon": "heating",
                  "mappings": [],
                  "item": {
                    "link": "/rest/items/Heating_GF_Corridor",
                    "state": "OFF",
                    "type": "Switch",
                    "name": "Heating_GF_Corridor",
                    "label": "Corridor",
                    "category": "heating",
                    "tags": [],
                    "groupNames": [
                      "GF_Corridor",
                      "Heating"
                    ]
                  },
                  "widgets": []
                }
              ]
            },
            "widgets": []
          }
        ]
      }
    ]
  }
}

**IDE**: Arduin0 1.8.3 with ESP boardmanager 2.3.0
**ArduinoJSON**: 5.10.1
**Target platform**: ESP8266 model 12E
**Compiler output**: 
Using library arduino_965753 at version 5.10.1 in folder: C:\Users\ewald\Documents\Arduino\libraries\arduino_965753 
Sketch uses 247541 bytes (23%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34848 bytes (42%) of dynamic memory, leaving 47072 bytes for local variables. Maximum is 81920 bytes.
C:\Users\ewald\Documents\Arduino\hardware\esp8266com\esp8266/tools/esptool/esptool.exe -vv -cd nodemcu -cb 256000 -cp COM8 -ca 0x00000 -cf C:\Users\ewald\AppData\Local\Temp\arduino_build_854428/sketch_jul19b.ino.bin 

I noticed two other issues:
1. When I put this JSON object in an Array by just adding [] before/after, the ArduinoJSON assistant produces compile errors.   The problem is that it uses the variable **root** for both the array as well as the first (and only) object of the array.  Of course, the manual fix is simple.   I noticed it as tried to change the object into an array to see if the problem would disappear with parsing an array.  The lines of coed that produce the error:
JsonArray& **root** = jsonBuffer.parseArray(json);
JsonObject& **root** = root[0];

2. Compiling my class with ArduinoJSON succeeds with version 5.10.1 but fails with 5.11.x
I am using JSON variant and StaticJsonBufferBase to get the benefits from DynamicArrays (e.g. size of JSON string can vary) with the speed of Static.  The compiler basically complains that StaticJsonBufferBase is a deleted class.  
Part of the code that fails:
	char *_sitemapsListBuffer = (char *) malloc (size);
	JsonVariant _siteMapsList;
	_sitemapsListBuffer = (char *) malloc (size);
	StaticJsonBufferBase sitemapsListJSONBuffer = StaticJsonBufferBase(_sitemapsListBuffer, size);
	_siteMapsList = sitemapsListJSONBuffer.parseArray(json);
Since the use of StaticJsonBufferBase was not supported in the first place, this is of course my fault, but I could not find another way to deal with variable json string sizes, yet get the efficiency of static buffers and managing the reuse of the buffer space (malloc/free) within my class on a platform with little memory.  

Thanks in advance,

Ewald

bblanchon, 2017-07-21:
Hi Ewald,

It's quite possible that the MCU is running out of memory.
Can you try to reduce the size/complexity of the input to get to a point where the program works?

Have a look at [How to reduce memory usage?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reduce-memory-usage/)
In particular, try to leverage the stream feature, this can save a lot of memory.

I added #556 for the bug in the Assistant, thank you very much for reporting this.

What you're trying to do with `StaticJsonBufferBase` is exactly what `DynamicJsonBuffer` does?
My advice is to use `DynamicJsonBuffer` instead.

Regards,
Benoit

ewaldc, 2017-07-22:
Benoit.
thanks for the rapid response and great suggestion on using DynamicJsonBuffer jsonBuffer(size) as substitute for StaticJsonBufferBase(size).  How did I manage to miss that perfect solution ?  Maybe the big code size and memory overhead of FAQ [https://bblanchon.github.io/ArduinoJson/faq/what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer/](url) made me ignore DynamicJsonBuffer.   As I understand it now, using a calculated size as argument to DynamicJsonBuffer makes it similar in size and performance as StaticJsonBufferBase, while retaining the benefits of dynamic scaling just in case the calculated size is a bit off.   
This will greatly simplify my code.

With respect to the "problem json string", I read the "how to reduce memory usage" before I started the project and applied it all throughout the code: no single string is constant. The part that puzzles me is that much larger JSON strings parse perfectly, even using the Assistant code which uses much more memory as it uses a const char. For example:
[json.txt](https://github.com/bblanchon/ArduinoJson/files/1167187/json.txt)
parses perfectly in my code @4052 bytes as well as with the Assistant's code @7604 bytes buffer size (due to const char*) .  That is >2x bigger than the string that fails, or ~7x when using a const char.

With respect to narrowing down the problem:  the following JSON string succeeds @1968 bytes (const) and 1116 bytes (char []):
[jsonOK.txt](https://github.com/bblanchon/ArduinoJson/files/1167241/jsonOK.txt)
The following string fails, the difference being just one single object of 3 characters "XYZ":
[jsonFAIL.txt](https://github.com/bblanchon/ArduinoJson/files/1167242/jsonFAIL.txt)
Yet the following parses also OK while requiring more memory @2004 bytes (const) than the failing string.
[jsonAlsoOK.txt](https://github.com/bblanchon/ArduinoJson/files/1167271/jsonAlsoOK.txt)
All using the code from the Assistant.

Changing "const char" to "char []" makes no difference, nor does changing "DynamicJsonBuffer jsonBuffer(bufferSize);" to "DynamicJsonBuffer jsonBuffer(1500)" (which we know if large enough for a non-const string giving the difference of just one object of 3 characters.

That is why, at least initially, I discarded "out of memory" as a possible cause.
Tonight I'll add some #ifdef code to ArduinoJSON to better understand where it fails.  This way I will be able to get better diagnosis of parsing failures (e.g. at what character and with what error) without growing the code size or reducing performance in normal use.  Right now I can't see a different approach to explain what is going wrong. Happy to contribute this back for your consideration.  
Ewald

ewaldc, 2017-07-24:
It seems not to be a memory error, but a parser error: ERROR_INVALID_VALUE.
    if (!parseAnythingTo(&value)) goto ERROR_INVALID_VALUE;
Need to insert more debug code to find out where it exactly fails.
Ewald

ewaldc, 2017-07-24:
**Problem solved.**  The JSON string exceeded the nestingLimit of 10.
Changing the Assistant code to 
   JsonObject& root = jsonBuffer.parseObject(json, 20);
is all it takes...
Maybe this is something to consider as an enhancement for the Assistant as I guess many users will be as lazy and spoiled as I am by the wonderful Assistant tool and totally rely on it for validation and code generation.   It's not obvious to analyze the recursive depth of a complex JSON string by hand to determine if the nestingLimit of 10 has been exceeded.
I'll post the debug code here for your consideration.

bblanchon, 2017-07-24:
Hi @ewaldc,

That's an excellent suggestion, I'll add it to the Assistant.

Sorry I didn't think about the nesting limit.
This is literally the first time it causes an issue.

Regards,
Benoit

ewaldc, 2017-07-24:
Benoit,
Please find attached the instrumented JsonParser code that allowed me to find what went wrong.
Definitely a "quick fix" with just one single file modified but at least I made sure no line of code is added/changed compared to the original version in the default run-time scenario. 
Enter "#define ArduinoJsonDebug" ahead of "#include <ArduinoJson.h>" in the project code to activate the additional error checking and verbosity.
[JsonParserImpl.zip](https://github.com/bblanchon/ArduinoJson/files/1169214/JsonParserImpl.zip).
It's based on the latest released version of ArdunoJson (5.11.1)
Ewald

closed 2017-08-27 by bblanchon