bblanchon/ArduinoJson #557: ArduinoJson Assistant for constructing XML

probonopd, 2017-07-23:
https://bblanchon.github.io/ArduinoJson/assistant/ is great for parsing XML. It would be extremely helpful if it could also be used for constructing, rather than parsing, XML.

For example, I want to construct the follwoing XML using ArduinoJson. It is very cumbersome, especially for someone new to the library, to write all the code by hand.

```
{
  "name": "whistle_v2.gcode",
  "size": 1468987,
  "date": 1378847754,
  "origin": "local",
  "refs": {
    "resource": "http://example.com/api/files/local/whistle_v2.gcode",
    "download": "http://example.com/downloads/files/local/whistle_v2.gcode"
  },
  "gcodeAnalysis": {
    "estimatedPrintTime": 1188,
    "filament": {
      "length": 810,
      "volume": 5.36
    }
  },
  "print": {
    "failure": 4,
    "success": 23,
    "last": {
      "date": 1387144346,
      "success": true
    }
  }
}
```

bblanchon, 2017-07-24:
Hi @probonopd,

Thanks you for this suggestion.
That was part of the initial plan, but I wasn't sure it would be needed as no one asked for it.

Regards,
Benoit

closed 2017-10-11 by bblanchon

Sovatna, 2017-10-22:
Dear bblanchon,
Please kindly help to determine the size of ArduinoJson with the following data: 
{
	"cod":"200",
	"message":0.0084,
	"cnt":37,
	"list":[
			{
				"dt":1508662800,
				"main":{
					"temp":303.58,
					"temp_min":301.161,
					"temp_max":303.58,
					"pressure":1019.79,
					"sea_level":1021.26,
					"grnd_level":1019.79,
					"humidity":94,
					"temp_kf":2.42
					},
				"weather":[
						{
							"id":500,
							"main":"Rain",
							"description":"light rain",
							"icon":"10d"
						}
					],
				"clouds":{
					"all":32
					},
				"wind":{
					"speed":1.76,
					"deg":135.512
					},
				"rain":{
					"3h":0.52
					},
				"sys":{
					"pod":"d"
					},
				"dt_txt":"2017-10-22 09:00:00"
			},
			{
				"dt":1508673600,
				"main":{
					"temp":300.63,
					"temp_min":299.015,
					"temp_max":300.63,
					"pressure":1021.07,
					"sea_level":1022.61,
					"grnd_level":1021.07,
					"humidity":96,
					"temp_kf":1.61
					},
				"weather":[
						{
							"id":500,
							"main":"Rain",
							"description":"light rain",
							"icon":"10n"
						}
					],
				"clouds":{
					"all":64
					},
				"wind":{
					"speed":1.4,
					"deg":190.509
					},
				"rain":{
[... 340 lines cut ...]
				"main":{
					"temp":296.99,
					"temp_min":296.99,
					"temp_max":296.99,
					"pressure":1023.04,
					"sea_level":1024.52,
					"grnd_level":1023.04,
					"humidity":100,
					"temp_kf":0
					},
				"weather":[
						{
							"id":500,
							"main":"Rain",
							"description":"light rain",
							"icon":"10n"
						}
					],
				"clouds":{"all":92},
				"wind":{"speed":2.5,"deg":45.0008}
			}
	}

Best regards, 
Sovatna

probonopd, 2017-10-22:
@Sovatna running your JSON through https://jsonformatter.curiousconcept.com/ shows that it has errors. After fixing those, pasting your JSON into https://bblanchon.github.io/ArduinoJson/assistant/ gives

```
13*JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(13) + 2*JSON_OBJECT_SIZE(0) + 35*JSON_OBJECT_SIZE(1) + 13*JSON_OBJECT_SIZE(2) + 14*JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(5) + 25*JSON_OBJECT_SIZE(8)
```

Platform | Size
-- | --
AVR 8-bit | 7617
ESP8266 | 10069
Visual Studio x86 | 15949
Visual Studio x64 | 17197

probonopd, 2017-10-22:
Thanks @bblanchon for this really, really useful tool! :+1:

Sovatna, 2017-10-22:
Thank you very much, but it seems to not work with my cords. Anyway, you can help tell me how to get this size from ArduinoJson assistance please. 

Sovatna

probonopd, 2017-10-22:
@Sovatna first your JSON through https://jsonformatter.curiousconcept.com/ to see what errors it has and correct these. Then paste your corrected JSON into https://bblanchon.github.io/ArduinoJson/assistant/ to get the values above.

Sovatna, 2017-10-22:
Thank you so much for your clarification. it works me out right now. 
Best regards, 

Sovatna

Sovatna, 2017-10-22:
@probonopd  may you please help tell me if the above data can be processed with ESP8266MOD? I have tried it many times, but it appears nothing data, except only disconnected. 

Best regards, 
Sovatna

Sovatna, 2017-10-22:
@probonopd @bblanchon Please kindly check this code below. I work it with ESP8266 model: ESP8266MOD. It does not work actually while I do not know why. Please help. 

#include <ArduinoJson.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);// Starts the serial port with 9600 baud
}

void loop() {
  // put your main code here, to run repeatedly:
  const size_t bufferSize = 30*JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(30) + 16*JSON_OBJECT_SIZE(0) + 74*JSON_OBJECT_SIZE(1) + 30*JSON_OBJECT_SIZE(2) + 31*JSON_OBJECT_SIZE(4) + 60*JSON_OBJECT_SIZE(8) + 9750;
  DynamicJsonBuffer jsonBuffer(bufferSize);
  
  const char* json = "{\"cod\":\"200\",\"message\":0.0054,\"cnt\":40,\"list\":[{\"dt\":1508684400,\"main\":{\"temp\":299.86,\"temp_min\":297.207,\"temp_max\":299.86,\"pressure\":1022.79,\"sea_level\":1024.26,\"grnd_level\":1022.79,\"humidity\":100,\"temp_kf\":2.66},\"weather\":[{\"id\":803,\"main\":\"Clouds\",\"description\":\"broken clouds\",\"icon\":\"04n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":2.77,\"deg\":164.003},\"rain\":{},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-22 15:00:00\"},{\"dt\":1508695200,\"main\":{\"temp\":298.93,\"temp_min\":296.942,\"temp_max\":298.93,\"pressure\":1021.75,\"sea_level\":1023.23,\"grnd_level\":1021.75,\"humidity\":100,\"temp_kf\":1.99},\"weather\":[{\"id\":803,\"main\":\"Clouds\",\"description\":\"broken clouds\",\"icon\":\"04n\"}],\"clouds\":{\"all\":56},\"wind\":{\"speed\":1.86,\"deg\":143.502},\"rain\":{},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-22 18:00:00\"},{\"dt\":1508706000,\"main\":{\"temp\":298.27,\"temp_min\":296.939,\"temp_max\":298.27,\"pressure\":1021.27,\"sea_level\":1022.75,\"grnd_level\":1021.27,\"humidity\":100,\"temp_kf\":1.33},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":92},\"wind\":{\"speed\":2.31,\"deg\":124.5},\"rain\":{\"3h\":0.0049999999999999},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-22 21:00:00\"},{\"dt\":1508716800,\"main\":{\"temp\":298.18,\"temp_min\":297.52,\"temp_max\":298.18,\"pressure\":1022.67,\"sea_level\":1024.17,\"grnd_level\":1022.67,\"humidity\":100,\"temp_kf\":0.66},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10d\"}],\"clouds\":{\"all\":92},\"wind\":{\"speed\":2.31,\"deg\":122.503},\"rain\":{\"3h\":0.06},\"sys\":{\"pod\":\"d\"},\"dt_txt\":\"2017-10-23 00:00:00\"},{\"dt\":1508727600,\"main\":{\"temp\":299.71,\"temp_min\":299.71,\"temp_max\":299.71,\"pressure\":1023.9,\"sea_level\":1025.41,\"grnd_level\":1023.9,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10d\"}],\"clouds\":{\"all\":92},\"wind\":{\"speed\":2.47,\"deg\":116.501},\"rain\":{\"3h\":0.18},\"sys\":{\"pod\":\"d\"},\"dt_txt\":\"2017-10-23 03:00:00\"},{\"dt\":1508738400,\"main\":{\"temp\":300.74,\"temp_min\":300.74,\"temp_max\":300.74,\"pressure\":1021.94,\"sea_level\":1023.36,\"grnd_level\":1021.94,\"humidity\":97,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10d\"}],\"clouds\":{\"all\":76},\"wind\":{\"speed\":2.01,\"deg\":103.505},\"rain\":{\"3h\":1.575},\"sys\":{\"pod\":\"d\"},\"dt_txt\":\"2017-10-23 06:00:00\"},{\"dt\":1508749200,\"main\":{\"temp\":299.967,\"temp_min\":299.967,\"temp_max\":299.967,\"pressure\":1021.42,\"sea_level\":1022.92,\"grnd_level\":1021.42,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":501,\"main\":\"Rain\",\"description\":\"moderate rain\",\"icon\":\"10d\"}],\"clouds\":{\"all\":56},\"wind\":{\"speed\":2.43,\"deg\":132.003},\"rain\":{\"3h\":5.45},\"sys\":{\"pod\":\"d\"},\"dt_txt\":\"2017-10-23 09:00:00\"},{\"dt\":1508760000,\"main\":{\"temp\":297.479,\"temp_min\":297.479,\"temp_max\":297.479,\"pressure\":1022.97,\"sea_level\":1024.46,\"grnd_level\":1022.97,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":501,\"main\":\"Rain\",\"description\":\"moderate rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":80},\"wind\":{\"speed\":2.06,\"deg\":176.005},\"rain\":{\"3h\":7.85},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-23 12:00:00\"},{\"dt\":1508770800,\"main\":{\"temp\":297.008,\"temp_min\":297.008,\"temp_max\":297.008,\"pressure\":1024.45,\"sea_level\":1025.95,\"grnd_level\":1024.45,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":68},\"wind\":{\"speed\":1.95,\"deg\":179},\"rain\":{\"3h\":0.025000000000002},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-23 15:00:00\"},{\"dt\":1508781600,\"main\":{\"temp\":297.16,\"temp_min\":297.16,\"temp_max\":297.16,\"pressure\":1024.03,\"sea_level\":1025.46,\"grnd_level\":1024.03,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":500,\"main\":\"Rain\",\"description\":\"light rain\",\"icon\":\"10n\"}],\"clouds\":{\"all\":76},\"wind\":{\"speed\":1.42,\"deg\":154.001},\"rain\":{\"3h\":0.064999999999998},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-23 18:00:00\"},{\"dt\":1508792400,\"main\":{\"temp\":296.705,\"temp_min\":296.705,\"temp_max\":296.705,\"pressure\":1023.41,\"sea_level\":1024.9,\"grnd_level\":1023.41,\"humidity\":100,\"temp_kf\":0},\"weather\":[{\"id\":803,\"main\":\"Clouds\",\"description\":\"broken clouds\",\"icon\":\"04n\"}],\"clouds\":{\"all\":76},\"wind\":{\"speed\":1.12,\"deg\":214},\"rain\":{},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-23 21:00:00\"},{\"dt\":1508803200,\"main\":{\"temp\":297.349,\"temp_min\":297.349,\"t [... cut ...] 1023.17,\"sea_level\":1024.65,\"grnd_level\":1023.17,\"humidity\":99,\"temp_kf\":0},\"weather\":[{\"id\":802,\"main\":\"Clouds\",\"description\":\"scattered clouds\",\"icon\":\"03n\"}],\"clouds\":{\"all\":32},\"wind\":{\"speed\":2.55,\"deg\":8.00641},\"rain\":{},\"sys\":{\"pod\":\"n\"},\"dt_txt\":\"2017-10-26 21:00:00\"}]}";
  
  JsonObject& root = jsonBuffer.parseObject(json);
  
  const char* cod = root["cod"]; // "200"
  //root.printTo(Serial);
  Serial.println(cod);
  delay(1000);
}

Sovatna

Sovatna, 2017-11-04:
Dear @probonopd, 

Normally, the result displays as below: 

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

I do not know what is really problem. please kindly help!

Sovatna

probonopd, 2017-11-04:
This means that you sketch crashes and the ESP reboots. For whatever reason.

Sovatna, 2017-11-05:
Dear @probonopd, 

Thank you very much for your sharing. May you have any idea how to fix this problem. It is really bad for me to always see the occurred problem.

probonopd, 2017-11-05:
Maybe @bblanchon can comment on this, or on how to best debug it?

bblanchon, 2017-11-05:
@Sovatna Usually a reset is caused by a stack overflow or a null pointer.
Can you open a new issue and provide an [MVCE](https://stackoverflow.com/help/mcve)?

See:

* [Why does my device crash or reboot?](https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot/)
* [Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)

Sovatna, 2017-11-05:
Dear @probonopd @bblanchon, 

Thank you very much for your kind help. I do not know how to debug with this code below: Please help: 
#include <ESP8266WiFi.h> //Include the library ESP8266
#define ARDUINOJSON_ENABLE_PROGMEM 0
#include <ArduinoJson.h>    //https://github.com/bblanchon/ArduinoJson

const char* ssid     = "ssid";
const char* password = "password";
String CityID = "lat=11.55&lon=104.87";//"253394"; //Sparta, Greece
String APIKEY = "APIKEY"; //"yourAPIkey";

WiFiClient client;
char* servername ="api.openweathermap.org";  // remote server we will connect to
String result;
int weatherID = 0;
unsigned long  iteration = 1 * 60 * 1000;
unsigned long currentTime = millis();
unsigned long lastTime = 0; 
String weatherDescription ="";
String weatherLocation = "";
float Temperature;
unsigned long minute = 60*1000;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  connectToWifi();
}

void loop() {
  // put your main code here, to run repeatedly:
   if(iteration == (1*60*1000))//We check for updated weather forecast once every hour
   {
     getWeatherData();
     iteration = 0;   
   }
   iteration++;
}

void connectToWifi()
{  
  delay(20);
  WiFi.begin(ssid, password);
  int wifi_ctr = 0;
  while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
}

void getWeatherData() //client function to send/receive GET request data.
{
  
  const int httpPort = 80;
  if (!client.connect(servername, httpPort)) {
        return;
    }
      // We now create a URI for the request
    String url = "/data/2.5/forecast?"+CityID+"&APPID="+APIKEY;

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + servername + "\r\n" +
                 "Connection: close\r\n\r\n");
    unsigned long timeout = millis();
    while (client.available() == 0) {
        if (millis() - timeout > 5000) {
            client.stop();
            return;
        }
    }

    // Read all the lines of the reply from server
    while(client.available()) {
        result = client.readStringUntil('\r');
        
    }

    result.replace('[', ' ');
    result.replace(']', ' ');
    char jsonArray [result.length()+1];
    result.toCharArray(jsonArray,sizeof(jsonArray));
    jsonArray[result.length() + 1] = '\0';
    const size_t bufferSize = 31*JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(31) + 26*JSON_OBJECT_SIZE(0) + 67*JSON_OBJECT_SIZE(1) + 32*JSON_OBJECT_SIZE(2) + 32*JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(5) + 62*JSON_OBJECT_SIZE(8) + 10090;
    DynamicJsonBuffer json_buf(bufferSize);
    //StaticJsonBuffer<bufferSize> json_buf;
    JsonObject &root = json_buf.parseObject(jsonArray);
    if (!root.success())
    {
      Serial.println("parseObject() failed");
    }
 
    String location = root["city"]["name"];
    String temperature = root["list"]["main"]["temp"];
    String weather = root["list"]["weather"]["main"];
    String description = root["list"]["weather"]["description"];
    String idString = root["list"]["weather"]["id"];
    String timeS = root["list"]["dt_txt"];
    
    weatherID = idString.toInt();
    Serial.print("\nWeatherID: ");
    Serial.print(weatherID);
}

probonopd, 2017-11-05:
@Sovatna please do as describeed in https://github.com/bblanchon/ArduinoJson/issues/557#issuecomment-341974789

Sovatna, 2017-11-05:
Dear @probonopd , 

I am trying with the comment mentioned above. Something wrong is still happening. I keep trying it. Thank you so much.

commit b9efa089 (2017-10-11): Added "Serializing program" in Assistant (closes #557)