bblanchon/ArduinoJson #558: Cannot parse `long long` or `unsigned long long`

aj-ptw, 2017-07-24:
I am in the process of writing unit tests for a library using ArduinoJSON, and while the AduinoJSON can set `unsigned long long` or `long long`.

Expected to be able to parse `unsigned long long` and `long long` when:
```
#define ARDUINOJSON_USE_LONG_LONG 1
```

Example:
```Arduino
void testGetJSONFromSampleCyton() {
  test.describe("getJSONFromSampleCyton");

  wifi.sampleReset(wifi.sampleBuffer);
  uint8_t numChannels = NUM_CHANNELS_CYTON;
  uint8_t numSamples = 1;
  uint8_t expected_sampleNumber = 29;
  unsigned long long expected_timestamp = 1500916934017000;
  long long expected_channelData[MAX_CHANNELS];
  for (int i = 0; i < MAX_CHANNELS; i++) {
    expected_channelData[i] = i * NANO_VOLTS_IN_VOLTS;
  }

  wifi.sampleBuffer->sampleNumber = expected_sampleNumber;
  wifi.sampleBuffer->timestamp = expected_timestamp;
  memcpy(wifi.sampleBuffer->channelData, expected_channelData, numChannels);

  String actual_serializedOutput = wifi.getJSONFromSample(wifi.sampleBuffer, numChannels, numSamples);

  // const char* json = "{\"chunk\":[{\"timestamp\":1500916934017000,\"sampleNumber\":255,\"data\":[8388607000000000,8388607000000000,8388607000000000,8388607000000000,8388607000000000,8388607000000000,8388607000000000,8388607000000000]}]}";
  const size_t bufferSize = JSON_ARRAY_SIZE(1) + JSON_ARRAY_SIZE(8) + JSON_OBJECT_SIZE(1) + JSON_OBJECT_SIZE(3) + 220;
  DynamicJsonBuffer jsonBuffer(bufferSize);

  JsonObject& root = jsonBuffer.parseObject(actual_serializedOutput);

  JsonObject& chunk0 = root["chunk"][0];
  unsigned long long actual_timestamp = chunk0["timestamp"]; // 1500916934017000
  test.assertEqual((unsigned long long)actual_timestamp, expected_timestamp, "should be able to set timestamp", __LINE__);

  uint8_t actual_sampleNumber = chunk0["sampleNumber"]; // 255
  test.assertEqual(actual_sampleNumber, expected_sampleNumber, "should be able to set sampleNumber", __LINE__);

  JsonArray& chunk0_data = chunk0["data"];
  for (int i = 0; i < numChannels; i++) {
    long long chunk0_tempData = chunk0_data[i];
    test.assertEqual(expected_channelData[i], chunk0_data[i]);
  }
}
```

Will give a heft error:
```bash
$ make -f makeEspWifiTest.mk
OpenBCI_Wifi_Test.ino
In file included from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:12:0,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,

                 from /Users/aj/Documents/Arduino/libraries/OpenBCI_Wifi/OpenBCI_Wifi.h:29,
                 from /Users/aj/Documents/Arduino/libraries/OpenBCI_Wifi/tests-ptw-assert/OpenBCI_Wifi_Test/OpenBCI_Wifi_Test.ino:12:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp: In instantiation of 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::Js
onObject::get_impl(TStringRef) const [with TStringRef = const char*; TValue = long long unsigned int; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = long long unsi
gned int]':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:181:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::Json
Object::get(const TString*) const [with TValue = long long unsigned int; TString = char; typename ArduinoJson::Internals::JsonVariantAs<T>::type = long long unsigned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJs
on::JsonObjectSubscript<TKey>::as() const [with TValue = long long unsigned int; TStringRef = const char*; typename ArduinoJson::Internals::JsonVariantAs<T>::type = long long unsi
gned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:63:35:   required from 'const typename ArduinoJson::Internals::JsonVariant
As<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = long long unsigned int; TImpl = ArduinoJson::JsonObjectSubscript<const char*>; typename ArduinoJson::Internal
s::JsonVariantAs<T>::type = long long unsigned int]'
[... 380 lines cut ...]
   as() const {
   ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:240:3: note:   template argument deduction/substitution failed:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp: In substitution of 'template<class T> typename ArduinoJson::TypeTraits::Enabl
eIf<ArduinoJson::TypeTraits::IsSame<T, ArduinoJson::JsonVariant>::value, T>::type ArduinoJson::JsonVariant::as() const [with T = long long unsigned int]':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:310:24:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson:
:JsonObject::get_impl(TStringRef) const [with TStringRef = const char*; TValue = long long unsigned int; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = long long u
nsigned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:181:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::Json
Object::get(const TString*) const [with TValue = long long unsigned int; TString = char; typename ArduinoJson::Internals::JsonVariantAs<T>::type = long long unsigned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJs
on::JsonObjectSubscript<TKey>::as() const [with TValue = long long unsigned int; TStringRef = const char*; typename ArduinoJson::Internals::JsonVariantAs<T>::type = long long unsi
gned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:63:35:   required from 'const typename ArduinoJson::Internals::JsonVariant
As<T>::type ArduinoJson::JsonVariantBase<TImpl>::as() const [with T = long long unsigned int; TImpl = ArduinoJson::JsonObjectSubscript<const char*>; typename ArduinoJson::Internal
s::JsonVariantAs<T>::type = long long unsigned int]'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:58:14:   required from 'ArduinoJson::JsonVariantBase<TImpl>::operator T()
const [with T = long long unsigned int; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]'
/Users/aj/Documents/Arduino/libraries/OpenBCI_Wifi/tests-ptw-assert/OpenBCI_Wifi_Test/OpenBCI_Wifi_Test.ino:85:59:   required from here
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:240:3: error: no type named 'type' in 'struct ArduinoJson::TypeTraits::EnableI
f<false, long long unsigned int>'
make: *** [/tmp/mkESP/OpenBCI_Wifi_Test_huzzah/OpenBCI_Wifi_Test_.cpp.o] Error 1
```

Thanks!!

aj-ptw, 2017-07-24:
I am able to continue working using `double` and then converting like:

```
  JsonObject& chunk0 = root["chunk"][0];
  double actual_timestamp = chunk0["timestamp"]; // 1500916934017000
  test.assertEqual((unsigned long long)actual_timestamp, expected_timestamp, "should be able to set timestamp", __LINE__);
```

bblanchon, 2017-07-26:
Hi @aj-ptw,

It works on wandbox:
https://wandbox.org/permlink/DQKqrtPqmlcph0lu

Can you verify that ARDUINOJSON_USE_LONG_LONG is define before the first include of ArduinoJson.h?

aj-ptw, 2017-07-28:
@bblanchon sorry! you are right

closed 2017-07-28 by aj-ptw