bblanchon/ArduinoJson #559: How to parse JsonObject if I don't know what it contains

makeacoffeeprj, 2017-07-24:
Hi everyone,

I'm currently stucked in iterating a JsonObject just knowing its size and not which field are on it.

For example, I'm unable to do something like this:

```
StaticJsonBuffer<200> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();

for (int i=0; i <= root.size(); i++){
    const char* payload_char = root[i];
  }
```
The error is the following:

> /Users/diegobraga/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:299:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'
       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;

Does anyone know how to parse every array element without previously knowing the key fields?

Thanks,
Diego

devyte, 2017-07-24:
> if (Internals::StringTraits::equals(key, it->key)) break;

DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.parse(somestring);

//root now contains a hierarchy of the unknown json. You can iterate
each nested level to figure out what's in it (see docs).

makeacoffeeprj, 2017-07-24:
Hi @devyte, can you point me to the docs describing this iteration?

devyte, 2017-07-24:
https://github.com/bblanchon/ArduinoJson/wiki/Examples

Example 3 gives you an idea, also look at the api reference.

bblanchon, 2017-07-25:
Hi Diego,

Have a look at:

* [API Reference: JsonArray::begin() / JsonArray::end()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/begin_end/)
* [API Reference: JsonObject::begin() / JsonObject::end()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/begin_end/)
* [F.A.Q: How to know the type of a value?](https://bblanchon.github.io/ArduinoJson/faq/how-to-know-the-type-of-a-value/)

Regards,
Benoit

closed 2017-08-08 by bblanchon

opkiler22789, 2017-09-17:
@bblanchon same problem ```void processing(String json){

    DynamicJsonBuffer jsonBuffer;

    JsonObject& array_data = jsonBuffer.parseObject(F(json));
}
```
ERROR:
```
ArduinoJson/JsonObject.hpp:299:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'

       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;

```
                                                                   ^

bblanchon, 2017-11-09:
Please apologize for the long delay @opkiler22789, I didn't see you message as it was made on a closed issue.

This error happens because your program uses an integer to index a `JsonObject`, but it should be a string.
Please see: http://arduinojson.org/faq/error-equals-is-not-a-member-of/