bblanchon/ArduinoJson #560: Print is losing data

willian-gvs, 2017-07-24:
I am not sure if I should ask this here.. but.. here we go!

I am using the library ArduinoJson to get some data from featherM0 and send it to the the computer.

The part of the code that is parsing and generating the json is:

```
char inChar; // Where to store the character read
byte index2; // Index into array; where to store the character
Datas dados;
uint32_t loop_timer = 0;

void loop() { //----------------------------------------------------------------

  StaticJsonBuffer<200> jsonBuffer;
  StaticJsonBuffer<200> jsonBuffer2;
  JsonObject& root = jsonBuffer.createObject();

  char json[200];
  byte index2 = 0;
  while(Serial.available() > 0) // Don't read unless there you know there is data
  {
    if(index2 < 199) // One less than the size of the array
    {
      inChar = Serial.read(); // Read a character
      json[index2] = inChar; // Store it
      index2++; // Increment where to write next
      json[index2] = '\0'; // Null terminate the string
    }
  }

  String jsonSTRING(json);

  JsonObject& root2 = jsonBuffer2.parseObject(json);

  if (root2.success()) {
    if(jsonSTRING.indexOf("DO1") > 0)
      dados.DO1 = root2["DO1"];
    if(jsonSTRING.indexOf("DO2") > 0)
      dados.DO2 = root2["DO2"];
    if(jsonSTRING.indexOf("AO1") > 0)
    {
      dados.AO1 = root2["AO1"];
      if(dados.AO1>=4095)
        dados.AO1 = 4095;
      if(dados.AO1<=0)
        dados.AO1 = 0;
    }

    if(jsonSTRING.indexOf("AO2") > 0)
    {
      dados.AO2 = root2["AO2"];
      if(dados.AO2>=4095)
        dados.AO2 = 4095;
      if(dados.AO2<=0)
        dados.AO2 = 0;
    }
    if(jsonSTRING.indexOf("RstFlow") > 0)
      RstFlow = root2["RstFlow"];
    if(jsonSTRING.indexOf("ActFlow") > 0)
      ActFlow = root2["ActFlow"];

    if(jsonSTRING.indexOf("SclFct") > 0)  // Scale Factor
      SclFct = root2["SclFct"];
      ScaleFactor  = 100;

    if(jsonSTRING.indexOf("RfsRt") > 0)  // RefreshRate
      Refreshrate = root2["RfsRt"];

    if(jsonSTRING.indexOf("RatEng") > 0) // Time/Multiplier - Rate Engen
      {
        if(root2["RatEng"]==1)            // SEGUNDOS
        TimeMultiplier = 1;
        else if(root2["RatEng"]==2)        // MINUTOS
        TimeMultiplier = 60;
        else if(root2["RatEng"]==3)        // HORA
        TimeMultiplier = 3600;
        else if(root2["RatEng"]==4)        // HORA
        TimeMultiplier = 86400;
      }
      digitalWrite(DigitalOutPin1, dados.DO1);
      digitalWrite(DigitalOutPin2, dados.DO2);
      AtivaDAC();
      dac.writeChannel(DAC_CHANNEL_A, dados.AO1);
      dac.writeChannel(DAC_CHANNEL_B, dados.AO2);
}

if ((Timerflag >= 1)&&(ActFlow == 1)) {                                  

  DigitalInValue1 = digitalRead(DigitalInPin1);
  DigitalInValue2 = digitalRead(DigitalInPin2);
  dados.DI1 = DigitalInValue1;
  dados.DI2 = DigitalInValue2;

  AtivaADC();
  dados.AI1voltage = 3*adc.toAnalog(dados.AI1raw);
  dados.AI2voltage = 3*adc.toAnalog(dados.AI2raw);
  dados.AI3voltage = 3*adc.toAnalog(dados.AI3raw);
  dados.AI4voltage = 3*adc.toAnalog(dados.AI4raw);

  root["AI1"] = dados.AI1voltage;
  root["AI2"] = dados.AI2voltage;
  root["AI3"] = dados.AI3voltage;
  root["AI4"] = dados.AI4voltage;
  root["AO1"] = dados.AO1;
  root["AO2"] = dados.AO2;
  root["DI1"] = dados.DI1;
  root["DI2"] = dados.DI2;
  root["DO1"] = dados.DO1;
  root["DO2"] = dados.DO2;

  DeltaPulses = DeltaIntPulses;                         //Save value of the interrupt in a new value
  if (InterruptCounter == 1){                           //Enter loop if and only if InterruptCounter is set 1
    DispTotal = DispTotal + (DeltaPulses*ScaleFactor);  //Calculation of the Units to display
  };
  FlowRate = DeltaPulses * ScaleFactor;                 //FlowRate Calculation where the pulses are multiplied 
  //Show the data results
  CheckRstFlow();

  root["DFR"] = (int)(DispFlowRate*100);
  root["DT"] = DispTotal;
  root["TP"] = TotalPulses;
  root["SclFct"] = ScaleFactor;
  root["RfsRt"] = Refreshrate;
  root["RatEng"] = TimeMultiplier;

  root.printTo(Serial);
  Serial.println();

  Timerflag = 0;                                        //Reset the Timerflag
}
```

`

I'm getting (parsing) the serial data correctly.

However, when I print the last 4 data (TP, SclFct,RfsRt and RatEng) is not being sent.
Json looks like this:
> {"AI1":0,"AI2":0,"AI3":0,"AI4":0,"AO1":0,"AO2":0,"DI1":false,"DI2":false,"DO1":false,"DO2":false,"DFR":0,"DT":0}

What am I doing wrong?

Thanks in advance for your help.

bblanchon, 2017-07-25:
Hi William,

Please read:

* [FAQ: Why some parts are missing?](https://bblanchon.github.io/ArduinoJson/faq/why-some-parts-are-missing/)

Regards,
Benoit

willian-gvs, 2017-07-25:
Hello Benoit,

Sorry for repeat a frequently asked question.

I've changed StaticJsonBuffer<200> to StaticJsonBuffer<400>;
And now I am geting the complete string:

> {"AI1":4890,"AI2":0,"AI3":0,"AI4":4902,"AO1":2000,"AO2":2000,"DI1":0,"DI2":0,"DO1":0,"DO2":1,"DFR":0,"DT":0,"TP":0,"SclFct":1,"RfsRt":1,"RatEng":1}

Thank you very much!

Another problem that is happening is that even though it works perfectly, I'm getting this giant warning:

> In file included from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:17:0,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/JsonSerializer.hpp:10,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:15,
> 
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:12,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonVariant.hpp:16,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:15,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/JsonParser.hpp:10,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonBufferBase.hpp:10,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/DynamicJsonBuffer.hpp:10,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.hpp:10,
> from .piolibdeps\ArduinoJson_ID64\src/ArduinoJson.h:10,
> from src\main.cpp:9:
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/../TypeTraits/FloatTraits.hpp: In instantiation of 'static T ArduinoJson::TypeTraits::FloatTraits<T, 8u>::nan() 
> [with T = double]':
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Polyfills/parseFloat.hpp:34:50:   required from 'T ArduinoJson::Polyfills::parseFloat(const char*) [with T = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonVariantImpl.hpp:92:56:   required from 'T ArduinoJson::JsonVariant::variantAsFloat() const [with T = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonVariant.hpp:168:30:   required from 'const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingP
> oint<T>::value, T>::type ArduinoJson::JsonVariant::as() const [with T = double; typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingPoint<T>::value, T>::type = doub
> le]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonObject.hpp:312:24:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::JsonObject::get_impl(TStringRef
> ) const [with TStringRef = const char*; TValue = double; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonObject.hpp:183:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObject::get(const TString*) cons
> t [with TValue = double; TString = char; typename ArduinoJson::Internals::JsonVariantAs<T>::type = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::a
> s() const [with TValue = double; TStringRef = const char*; typename ArduinoJson::Internals::JsonVariantAs<T>::type = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:63:35:   required from 'const typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonVar
> iantBase<TImpl>::as() const [with T = double; TImpl = ArduinoJson::JsonObjectSubscript<const char*>; typename ArduinoJson::Internals::JsonVariantAs<T>::type = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Data/JsonVariantComparer.hpp:39:46:   required from 'static bool ArduinoJson::Internals::JsonVariantComparer<TComparand, typename ArduinoJson::Ty
> peTraits::EnableIf<((! ArduinoJson::TypeTraits::IsVariant<T>::value) && (! ArduinoJson::TypeTraits::IsString<Print, void>::value))>::type>::equals(const ArduinoJson::JsonVariantBase<TVariant
> &, const TComparand&) [with TVariant = ArduinoJson::JsonObjectSubscript<const char*>; TComparand = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonVariantComparisons.hpp:16:79:   required from 'bool ArduinoJson::operator==(const ArduinoJson::JsonVariantBase<T>&, TComparand) [with TVarian
> t = ArduinoJson::JsonObjectSubscript<const char*>; TComparand = double]'
> src\main.cpp:180:27:   required from here
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/../TypeTraits/FloatTraits.hpp:48:36: warning: dereferencing type-punned pointer will break strict-aliasing rules
>  [-Wstrict-aliasing]
> return *reinterpret_cast<T*>(&x);
> 
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/../TypeTraits/FloatTraits.hpp: In instantiation of 'static T ArduinoJson::TypeTraits::FloatTraits<T, 8u>::inf() 
> [with T = double]':
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Polyfills/parseFloat.hpp:36:43:   required from 'T ArduinoJson::Polyfills::parseFloat(const char*) [with T = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/JsonVariantImpl.hpp:92:56:   required from 'T ArduinoJson::JsonVariant::variantAsFloat() const [with T = double]'
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../JsonVariant.hpp:168:30:   required from 'const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::TypeTraits::IsFloatingP
[... cut ...]
> t = ArduinoJson::JsonObjectSubscript<const char*>; TComparand = double]'
> src\main.cpp:180:27:   required from here
> .piolibdeps\ArduinoJson_ID64\src/ArduinoJson/Deserialization/../Serialization/../TypeTraits/FloatTraits.hpp:53:36: warning: dereferencing type-punned pointer will break strict-aliasing rules
>  [-Wstrict-aliasing]
> return *reinterpret_cast<T*>(&x);
> ^

I tried to read it, but could not get any information to delete it.
Should I open another question with this?

Again, thanks in advance, Benoit!.

Regards,
Willian

bblanchon, 2017-07-25:
Hi William,

Thank you for confirming that the fix works.

About the warning, the important part is:

    warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

It's supposed to be fixed in [5.11.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.11.1)
What version are you using?

Regards,
Benoit

willian-gvs, 2017-07-26:
Hello Benoit,

I am using:
>  "version": "5.11.0"

I will change the library to the new version.
Soon I'll post here if the warning is still showing up.

About my name, no problem!
I've gotten used to it.

Thanks again.

Regards,
Willian

willian-gvs, 2017-07-26:
Perfect!

The warning is no longer appearing.

Thank you very much for your support, Benoît!

Regards,
Willian

closed 2017-07-26 by willian-gvs