bblanchon/ArduinoJson #561: using DynamicJsonBuffer inside of a class instead of StaticJsonBufferBase (pre 5.11 version)

ewaldc, 2017-07-26:
Hi Benoit,
Apologies for filing a new question for this really cool library.
Pre-5.11, I used StaticJsonBufferBase inside in my class library.  This gave me full control of the allocation of the buffer space (stack, heap, flash), the scope in relation to the class objects, ability to deal with JSON strings of dynamic length (unknown at compile time) and the performance of StaticJsonBuffer (a single flat buffer space) etc.  A major disadvantage is that I have to get the length of the buffer right (no auto extension) using estimation code and now, since 5.11, the reality that it longer works.
Hence, I followed your suggestion to use DynamicJsonBuffer(bufferSize) instead.  Recoding was easy, the clear() function allows me to free the memory when the buffer is no longer needed ahead of the disposal of an instantiated class object, performance difference is small, but..  I was unable to solve the scope issue :-(.

The original class definition (shortened):
``
class OpenHab {
protected:
  char *_sitemapBuffer;
  StaticJsonBufferBase *_sitemapJsonBufferPtr;
   JsonVariant _siteMap;
}
``

and code
```
_sitemapBuffer = (char *) malloc (size);  // allocate the buffer
StaticJsonBufferBase sitemapJsonBuffer = StaticJsonBufferBase(_sitemapBuffer, size);
_sitemapJsonBufferPtr = &sitemapJsonBuffer;   // save buffer just in case
_siteMap = sitemapJsonBuffer.parse(sitemapJson, 20);  // increase nestingLimit !
```

The new class definition (shortened):
``
class OpenHab {
protected:
  DynamicJsonBuffer _sitemapJsonBuffer;
   JsonVariant _siteMap;
}
``

and parsing code
```
DynamicJsonBuffer _sitemapJsonBuffer(size);
_siteMap = _sitemapJsonBuffer.parse(sitemapJson, 20);  // increase nestingLimit !
```
The problem is that _sitemapJsonBuffer seems to go out of scope at the end of the function containing the parsing code.  Adding a class initializer e.g. _sitemapJsonBuffer(1) did not solve it either.

I am sure you answered this once before, but I searched for an hour and could not find it.

**Platform**:  NodeMCU 1.0/ESP 12E with ESP8266 board manager 2.3.0
**Library**:    ArduinoJson 5.11.1
**IDE**: Arduino IDE 18.3

Thanks in advance,
Ewald

bblanchon, 2017-07-26:
Hi Ewald,

Try something like this:

```c++
class OpenHab { 
public:
   OpenHab() : _sitemapJsonBuffer(size)
   {
   }

   void parse(const char* sitemapJson)
   {
       _sitemapJsonBuffer.clear();
       _siteMap = _sitemapJsonBuffer.parse(sitemapJson, 20);
   }

protected:
   DynamicJsonBuffer _sitemapJsonBuffer;
   JsonVariant _siteMap;
};
```

The important part is the constructor of `OpenHab` that creates the `DynamicJsonBuffer` of the right size.

Regards,
Benoit

ewaldc, 2017-07-26:
Hi Benoit,
Thanks for this quick response.
The OpenHab class encapsulates a REST based web server. JSON strings are read from files or flash or can be loaded. As they get uploaded, they are parsed with DyamicJsonBuffer's and the buffer size is stored with the string.
Hence, at the time of OpenHab class instance creation, the actual size of the JSON strings is unknown.
Using an initializer of size=1 (OpenHab() : _sitemapJsonBuffer(1)) works, but now I am back with the traditional (slower) DyamicJsonBuffer mode of operation which causes "out of memory" situations for large sized JSON strings due to the way it doubles the buffer sizes of each segment.  That is why I ended up using StaticJsonBufferBase.    
Using this model, the only way I see it come close to StaticJsonBufferBase is if the clear() function would take an optional size argument.    Having followed the previous discussions around the need for the clear() function, I know this is delicate subject.   But now that StaticJsonBufferBase no longer works, maybe there is a case for it...
Ewald

bblanchon, 2017-07-26:
Hi Ewald,

I think you can get what you want by allocating a `StaticJsonBuffer` in the heap:

```c++
_sitemapJsonBufferPtr = new StaticJsonBuffer<size>();
```
Regards,
Benoit

ewaldc, 2017-07-26:
Interesting. Will give it a try right away and report back.
My lack of knowledge on C++ and it’s evolution is clearly giving me some challenges to come up with the solution myself despite having access to all the ArduinoJson source code.
One again, tx so much for this great library and your support.
Ewald

ewaldc, 2017-07-26:
Hmm, on version 5.10.1 it expects <size> to be a constant which it isn't since I read the size from a pre-parsed string from file or memory.  Changing size to a const does not help either.
`...\OpenHAB.cpp:138:48: error: 'size' is not a constant expression`
When I change to this, it all works of course ...
`_sitemapJsonBufferPtr = new StaticJsonBuffer<5820>();`

I will switch to 5.11.1 and test again

bblanchon, 2017-07-26:
What if I change`DynamicJsonBuffer` so that it doesn't double it's size after calling `clear()`?
Would that solve you problem?

ewaldc, 2017-07-26:
The fastest validated fix:
`
void clear(size_t size) {
    clear();
   _nextBlockCapacity = size;
  }
`
and everything works using class initializer sitemapJsonBuffer(0) and _sitemapJsonBuffer.clear(size) in class functions.  Identical functionality to DynamicJsonBuffer(size) but without class instantiation. 
Or in a single function
`  void clear(size_t size = 0) {
    freeAllBlocks();
    _head = 0;
    if (size != 0) _nextBlockCapacity = size;
  }
` 
Ewald.

bblanchon, 2017-07-26:
I was thinking of this:

```c++
void clear()
{
    _nextBlockCapacity = capacity();
    clear();
}
```

where `capacity()` returns the sum of all blocks capacity.

Sounds good?

ewaldc, 2017-07-26:
In my case that won't work since I initialize the class with size = 0 `sitemapJsonBuffer(0)` as I have no idea what the first JSON string to parse will be as it's the client user interface on Android that will determine that.
I would assume that in that case capacity would be zero.
Passing an optional size would make the first block the exact size, which is the closest one can get to a StaticJsonBuffer, but with the comfort of auto-extension just-in-case.
Ewald

ewaldc, 2017-07-26:
Better than my previous suggestion for a combined solution would be:

ewaldc, 2017-07-26:
Maybe a better solution would be
`  void clear(size_t size = 0) {
    freeAllBlocks();
    _head = 0;
	_nextBlockCapacity = (size == 0) ? 256 : size;
  }
`
When clear() (or clear(0)) is called it resets _nextBlockCapacity to the default initial capacity of a freshly instantiated DynamicJsonBuffer.   In the current implementation, it looks to me (but I could be wrong) that the first block allocated after a clear() would be twice the last allocated buffer capacity and if so, that would not be a desired behavior. 
When clear(size) is called however, the next (= first) block will get the specified size, but the subsequent one will get double that (potentially large) size.   For large strings that better be the correct size on an embedded platform like Ardiuino/ESP8266.  Maybe here the trick would be to cap the _nextBlockCapacity to e.g. 2K or 4K or, better, to start the incremental size increase from the _second_ block (second block would be 256 and double in size thereafter).  Or, alternatively, when using clear(size) pre-allocate the first block to the requested size and set _nextBlockCapacity to the default initial size of 256.   
A quick solution would be to pre-allocate the first block (either default size of 256 or user requested size)
and set _nextBlockCapacity to the default size (256 today) when means the doubling of block starts at block two with moderate increase.
This sounds the best solution to me, even though one loses the lazy memory allocation feature of DynamicJsonBuffer, but a potential loss would be limited to 256 bytes.
The advantage: if you use DynamicJsonBuffer(16384) and you are wrong by one byte, it would grow the buffer initially with just 256 bytes versus 32K. 

Hope that makes sense.  Time for a glass of wine now :-)
Ewald

ewaldc, 2017-07-26:
Hello Benoit,
In code the suggestion above comes down to
```c++
  DynamicJsonBufferBase(size_t initialSize = 256)  : _head(NULL), _nextBlockCapacity(256) {
	if (initialSize != 256) addNewBlock(initialSize);  // keep lazy allocation for default case
	// addNewBlock(initialSize);   // alternative: no more lazy malloc
  }

```
which could be refined to handle failed memory allocation and define the initialSize 256 as a constant.
And for the combined clear() function:
```c++
void clear(size_t size = 0) {
    freeAllBlocks();
    _head = 0;
     if (size) addNewBlock(size);   //maintain lazy allocation for default case
     _nextBlockCapacity = 256;    //ensure first block is initialSize
  }
```
Test cases verified:
- no code or behavior changes for existing code
- default allocation DynamicJsonBufferBase()  --> 256, 512, etc block build up
- allocation of a 4-byte short but large buffer DynamicJsonBufferBase(16380) --> second block is only 256 bytes
- clear() scenario --> first block is now 256 bytes vs. determined by previous use
- clear(16380) scenario  --> second block is only 256 bytes

All in all, very minor code changes and no new functions. For sure the glass of wine helped to keep it simple...
Ewald

ewaldc, 2017-07-27:
Benoit,
I've come to realize there are at least 3 enhancement requests berried in one single question which is all but best practice from my side.

1. clear() function  does not reset _nextBlockCapacity 
2. DynamicJsonBufferBase(size) with size being large, grows with size*2 where 256 may be preferred
3. request for clear(size) function

If you are OK, I'll verify each of these one more time and provide additional data (in order not to waste your time) and file a separate issue for each.
Ewald

ewaldc, 2017-07-28:
Back to the original question, I found how to use StaticJsonBufferBase in 5.11.1.  It has actually become more easy to use since there is no longer the need to initialize StaticJsonBufferBase instances in your class.  Since the buffer is managed outside, it's OK to create StaticJsonBufferBase as a disposable instance with limited scope wherever needed.    I won't go back to StaticJsonBufferBase though, as got to appreciate the auto-extension capability of DynamicJsonBuffer(size).  It has already solved/prevented a number of difficult to trace crashes.

``
  size_t sz = 5840;
  char* buffer = (char *)malloc(sz);
  StaticJsonBufferBase jsonBuffer(buffer, sz);
``

bblanchon, 2017-08-08:
Hi @ewaldc,

I'm sorry to say that, but I won't make the requested changes.

I'm sure you understand that I cannot trade the simplicity and the genericity of the library for the needs of one developper.
ArduinoJson has existed for years without the `clear()` function, and I still think it should rarely be used.

What I can do is reset the size in the `clear()` function based on the current memory usage.
But I'll not change the size-doubling in `DynamicJsonBuffer` and I'll not add a parameter to `clear()`.

Regards,
Benoit

ewaldc, 2017-08-09:
Hi Benoit,

> I'm sure you understand that I cannot trade the simplicity and the genericity of the library for the needs of one developer.

Understood.   However, I do think that item 1. can be considered a defect.  One could expect the clear() function to restore operation as if it were a freshly created buffer and hence not to increase the size by whatever was the size of the last buffer increase as shown in the example.  Unless it's clearly documented, it will result in unpredictable behavior as with each clear() call the buffer size actually doubles form it's last usage, which renders the clear() function practically useless as shown in my example parsing a small Json after a large one.   In the current implementation, I would argue it's better to remove it. As you said: ArduinoJson has existed for years without it.

With respect to item 2. it's not really about changing the size-doubling algorithm/concept (I kept that in place) , but rather to apply the default/standard growth pattern for all DynamicjsonBuffer's to 256:512:1024 etc.
Now when one allocates DynamicJsonBuffer(2000) and this happens to be wrong by 1 byte, it would add 4000 bytes which on an embedded platform is a lot of wasted memory for 1 extra byte.
The change I proposed would be to grow the buffer always with the default growth doubling pattern: 2000+256+512+1024...   In other words: the growth doubling pattern is independent of the initial size.

Hope that makes sense,   
Ewald

bblanchon, 2017-08-20:
Hi Ewald,

I understand your position.
I initially though that restarting from the current memory usage would be a nice optimization; but, as you stated, it fails when the shape of the input changes a lot.

I now agree that `clear()` should reset to the initial conditions.
I think that this can be accomplish without any additional new member, which is good.

However, I still don't think `clear()` should have a parameter.

Regards,
Benoit

commit 57defe00 (2017-08-30): Fixed `DynamicJsonBuffer::clear()` not resetting allocation size (fixes #561)
closed 2017-08-30 by bblanchon