bblanchon/ArduinoJson #563: What should I use to store the content of a json ?

redge76, 2017-08-10:
> Indeed ArduinoJson is just a serialization library; it's not intended to be a framework that you use as a backbone for your app.

OK I get it. But I was planing to save all the configuration data of my application in a structured form as a json file.
How bad is it to keep a global Json object. For sure I could deserialize my config file to a std::map but I need to handle the different type of value (String, int, float...)
Do you know any lib/class that can store the content of a Json in an acceptable form (memory wise)?
Something like this... but for the arduino world:
http://www.boost.org/doc/libs/1_61_0/doc/html/property_tree.html

devyte, 2017-08-10:
Just implement a hierarchy of structs, convert from that to json, and
serialize to file. Reverse process to get your in-mem structs back.

redge76, 2017-08-11:
a hierarchy of structs has to be maintained each time I modify the structure of my json.
:-(

devyte, 2017-08-11:
Correct. Keep in mind that this is an embedded system with minimal RAM
available, so mem usage reduction is high priority, while maintainability
is not. A json object must be accompanied by a jsonbuffer, and jsonbuffers
can be demanding on memory.
In addition, keeping a json hierarchy in memory ties your data structure to
the json format representation, which makes for a poor design choice (e.g.:
say in the future you need xml instead of json?)
Also, consider that you should design your data hierarchy once, and modify
it few times. That means that maintenance shouldn't happen often if you
designed well, and then only during development of your application.

There are other possibilities, such as tuples or variants, but I don't know
if they are supported by the compiler.

bblanchon, 2017-08-19:
I agree with @devyte: from my experience it is best to decouple the memory representation and the serialized form as they change for different reasons.

@redge76, why don't you keep the configuration in its serialized form?

redge76, 2017-08-23:
(sorry for the late reply)
What do you mean by keeping in serialized form ? 
My config file is stored on the SPIFF filesystem. Do you mean "open, read file, parse, query value" each time I want to read a value ? I could do this. My app does not really need performance... 

Concerning the tradeoff between memory and maintainability, I'm using a ESP8266. Memory is scarce but so far it's not an issue. Compared to an ATmega328P, I have plenty of RAM

bblanchon, 2017-08-23:
Yes, that's what I meant.

However, if all the program do is read the JsonObject, then it's OK to use a global JsonObject.
The problem appears when the JsonObject is modified, because the JsonBuffer grows and there is no mechanism to shrink it.

redge76, 2017-08-23:
All it does is read and when I change config, I change a value, and save it to the file.
But at this moment I can destroy my json object and recreate it.
So I will go with a single json and see how it behaves.

bblanchon, 2017-08-30:
@redge76 What is the status? Can we close this issue?

redge76, 2017-08-30:
yes. I will choose the easy path.... go with a single JSON :-P