bblanchon/ArduinoJson #564: Very long json from stream

emilekm2142, 2017-08-18:
I use the following code to read json from Serial
`void loop() {
	if (Serial.available()) {
		DynamicJsonBuffer jsonBuffer(90000);
		JsonObject& root = jsonBuffer.parse(Serial);
		const char* a = root["type"];
		Serial.println(a);
	}}`
It works until I want to load an object that is 14200 bytes in size ( https://pastebin.com/TmS7T6hz it contains 'type' and 'imageData' that is a jpg in base64 ). I can not get any values. `Serial.println(a)` prints an empty line.
I am on ESP32 so memory is not a problem here :/. What should I do?

bblanchon, 2017-08-20:
Hi @emilekm2142,

As you can see on the link below, It works on a computer.
https://wandbox.org/permlink/BJ02ALAY5QXEATRM

How can you be sure it's not a memory problem?

bblanchon, 2017-08-30:
@emilekm2142 Did you make progress? Can we close this issue?

closed 2017-09-03 by bblanchon

emilekm2142, 2017-09-05:
It works only when the JSON is specified in the sketch. The following sketch works only with short input. Is it an arduino issue?
```

void setup()
{
  Serial.begin(115200);  
}
void loop(){  
 if (Serial.available()){  
  const size_t bufferSize = JSON_OBJECT_SIZE(2) + 14200;
  DynamicJsonBuffer jsonBuffer(bufferSize);
  JsonObject& root = jsonBuffer.parseObject(Serial);
  const char* d = root["type"];
  const char* o = root["imageData"];
  Serial.println(d);
  Serial.println(o);
  }
}
```

bblanchon, 2017-09-06:
@emilekm2142, I'm glad you finally fixed your problem 😄 

Thank you very much for the update.

Feel free to add links to your projects in the [wiki](https://github.com/bblanchon/ArduinoJson/wiki/Projects-using-ArduinoJson)