bblanchon/ArduinoJson #565: JSON parsing failed!   --> Arduino / ESP8266 Beginner need help :-)

Torben80, 2017-08-18:
Hey, i am sorry to disturb u.
Thats my first Post on GitHub, so i hope its ok, what i write.

I am in the beginning of ESP8266 or Arduino and i try to get the Status of my Alarmpanel.

If i use my MozillaBrowser to get the JSON i will open: 

http://192.168.11.65/action/sensorListGet

```
/*-secure-
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "8", type : "Door Contact", area : "1", zone : "8", name : "Wohnzimmer_DF_Links", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "9", type : "Door Contact", area : "1", zone : "9", name : "Terassentuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "10", type : "Door Contact", area : "1", zone : "10", name : "Fenster_Flur_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "11", type : "Door Contact", area : "1", zone : "11", name : "Kueche", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "12", type : "Door Contact", area : "1", zone : "12", name : "Bad_EG", attr : "Burglar", cond : "Open", battery : "", tamp : "", bypass : "No"}]
}
*/`
```

After i went step by step through your Examples (using WiF), my SERIAL prompt me:

```
connecting to 192.168.11.65
Requesting URL: /action/sensorListGet
HTTP/1.0 200 OK
Server: GoAhead-Webs
Pragma: no-cache
Cache-control: no-cache
Expires: 0
Content-Type: application/json; charset=ISO-8859-1

/*-secure-
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "8", type : "Door Contact", area : "1", zone : "8", name : "Wohnzimmer_DF_Links", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "9", type : "Door Contact", area : "1", zone : "9", name : "Terassentuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "10", type : "Door Contact", area : "1", zone : "10", name : "Fenster_Flur_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "11", type : "Door Contact", area : "1", zone : "11", name : "Kueche", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "12", type : "Door Contact", area : "1", zone : "12", name : "Bad_EG", attr : "Burglar", cond : "Open", battery : "", tamp : "", bypass : "No"}]
}
*/
----------------------JSON PARSER-------------------------
JSON parsing failed!
-------------------------------------------------------
closing connection
```

Some CodeSnippet from my Code:

```
char charVar[10];

struct UserData {
  char no[32];
  char type[32];
  char area[32];
  char zone[32];
  char name[32];
  char attr[32];
  char cond[32];
  char battery[32];
  char tamp[32];
  char bypass[32];
};
```

```
// Read all the lines of the reply from server and print them to Serial
  while(client.available()){
    line = client.readStringUntil('\r');
    Serial.print(line);
  }
  Serial.println();
  Serial.println("----------------------JSON PARSER-------------------------");

  char* charBuf = (char*)malloc(sizeof(char)*1024);  //dynamic memory allocation avoid most of ESP watch dog exeptions
  line.toCharArray(charBuf,1024);
      if (parseUserData(charBuf, &userData)) {
        printUserData(&userData);
      }
  Serial.println("-------------------------------------------------------");
 
  free(charBuf); // remember to free dynamic memory
    

  Serial.println("closing connection");
}
```

```
bool parseUserData(char* content, struct UserData* userData) {
  // Compute optimal size of the JSON buffer according to what we need to parse.
  // This is only required if you use StaticJsonBuffer.
  const size_t BUFFER_SIZE =
      JSON_OBJECT_SIZE(10);

  // Allocate a temporary memory pool on the stack
   StaticJsonBuffer<200> jsonBuffer;
  // If the memory pool is too big for the stack, use this instead:
   //DynamicJsonBuffer jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(content);

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  }
 
  //root.prettyPrintTo(Serial);   //print raw JSON data
 
  // Here were copy the strings we're interested in
  strcpy(userData->no, root["no"]);
  strcpy(userData->type, root["type"]);
  strcpy(userData->area, root["area"]);
  strcpy(userData->zone, root["zone"]);
  strcpy(userData->name,root["name"]);
  strcpy(userData->attr,root["attr"]);
  strcpy(userData->cond,root["cond"]);
  strcpy(userData->battery, root["battery"]);
  strcpy(userData->tamp, root["tamp"]);
  strcpy(userData->bypass, root["bypass"]);
 
  // It's not mandatory to make a copy, you could just use the pointers
  // Since, they are pointing inside the "content" buffer, so you need to make
  // sure it's still in memory when you read the string

  return true;
}

void printUserData(const struct UserData* userData) {
  Serial.println("");
  Serial.print("no = ");
  Serial.println(userData->no);
 
  Serial.print("type = ");
  Serial.println(userData->type);

  Serial.print("area = ");
  Serial.println(userData->area);

  Serial.print("zone = ");
  Serial.println(userData->zone);

  Serial.print("name = ");
  Serial.println(userData->name);

  Serial.print("attr = ");
  Serial.println(userData->attr);
 
  Serial.print("cond = ");
  Serial.println(userData->cond);

  Serial.print("battery = ");
  Serial.println(userData->battery);
  
  Serial.print("tamp = ");
  Serial.println(userData->tamp);
 
  Serial.print("bypass = ");
  Serial.println(userData->bypass)
tamp);
 
  Serial.print("bypass = ");
  Serial.println(userData->bypass);
```

Can u help me to Debug? 
Sorry again, but i am a real NEWBEE ;-)

bblanchon, 2017-08-20:
Hi @Torben80,

It seems that the input is a comment, so the JSON parse won't read it.
You need to skip the `/*-secure-` part.

Then, I don't think the `malloc/free` part is necessary.
Why don't you let the JSON parser pull the bits from the input stream?
See [an example here](https://bblanchon.github.io/ArduinoJson/example/http-client/).

Torben80, 2017-08-20:
Hi, and thanks for your answer. I am sitting at that problem since my post. I in the beginning of that all, and i read your examples up and down, but i cant understand :-(
I am using the wificlient and i dont know how to use the SKIP Examples in my Code :-(

I figured out, that i have to skip, thats right.

Thanks for your support on Sunday ;-)

Torben80, 2017-08-20:
Update:
I took your Script from your Post, after hours (for me) of changing it to WiFi, i have  
"No response or invalid response!" in SERIAL.

Now i Stuck here:
```

// Send the HTTP GET request to the server
bool sendRequest(const char* host, const char* resource) {
  Serial.print("GET ");
  Serial.println(resource);

  client.print("GET ");
  client.print(resource);
  client.println(" HTTP/1.0");
  client.print("Host: ");
  client.println(host);
  client.print("Authorization: Basic ");
  client.println(USER_PASS);
  client.println("Connection: close");
  client.println();
                                  
  return true;
}
```

If dont add my Part:
```
delay(10);
 while(client.available()){
 String line = client.readStringUntil('\r');
 Serial.print(line);
   }
```
 My Serial will Prompt:
Serial ready

Connecting to Foch155_WZ
```
.......WiFi connected
IP address: 
192.168.11.75
Connect to 192.168.11.65
Connected
GET /action/sensorListGet
JSON parsing failed!
Disconnect
Wait 60 seconds
```

If i change your Code to:

```
// Send the HTTP GET request to the server
bool sendRequest(const char* host, const char* resource) {
  Serial.print("GET ");
  Serial.println(resource);

  client.print("GET ");
  client.print(resource);
  client.println(" HTTP/1.0");
  client.print("Host: ");
  client.println(host);
  client.print("Authorization: Basic ");
  client.println(USER_PASS);
  client.println("Connection: close");
  client.println();

                delay(10);
                while(client.available()){
                String line = client.readStringUntil('\r');
                Serial.print(line);
              }
                                        
  return true;
}
```

...my Serial prompt me:

```
⸮Serial ready

Connecting to Foch155_WZ
.......WiFi connected
IP address: 
192.168.11.75
Connect to 192.168.11.65
Connected
GET /action/sensorListGet
HTTP/1.0 200 OK
Server: GoAhead-Webs
Pragma: no-cache
Cache-control: no-cache
Expires: 0
Content-Type: application/json; charset=ISO-8859-1

/*-secure-
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "8", type : "Door Contact", area : "1", zone : "8", name : "Wohnzimmer_DF_Links", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "9", type : "Door Contact", area : "1", zone : "9", name : "Terassentuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "10", type : "Door Contact", area : "1", zone : "10", name : "Fenster_Flur_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "11", type : "Door Contact", area : "1", zone : "11", name : "Kueche", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "12", type : "Door Contact", area : "1", zone : "12", name : "Bad_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"}]
}
*/No response or invalid response!
Disconnect
Wait 60 seconds

```
Any Ideas?

bblanchon, 2017-08-23:
Hi @Torben80,

I cannot spot the exact problem.

However, it is very likely that the input given to `parseObject()` is incorrect (e.g starts with `/*...`) or incomplete (e.g only the first line).
Please try to print the content of the variable passed to `parseObject()`.

Regards,
Benoit

Torben80, 2017-08-23:
OK: UPDATE :-) 
after i add: "Serial.println(content);" to:

```
bool parseUserData(char* content, struct UserData* userData) {
  // Compute optimal size of the JSON buffer according to what we need to parse.
  // This is only required if you use StaticJsonBuffer.
  const size_t BUFFER_SIZE =
      JSON_OBJECT_SIZE(10);
Serial.println(content);
  // Allocate a temporary memory pool on the stack
  // StaticJsonBuffer<200> jsonBuffer;
  // If the memory pool is too big for the stack, use this instead:
   DynamicJsonBuffer jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(content);
  

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  }
```
 

My Serial prints me:
```
Connecting to Foch155_WZ
.......WiFi connected
IP address: 
192.168.11.75
connecting to 192.168.11.65
Requesting URL: /action/sensorListGet
HTTP/1.0 200 OK
Server: GoAhead-Webs
Pragma: no-cache
Cache-control: no-cache
Expires: 0
Content-Type: application/json; charset=ISO-8859-1

/*-secure-
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "Open", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "8", type : "Door Contact", area : "1", zone : "8", name : "Wohnzimmer_DF_Links", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "9", type : "Door Contact", area : "1", zone : "9", name : "Terassentuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "10", type : "Door Contact", area : "1", zone : "10", name : "Fenster_Flur_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "11", type : "Door Contact", area : "1", zone : "11", name : "Kueche", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "12", type : "Door Contact", area : "1", zone : "12", name : "Bad_EG", attr : "Burglar", cond : "Open", battery : "", tamp : "", bypass : "No"}]
}
*/
----------------------JSON PARSER-------------------------

/*-secure-
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "Open", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "",
JSON parsing failed!
-------------------------------------------------------
closing connection

```

So i changed:
```
Serial.println("----------------------JSON PARSER-------------------------");

  char* charBuf = (char*)malloc(sizeof(char)*1024);  //dynamic memory allocation avoid most of ESP watch dog exeptions
  line.toCharArray(charBuf,1024);
      if (parseUserData(charBuf, &userData)) {
        printUserData(&userData);
      }
  Serial.println("-------------------------------------------------------");
```

to:
```
Serial.println("----------------------JSON PARSER-------------------------");

  char* charBuf = (char*)malloc(sizeof(char)*2048);  //dynamic memory allocation avoid most of ESP watch dog exeptions
  line.toCharArray(charBuf,2048);
      if (parseUserData(charBuf, &userData)) {
        printUserData(&userData);
      }
  Serial.println("-------------------------------------------------------");
```

And that i have guessed ;-)

But now i stuck again :-) Holy sh...!

Torben80, 2017-08-26:
Update:
After Cutting first and last line of my Code JsonCode
```
int vonPos = line.indexOf("/*-secure-"); 
    int bisPos = line.indexOf("*/");
    //Serial.print(line.substring(vonCondPos+8, bisCondPos));
    if (line.substring(vonPos, bisPos) != ""){
 //     Serial.print(line.substring(vonPos+11, bisPos-1));
      line = line.substring(vonPos+11, bisPos-1);
    }
```

Now my SERIAL prints:

```
Connecting to Foch155_WZ
.......WiFi connected
IP address: 
192.168.11.24
connecting to 192.168.11.65
Requesting URL: /action/sensorListGet

----------------------JSON PARSER-------------------------
{	senrows : [
{no : "1", type : "Remote Controller", area : "1", zone : "1", name : "Benutzer", attr : "Personal Att", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "2", type : "Remote Keypad", area : "1", zone : "2", name : "Flur_unten", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "3", type : "Door Contact", area : "1", zone : "3", name : "Haustuer", attr : "Entry", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "4", type : "BX15", area : "1", zone : "4", name : "Blinklicht", attr : "", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "5", type : "Door Contact", area : "1", zone : "5", name : "Kellertuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "6", type : "Door Contact", area : "1", zone : "6", name : "Esszimmer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "7", type : "Door Contact", area : "1", zone : "7", name : "Wohnzimmer_DF_Rechts", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "8", type : "Door Contact", area : "1", zone : "8", name : "Wohnzimmer_DF_Links", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "9", type : "Door Contact", area : "1", zone : "9", name : "Terassentuer", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "10", type : "Door Contact", area : "1", zone : "10", name : "Fenster_Flur_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "11", type : "Door Contact", area : "1", zone : "11", name : "Kueche", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"},
{no : "12", type : "Door Contact", area : "1", zone : "12", name : "Bad_EG", attr : "Burglar", cond : "", battery : "", tamp : "", bypass : "No"}]
}

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef500 end: 3ffef770 offset: 01a0

>>>stack>>>
3ffef6a0:  3ffe86d0 00000000 3ffef6c0 402021c7  
3ffef6b0:  3ffef6c0 3fff1745 3fff14ce 4020310a  
3ffef6c0:  3ffe86d0 0000071e 3fff1f6c 00001000  
3ffef6d0:  00000800 0000071e 3ffee710 00000000  
3ffef6e0:  3ffee5f0 3fff1034 3ffee710 40202425  
3ffef6f0:  3ffe86e8 00000000 000003e8 0000115c  
3ffef700:  00000000 3fff0674 00000000 00000000  
3ffef710:  00000000 00000000 00000000 00000000  
3ffef720:  3fff069c 0000001f 00000015 3fff08fc  
3ffef730:  0000072f 00000711 3ffee710 40201d7c  
3ffef740:  3ffe8890 180ba8c0 feefeffe 3ffee73c  
3ffef750:  3fffdad0 00000000 3ffee734 4020392c  
3ffef760:  feefeffe feefeffe 3ffee750 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
⸮
Connecting to Foch155_WZ
.......WiFi connected

```

Any Idea?

devyte, 2017-08-27:
Start by decoding the stack dump.

bblanchon, 2017-08-28:
@Torben80, see https://github.com/me-no-dev/EspExceptionDecoder

Torben80, 2017-08-31:
Ok, i did?!
```
Decoding 6 results
0x402021c7: findKey  at C:\Users\Torben\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/JsonObject.hpp line 305
:  (inlined by) get_impl  at C:\Users\Torben\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/JsonObject.hpp line 311
:  (inlined by) get  at C:\Users\Torben\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/JsonObject.hpp line 183
:  (inlined by) as  at C:\Users\Torben\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/JsonObjectSubscript.hpp line 64
:  (inlined by) operator char const*  at C:\Users\Torben\Documents\Arduino\libraries\ArduinoJson-master\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp line 52
:  (inlined by) parseUserData(char*, UserData*) at C:\Users\Torben\Documents\Arduino\JSON-Test2/JSON-Test2.ino line 170
0x4020310a: IPAddress::printTo(Print&) const at C:\Users\Torben\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/IPAddress.cpp line 87
0x40202425: loop at C:\Users\Torben\Documents\Arduino\JSON-Test2/JSON-Test2.ino line 122
0x40201d7c: setup at C:\Users\Torben\Documents\Arduino\JSON-Test2/JSON-Test2.ino line 56
0x4020392c: loop_wrapper at C:\Users\Torben\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Torben\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
```

And now? I cant see an error-message or something else. :-(
Maybe i am to stupid for doing my project...

King Regards Torben

devyte, 2017-09-01:
The stack dump shows that parseUserData() is crashing. On closer
inspection, you are allocating a charBuf of size 1024, and yet I estimate
your example string at slightly below 2KB. Of course your code will crash,
you're overrunning your buffer size.

bblanchon, 2017-09-12:
@Torben80 Did you make any progress?

Torben80, 2017-09-12:
Didnt have Time the last two week. I think, i will try @weekend ;-)
Thx for asking, i will inform u.

Thx

bblanchon, 2017-09-20:
@Torben80, anything new?

Torben80, 2017-09-22:
Sorry, cant do anything at the moment, pls let that issue open. I will report in a few days, i hope so.

Thx, Torben

Torben80, 2017-09-28:
No time any more, sorry.

closed 2017-09-28 by Torben80