bblanchon/ArduinoJson #566: Add support for variable float precision when generating JSON strings

mill1000, 2017-08-18:
Float values can have a large number of decimal places that may or may not be necessary to convey in a JSON string.

Being able to control the max precision of the output to a certain number of decimal places would be a nice feature.

bblanchon, 2017-08-20:
Temporary work around:

```c++
template<typename TJsonBuffer, typename TFloat>
RawJson floatToJson(TJsonBuffer& jb, TFloat value, uint8_t digits) {
  return RawJson(jb.strdup(String(value, digits)));
}

// to use it:
root["pi"] = floatToJson(jsonBuffer, 3.14156, 2);
```

Caution:
The Arduino mangler behaves incorrectly with template functions like this one.
I had to wrap the definition with an anonymous namespace.

universam1, 2017-08-29:
@bblanchon In this case the float becomes a String which is sometimes a problem.

Gulaschcowboy, 2017-09-29:
Any progress on this? The resulting string causes other problems for me. Thanks

bblanchon, 2017-10-03:
Does switching to single precision storage provides an acceptable solution?
See: https://github.com/bblanchon/ArduinoJson/issues/578#issuecomment-331409853

Gulaschcowboy, 2017-10-03:
This doesn't seem to change anything. Maybe because of using homie-esp8266 which sources ArduinoJson.h in multiple places?
So a configurable precision would be a nice thing

bblanchon, 2017-10-05:
@Gulaschcowboy, are you sure that your computation is using `double` and not `float`?

I only have this problem when using `float`, see:
https://wandbox.org/permlink/UsdaYVJ6uAfQSXuO

Gulaschcowboy, 2017-10-05:
Hi @bblanchon thanks for your support.

I'm doing roughly this (after changing all floats to double):

`
#define ARDUINOJSON_USE_DOUBLE 1
...
double weight;
double temperature0;
double temperature1;
...
void getWeight() {
  scale.set_scale(kilogramDividerSetting.get());  //initialize scale
...
    double WeightRaw = scale.get_units(3)
}
...
weight = weight / 1000 ;
getTemperatures();
and so on...

...
StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["Weight"] = weight;
    root["Temp1"] = temperature0;
    root["Temp2"] = temperature1;
    String values;
    root.printTo(values);
    Homie.getLogger() << "Json data:" << values << endl;
    jsonNode.setProperty("__json__").setRetained(false).send(values);
`
As a result I get in my mqtt:

`
homie/beute-4/temperature0/degrees 19.25
homie/beute-4/temperature1/degrees 19.31
homie/beute-4/weight/kilogram 0.06
homie/beute-4/battery/volt 3.32
homie/beute-4/data/__json__ {"Weight":0.058343151,"Temp1":19.25,"Temp2":19.3125}
`
So the measured values which are transmitted without ArduinoJson have the correct precision, while the JSON ones have not.

Only if I fake my values with a static "weight = 21.21; " your hint with #define ARDUINOJSON_USE_DOUBLE 1 works

Any ideas? 
Thank you,
Alex

bblanchon, 2017-10-09:
I'm sorry Alex, I don't understand.
Can you simplify?

Gulaschcowboy, 2017-10-09:
@bblanchon I'll try:

Your example code works only when I statically define an artificial value to e.g. weight:

    #define ARDUINOJSON_USE_DOUBLE 1
    double weight;
    weight=21.21;
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["Weight"] = weight;
    root["Temp1"] = temperature0;
    root["Temp2"] = temperature1;
    String values;
    root.printTo(values);
    Homie.getLogger() << "Json data:" << values << endl;
    jsonNode.setProperty("json").setRetained(false).send(values);

When I compute e.g. weight or temperature from received sensor values it doesn't work:

    #define ARDUINOJSON_USE_DOUBLE 1
    double weight;
    void getWeight() { blabla };
    weight = weight/1000; //g to kg
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();
    root["Weight"] = weight;
    root["Temp1"] = temperature0;
    root["Temp2"] = temperature1;
    String values;
    root.printTo(values);
    Homie.getLogger() << "Json data:" << values << endl;
    jsonNode.setProperty("json").setRetained(false).send(values);

bblanchon, 2017-10-10:
This sentence is very curious:

> all my computed values have 2 decimals

I think what you meant is that the values are printed with two decimals, which is the default for `Serial.print()`.
In this case, ArduinoJson is doing a much better job.

I am reluctant to add a variable number of decimal places, as it comes with another package of issues.
It used to be the case before ArduinoJson 5.10 and users complained a lot!

Why don't you just round the value?
Alternatively, why don't you just use this workaround: https://github.com/bblanchon/ArduinoJson/issues/566#issuecomment-323580444

PS: only use [ARDUINOJSON_USE_DOUBLE](https://bblanchon.github.io/ArduinoJson/api/config/use_double/) if it's mandatory, as it increases memory usage.

universam1, 2017-10-10:
your workaround is converting a float to a string, right? This is an issue if a specific type is required.

Gulaschcowboy, 2017-10-10:
@bblanchon Thanks for your answer. But I don't understand how to solve it.

As you can see in the above MQTT output (nope, not done by Serial.print() ) I need the values withing `__json__` as float, like in the non ArduinoJson MQTT output.
Wouldn't workaround #566 change it to:  `{"Weight":"0.06","Temp1":"19.25","Temp2":"19.31"}`?

About rounding the value: I never had to do that before. Because whatever function or library I used it defaulted to 2 decimals or had something like (val, format)

You may have good reasons for disabling (val, format):
> I am reluctant to add a variable number of decimal places, as it comes with another package of issues.
It used to be the case before ArduinoJson 5.10 and users complained a lot!

But still it was practical. At least 3 users are complaining now - see thread above ;-)

So as I'm more a beekeeper than a coder: Can someone provide an example how to round my `weight` and `temepratureX` floats to two decimal places before feeding it into ArduinoJson?

Thanks a lot

bblanchon, 2017-10-14:
@Gulaschcowboy the funcion `floatToJson()` presented above would write `{"Weight":0.06,"Temp1":19.25,"Temp2":19.31}`.
Did you give it a try?

Gulaschcowboy, 2017-10-15:
I tried to, but it doesn't compile: `error: 'RawJson' does not name a typeerror: 'RawJson' does not name a type`
To be honest, I don't get what it means and how to use, as the template concept is new to me.

bblanchon, 2017-10-19:
@Gulaschcowboy, why don't you just round the values ?

Here is a function to do that:

```c++
// rounds a number to 2 decimal places
// example: round(3.14159) -> 3.14
double round2(double value) {
   return round(value * 100) / 100.0;
}
```

universam1, 2017-10-19:
or you can do without requiring the math lib following:
```
(int)(3.14159 * 100 + 0.5) / 100.0
```

Gulaschcowboy, 2017-10-27:
Thanks guys, that solves my problem.

bblanchon, 2018-02-05:
A new workaround is available in ArduinoJson 5.13:

```c++
root["pi"] = RawJson(String(3.14159,2));
```

But I still recommend using the `round2()` defined above.

hpsaturn, 2019-04-08:
Hi, Do you know for ArduinoJson 6? I tried with [JsonVariant](https://arduinojson.org/v6/api/jsonvariant) with all forms but don't have a solution for `float` value. What is the example for **generate json** with a float value?

Some like to:

```C
void write( float hum, float tmp,  int stime)
{   
    const int capacity = JSON_ARRAY_SIZE(1) + JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(4);
    StaticJsonDocument <capacity> doc;
    JsonObject data = doc.createNestedObject();
    data["nameId"] = _nameId;
    data["sensorId"] = _sensorId;
    JsonObject fields = data.createNestedObject("fields");
    fields["hum"] = hum;
    fields["tmp"] = tmp;
    fields["stime"] = stime;
    Serial.print("payload: ");
    serializeJson(doc, Serial);
}
```
Wrong output:
```json
[{
	"nameId": "station00",
	"sensorId": "962DE6B4",
	"fields": {
		"hum": 0,
		"tmp": 0,
		"stime": 5
	}
}]
```
Expected output:
```json
"fields": {
		"hum": 0.0,
		"tmp": 0.0,
		"stime": 5
	}
```

bblanchon, 2019-04-08:
The equivalent with v6 is:

```c++
doc["pi"] = serialized(String(3.14159,2));
```

bblanchon, 2019-05-18:
I added an entry in the FAQ:

* [How to configure the serialization of floats?](https://arduinojson.org/v6/faq/how-to-configure-the-serialization-of-floats/)

closed 2019-05-18 by bblanchon