bblanchon/ArduinoJson #567: Input modification of char* produces corruption in first key of parsed object

teejK, 2017-08-22:
Hi, first of all, thank you for this amazing library.
I got some interesting behaviour using ArduinoJson to parse an MQTT message of 
AWS IoT's device shadow document.

[template JSON](https://docs.aws.amazon.com/iot/latest/developerguide/thing-shadow-document-syntax.html#thing-shadow-example-response-json)

Target platform: ESP8266
Compiler: PlatformIO 3.4.1

**Parsing a char* to JsonObject produces the following in the serial monitor:**
`
{"state":{"desired":{"welcome":"aws-iot","moisSensorCalibration":151...}}}

{"�\b":{"desired":{"welcome":"aws-iot","moisSensorCalibration":151...}}}
`
**CODE**
`
MQTT::Message &message = md.message;
char* msg = new char[message.payloadlen+1]();
memcpy(msg, message.payload, message.payloadlen);                                               
Serial.println(msg); 

DynamicJsonBuffer shadowDeltaJSONBuff;
JsonObject& rootSD = shadowDeltaJSONBuff.parseObject(msg);                                
delete msg; 
rootSD.printTo(Serial);  
 `

**The corruption of the first key makes its nested objects unreachable**
`
int desiredMoisSensorCalibration = rootSD["state"]["delta"]["moisSensorCalibration"];
if (desiredMoisSensorCalibration); // returns false
`

**I got it to work properly by first converting the char* into a String:**
`
{"state":{"desired":{"welcome":"aws-iot","moisSensorCalibration":151...}}}

{"state":{"desired":{"welcome":"aws-iot","moisSensorCalibration":151...}}}
`
**CODE**
`
MQTT::Message &message = md.message;
char* msg = new char[message.payloadlen+1]();
memcpy(msg, message.payload, message.payloadlen);
Serial.println(msg);
String msgString = String(msg);

DynamicJsonBuffer shadowDeltaJSONBuff;
JsonObject& rootSD = shadowDeltaJSONBuff.parseObject(msgString);
delete msg;
rootSD.printTo(Serial);
`

bblanchon, 2017-08-23:
Hi @teejK,

There is a "use-after-free" in this program:

```c++
DynamicJsonBuffer shadowDeltaJSONBuff;
JsonObject& rootSD = shadowDeltaJSONBuff.parseObject(msg);                                
delete msg; // free
rootSD.printTo(Serial); // use-after-free  
```

Indeed, when `parseObject()` is called with a `char*`, it doesn't make a copy of the input, but instead store pointers; this means that `rootSD` contains pointers to `msg`.

There are tree possible solutions:

1. Postpone the `delete msg`
2. Cast `msg` to a `const char*`
3. Pass `message.payload` directly to `parseObject()`

The last one is probably the best as it would avoid a memory allocation and copy.

See:

* [API Reference: JsonBuffer::parseObject()](https://bblanchon.github.io/ArduinoJson/api/jsonbuffer/parseobject/)

Regards,
Benoit

teejK, 2017-08-24:
@bblanchon thanks for the info! 

Sorry, still new to programming in C++ (and programming in general).

closed 2017-08-24 by teejK