bblanchon/ArduinoJson #569: JsonVariant has no iterator

universam1, 2017-08-28:
I need to use the JsonVariant (class member declaration) but I face an issue for not being able to iterate through a JsonVariant Object as begin() and end() are not available.

Any ideas in this regard?

Thank you for the great work with this library!

bblanchon, 2017-08-30:
Hi @universam1,

Indeed there is no iterator on `JsonVariant` because it would be ambiguous.
Before iterating on a variant, the program must cast it to either a `JsonArray` or `JsonObject`.

For instance, for an object:

```c++
for( const auto& kv : variant.as<JsonObject>() ) {
    Serial.println(kv.key);
    Serial.println(kv.value.as<char*>());
}
```

and for an array:

```c++
for( const auto& value : variant.as<JsonArray>() ) {
    Serial.println(value.as<char*>());
}
```

Regards,
Benoit

universam1, 2017-08-30:
Amazing, thanks for the explanation!

You got my 🌟

bblanchon, 2017-08-30:
You're welcome.
Please apology for the late answer.

closed 2017-08-30 by bblanchon

universam1, 2017-08-30:
Never mind! Would it make sense to add it to the bag of tricks?

bblanchon, 2017-08-30:
I'll add it in the FAQ as soon as possible, but feel free to edit the wiki.

universam1, 2017-08-30:
did it:
https://github.com/bblanchon/ArduinoJson/wiki/Examples#jsonvariant
hope thats fine!

universam1, 2017-10-06:
Hi @bblanchon I think the examples are not totally correct. 

I can't get `Serial.println(kv.value.as<char*>());` working at all, actually it breaks something and the object is gone!
Now, it looks like this expression without casting works as desired: `Serial.println(kv.value);`
What do you think?