bblanchon/ArduinoJson #571: Suggestion for an example.

nardev, 2017-09-02:
I know that you dislike global jsonbuffer/object but i think that many ppl just want to have set of seettings available with boot time and most of them should be unchanged as long as device is on.

IMHO that would be often used pattern.

bblanchon, 2017-09-03:
In that situation, I use custom data structures as suggested here in [FAQ: What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library/)

This is a win in three levels:

1. Faster code (the object tree is walked only once at boot time)
2. Smaller RAM footprint (JsonObject, JsonVariant... have significant overhead)
3. Smaller code

Reducing the size of global variables is crucial as they reduce the RAM remaining for the rest.
Moreover, the JSON object tree consumes memory even if the program doesn't use it.
For example, if someone adds values in the JSON settings that are not actually used by the program, the memory will be used anyway.
One could even get to a situation where the program is unable to run because all the memory will be consumed by the settings.

So I see this more as an anti-pattern, but I'd like to get you opinion @nardev...
What is the benefit compare to custom data structures?

bblanchon, 2017-09-04:
PS: I forgot another strong advantage of custom structures:

 4. It decouples the memory representation from the file format,
     which will be very handy when the file format evolves

nardev, 2017-09-04:
Didn't notice this one in the F.A.Q. thnx.

nardev, 2017-09-05:
Should we close this..??

bblanchon, 2017-09-06:
This is a question that pops up very often, let's keep this open for a while and see if we get some reactions.

Then, I'll add my answer to the FAQ, where it belongs, and close this issue.

Thank you very much for asking this @nardev, I'm sure this will help a lot of users.

commit f4332aa6 (2017-09-14): Add the "global JsonObject configuration" anti-pattern (closes #571)
closed 2017-09-14 by bblanchon