bblanchon/ArduinoJson #573: Parsing JSON from server.arg

MPEDI, 2017-09-08:
Hi

I have some problems in parsing Json data received through a web server on an ESP8266 (Wemos D1).
Here the parsing function i use (created and checked by using the ArduinoJson assistant) :

    void setSaveData()
    {
      Serial.println("Data Received, saving to internal variable!");
      Serial.print("Data received: ");
      String data = server.arg("plain");
      Serial.println(data);

      const size_t bufferSize = 10*JSON_ARRAY_SIZE(5) + JSON_ARRAY_SIZE(10) + JSON_OBJECT_SIZE(2) + 180;
      DynamicJsonBuffer jsonBuffer(bufferSize);

      JsonObject& root = jsonBuffer.parseObject(data);
      if (!root.success())
        {
          Serial.println("parseObject() failed");
          server.send(204,"");
          return;
        }

      int Idx = root["t"]; // 0
      JsonArray& V = root["V"];

      for (int i = 0; i<10; i++)
      {
        Serial.print("time slot : ");
        Serial.println(Idx + i);
        JsonArray& Vi = V[Idx + i];
        for (int j = 0; j<5; j++)
        {
          char cara = Vi[j];
          pattern[Idx][j] = Vi[j];
          Serial.print("     Value     : ");
          Serial.println(cara);
        }
      }
      server.send(204,"");
    }

I send data using the following command:

    >curl -H "Content-Type: application/json" -X POST -d '{"t":0,"V":[[82,71,87,71,82],[71,82,87,82,71],[82,71,87,71,82],[82,71,87,71,82],[89,89,89,89,89],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71]]}' http://192.168.1.146:8001/savedata	

I always get the following in the serial monitor:

    Data Received, saving to internal variable!
    Data received: '{t:0,V:[[82,71,87,71,82],[71,82,87,82,71],[82,71,87,71,82],[82,71,87,71,82],[89,89,89,89,89],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71]]}'
    parseObject() failed

So parsing always fails!

If instead of getting the data from the server.arg i put in the code:

    const char* data = "{\"t\":0,\"V\":[[82,71,87,71,82],[71,82,87,82,71],[82,71,87,71,82],[82,71,87,71,82],[89,89,89,89,89],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71],[71,82,87,82,71]]}";

The parser works! so it shall not be a buffer size problem (i also tried to double or even triple the buffer size).

Do you have some hints (did not find anything working in the examples and faq)

ewaldc, 2017-09-12:
Hello,
Take a look [here](https://stackoverflow.com/questions/37468775/server-args-esp8266-arduino), looks similar and highlights some gotchas which I also encountered.  That said, your code example does not seem to hit some of the issues I hit such as "=" sign in the Json data,  Json parsing depth of 10+ etc.  
I am doing something similar with no issues. 

What happens when you change to **const** String data = server.arg("plain")?   Are you using Arduino ESP?  If so, list version.   I ended up modifying JsonParserImpl.hpp with #ifdefs to add debug information in order to find out where parsing fails (code attached).   On an ESP platform with it's limited memory, parsing can fail for many reasons...  You may want to give that a try...
[JsonParserImpl.zip](https://github.com/bblanchon/ArduinoJson/files/1295207/JsonParserImpl.zip).
Ewald

MPEDI, 2017-09-12:
Hi, I had a look at your link, yes the issue is similar but as you wrote I do not have problems with "=" sign or parsing depth. In fact if I put the Json directly in the code everything works well.

Even following your suggestion to use const String data... doe not help. the result is always the same.

To answer your questions, yes i'm using Arduino ESP, I use the latest available version (2.3.0) with Arduino IDE 1.8.3.
I will try with your modified code to see if I can get more debug info, to understand what is going wrong where, the things that are fully unclear to me is again that if I put the data directly in the code with const String data = server.arg("plain") the things are working so the available and used memory shall be the same. 
I will keep posted here after trying the "Debug enhanced" code.
Thanks for your help 
mpedi

bblanchon, 2017-09-12:
Hi @MPEDI,

> I send data using the following command:
>
>     curl -H "Content-Type: application/json" -X POST -d [...]

That's a little disturbing.. what happened to the quotes?

Did you try to `strcmp()` the string that works with the data received?
It seems to me that they don't match.

Regards,
Benoit

closed 2017-09-25 by bblanchon