bblanchon/ArduinoJson #574: Compiling error using Arduino IDE for NodeMCU V1.0

asim97, 2017-09-12:
I added the ArduinoJson in my project like that

`#include <ArduinoJson.h>`

and i got  a huge error 

`Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\ABDULL~1\AppData\Local\Temp\arduino_cache_693802\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadTool_esptool,UploadSpeed_115200,FlashSize_4M3M_0a0e64a748c0fba0e230876f529f0a28.a
C:\Users\ABDULL~1\AppData\Local\Temp\arduino_build_70921/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'

C:\Users\ABDULL~1\AppData\Local\Temp\arduino_build_70921/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'

C:\Users\ABDULL~1\AppData\Local\Temp\arduino_build_70921/arduino.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':

C:\Users\Abdullah Assem\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp:97: undefined reference to `setup'

C:\Users\Abdullah Assem\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.1.0\cores\esp8266/core_esp8266_main.cpp:97: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

asim97, 2017-09-12:
UPDATE :::::: problem fixed 
i was using Firebase Arduino library which depends on ArduinoJson 
there was conflict between the two libraries 
i found there is a version of  ArduinoJson  incuded in Firebase library
link 

> https://github.com/firebase/firebase-arduino/tree/master/src/third-party/arduino-json-5.6.7

added this to my Arduino library folder instead of official one and worked fine

bblanchon, 2017-09-14:
Thanks for the update @asim97

closed 2017-09-14 by bblanchon

lprasanth, 2018-05-21:
Hello can you please update the link Iam not able to find the page@asim977

asim97, 2018-05-21:
[image]

@lprasanth you don't need to download the third party library any more !!

but if you still need the modified library, I still have it !
[arduino-json-5.6.7.zip](https://github.com/bblanchon/ArduinoJson/files/2022492/arduino-json-5.6.7.zip)

Have a nice Day 🥇

lprasanth, 2018-05-22:
Thank you so much can you tell me through which it can be configured.

asim97, 2018-05-22:
if you still want to use the "old library of arduino-json-5.6.7" along with "old library of firebase"
which i don't recommend....
you have to extract "arduino-json-5.6.7.zip" at 
[image]
then you can add the #included  at you code without any problem like that
[image]
and to prevent any problem , I will upload the firebase library i used (OLD)
[firebase-arduino.zip](https://github.com/bblanchon/ArduinoJson/files/2026360/firebase-arduino.zip)
I hope that helped you ,,,,,, Have a great Day 😄