bblanchon/ArduinoJson #575: Parse response JSON from localhost

francisco0327, 2017-09-16:
Hi, in advance thanks for your time and help, also apologize if I use some term not correct; I am a beginner in arduino subjects. I have an arduino yun which I would like to be able to read the following response from a url in json.

`{
    "status": 1,
    "pir": 0,
    "pro": 1
}`

I have tried to parse this answer through the example guide provided by the library, but I can not. The code:

```
int pir;
int pro;
int estado;
char* jsonRaw;
Process device;

String ID_DEVICE = "A-2DSSW2";
StaticJsonBuffer<200> jsonBuffer;
String URL_POST = "localhost:8080/post";

void setup() {
  Bridge.begin();
  Serial.begin(9600);
}

void loop() {
  device.begin("curl");
  device.addParameter("-k");
  device.addParameter("-X");
  device.addParameter("POST");
  device.addParameter("-H");
  device.addParameter("Content-Type:application/x-www-form-urlencoded");
  device.addParameter("-H");
  device.addParameter("Accept:application/json");
  device.addParameter("-d");
  device.addParameter("device=" + ID_DEVICE);
  device.addParameter(URL_POST);
  device.run();
  
  while (device.available()) {
    jsonRaw = device.read();
  }

  JsonObject& jsonParsed = jsonBuffer.parseObject(jsonRaw);
  estado = jsonParsed["status"];
  
  Serial.println(estado);
  Serial.flush();
  delay(5000);
}
```

I'm sorry for the inconvenience, thank you all.

bblanchon, 2017-09-16:
Hi @francisco0327,

I noticed something really strange in the program:

```c++
char* jsonRaw;
...
jsonRaw = device.read();
```

It seems that `jsonRaw` stores a pointer to a temporary object.
I think `jsonRaw` should be a `char[]` or a `String`.

Did you try to print the content of `jsonRaw` to the Serial?

francisco0327, 2017-09-16:
thanks for your prompt response.

I tried to save what `device.read ()` returns in a `String`, but when it prints it returns the length.

I have printed the contents returned through a variable `Char` and also directly to the Serial.

francisco0327, 2017-09-17:
Good day to all. I solved the problem. What I did was replace the` read()` with the `readString() ` and instead of storing the response in a` char` I did it in a `String`.

This works, but after the third `loop` everything was returned in` 0`, apparently the `StaticJsonBuffer` was filled and what I did was move it from the beginning where the variables are declared to the loop function. Then work with liberia as indicated in the examples. **Then I leave the code**. Thanks for your help @bblanchon 

**En Español**
Buen dia para todos. Resolvi el problema que plantea. Lo que hice fue reemplazar el `read()` por el `readString()` y en vez de almacenar la respuesta en un `char` lo hice en un `String`.

Así funciona, pero despues de tercer `loop` todo era devuelto en `0`, al parecer el `StaticJsonBuffer` se llenaba y lo que hice fue moverlo del inicio donde se declaran las variables a la funcion loop. Luego se trabaja con la liberia como indican en los ejemplos. **A continuación les dejo el código**. 

```
#include <Bridge.h>
#include <Process.h>
#include <ArduinoJson.h>

const ID_DEVICE = "A-ADAD323SD2"; 
const String URL_STATUS = "http://YOUR_IP:8080/status";

int pir;
int pro;
Process device;
String jsonRaw;
int status = 0;

void setup() {
  Bridge.begin();
  Serial.begin(9600);
}

void loop() {

  StaticJsonBuffer<200> jsonBuffer;

  device.begin("curl");
  device.addParameter("-k");
  device.addParameter("-X");
  device.addParameter("POST");
  device.addParameter("-H");
  device.addParameter("Content-Type:application/x-www-form-urlencoded");
  device.addParameter("-H");
  device.addParameter("Accept:application/json");
  device.addParameter("-d");
  device.addParameter("device=" + ID_DEVICE);
  device.addParameter(URL_STATUS);
  device.run();

  while (device.available()) {
    jsonRaw = device.readString();
  }
  
  JsonObject& jsonParsed = jsonBuffer.parseObject(jsonRaw);
  
  status = jsonParsed["status"];
  pir = jsonParsed["pir"];
  pro = jsonParsed["pro"];
  
  Serial.println(status);
  Serial.println(pir);
  Serial.println(pro);

  delay(60000);
}
```

bblanchon, 2017-09-18:
@francisco0327 I'm glad you made it work!

You can probably improve the program by replacing:

```c++
while (device.available()) {
    jsonRaw = device.readString();
}
  
JsonObject& jsonParsed = jsonBuffer.parseObject(jsonRaw);
```

by simply:

```c++
JsonObject& jsonParsed = jsonBuffer.parseObject(device);
```

This avoids the temporary `String`.

See:

* [Can I parse data from a stream?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-data-from-a-stream/)
* [How to reduce memory usage?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reduce-memory-usage/)

closed 2017-09-18 by bblanchon