bblanchon/ArduinoJson #577: Avoiding pitfalls No. 6 - duplicates for other data types

fedy0, 2017-09-17:
As shown in example of "Avoiding pitfalls" section No. 6, How can I duplicate for an array of integers?

Thanks in advance.

bblanchon, 2017-09-20:
Hi @fedy0,

I'm not sure I understand the question...
The duplication problem only occurs with strings because a `JsonVariant` doesn't store the string itself but only a pointer.
On the other hand, integers are directly stored in the `JsonVariant`, ie a copy is always made.

Do you have a problem in a particular program?
If so, please share an [MCVE](https://stackoverflow.com/help/mcve).

Regards,
Benoit

fedy0, 2017-09-20:
Here is an MCVE:

```

#include <ArduinoJson.h>
void appendData (JsonArray& data);
char times[][20] = {
  "2017-09-20 20:36:17",
  "2017-09-20 21:09:56",
  "2017-09-20 22:02:33",
};
void setup() {
  Serial.begin(115200);
  Serial.println();
}

void loop() {
  DynamicJsonBuffer jsonBuffer;
  JsonObject& response = jsonBuffer.createObject();
  response["sensor"] = "gps";
  // Including data k-v pair. data's value is an array of arrays
  JsonArray& data = response.createNestedArray("data");
  appendData (data);
  response.printTo(Serial);
  Serial.println();
  delay(1000);
}

void appendData (JsonArray& data) {
  // ...
  for (uint8_t i = 0; i < 3; i++) {
    DynamicJsonBuffer jsonBuffer2;
    JsonArray& array = jsonBuffer2.createArray();

    array.add(times[i]);      // Include time in array (data type is char string)

    for (uint16_t j = 0; j < 5; j++)
      array.add(i + j);       // Include other data in array (data type is int)

    array.printTo(Serial);    // Included for inspection in this for...loop
    Serial.println();
    data.add(array);          // array content here is of MIXED data type
  }
}

// Result:
/*
  {
  "sensor":"gps"
  "data":[
        ["2017-09-20 22:02:33",2,3,4,5,6],
        ["2017-09-20 22:02:33",2,3,4,5,6],
        ["2017-09-20 22:02:33",2,3,4,5,6]
       ]
  }
*/

// Expected result:
/*
  {
  "sensor":"gps"
  "data":[
        ["2017-09-20 20:36:17",0,1,2,3,4],
        ["2017-09-20 21:09:56",1,2,3,4,5],
        ["2017-09-20 22:02:33",2,3,4,5,6]
       ]
  }
*/

```
Thanks

bblanchon, 2017-09-21:
Hi @fedy0 ,

Indeed, this program adds references to temporary arrays.

The solution is to create the nested array within the original `JsonBuffer`.
`JsonArray` provides a shortcut for that: `createNestedArray()`.

Just replace:

```c++
DynamicJsonBuffer jsonBuffer2;
JsonArray& array = jsonBuffer2.createArray();
```

with:

```c++
JsonArray& array = data.createNestedArray();
```

This is equivalent to `createArray()` and `add()`, so you need also to remove `data.add(array)`.

See:

* [API Reference: JsonArray::createNestedArray()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/createnestedarray/)

Regards,
Benoit

fedy0, 2017-09-21:
Excellent!!! It worked
Thanks a lot Benoît

closed 2017-09-21 by fedy0