bblanchon/ArduinoJson #579: Post in webservice

LucasSantosSilva, 2017-09-21:
I want send for my webservice, but this code does not send. I used postman and structure is ok! 

`StaticJsonBuffer<256> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
root["categoria"] = categoria;
root.prettyPrintTo(Serial);

  if (client.connect(serverClient, 3000)) {
   // Serial.println("conectando...");
    // send the HTTP PUT request:
    client.println("POST /Proto/ HTTP/1.1");
    client.println("Host: 192.168.20.136");
    client.println("User-Agent: Arduino/1.0");
    client.println("Accept: application/json");
    client.print("Content-Length: ");

    root.prettyPrintTo(client);
    
    client.println("Content-Type: application/json");
   client.println("Connection: close");
    client.println();

  } 
  else {
    // if you couldn't make a connection:
   Serial.println("falha na conexao");
    Serial.println();
    //Serial.println("disconectado.");
    client.stop();
  }
   // note the time that the connection was made or attempted:
  //lastConnectionTime = millis();
}`

How I do?

bblanchon, 2017-09-23:
Hi @LucasSantosSilva,

From looking at this program, it seems that the HTTP POST is incorrect:

```http
POST /Proto/ HTTP/1.1
Host: 192.168.20.136
User-Agent: Arduino/1.0
Accept: application/json
Content-Length: {
  "categoria": "content of the categoria variable"
}Content-Type: application/json
Connection: close
```

The correct syntax is:

```http
POST /Proto/ HTTP/1.1
Host: 192.168.20.136
User-Agent: Arduino/1.0
Accept: application/json
Content-Type: application/json
Connection: close

{
  "categoria": "content of the categoria variable"
}
```

`Content-Length` is optional; if you want it, you can call [`JsonObject::measurePrettyLength()`](https://bblanchon.github.io/ArduinoJson/api/jsonobject/measureprettylength/).

Regards,
Benoit

LucasSantosSilva, 2017-09-24:
Thanks man, that solved 👍 .

In my case I needed use Content-Length.

closed 2017-09-24 by LucasSantosSilva