bblanchon/ArduinoJson #582: Can't load JsonObjects from SPIFFS

zibous, 2017-09-26:
I try to load the configuration, the loading process works, but the configuration is not available. 
If this is done in the memory, then it works.

What am I doing wrong or can not this be done with ArduinoJson?

Testcase 1:
1. Loading from SPIFFS works
2. cfgBuffer.parseObject works

Result: Configuration not loaded (see setup)

Testcase 2:
1. createNestedObject... works

Result: Configuration loaded !

```

#include <ArduinoJson.h>
#include "FS.h"
#include "config.h"

DynamicJsonBuffer cfgBuffer;
JsonObject& cfg = cfgBuffer.createObject();

bool getConfig() {

    bool result = SPIFFS.begin();
    size_t n = 0;
    if (result) {
        // try to load the confiaguration data
        File f = SPIFFS.open(APP_CONFIGFILE, "r");
        if (f) {
          size_t size = f.size();
          if ( size == 0 ) {
             f.close();
          }else{
            std::unique_ptr<char[]> buf (new char[size]);
            f.readBytes(buf.get(), size);
            JsonObject& cfg = cfgBuffer.parseObject(buf.get());
            if (!cfg.success()) {
              Serial.println("Parsing json failed !");
            }
            f.close();
            n = cfg.size();
            Serial.print("Configuration from SPIFFS loaded: ");
            Serial.print(n);
            Serial.println(" records found.");
            n = 0;
          }
        }
    }

    if(n == 0){
      // no data loaded, create the default
      JsonObject &_wifi = cfg.createNestedObject("wifi");
      _wifi["ssid"]     = WIFI_SSID;
      _wifi["password"] = WIFI_PASSWORD;
      _wifi["hostname"] = "SMDK12";

      JsonObject &_ntp = cfg.createNestedObject("ntp");
      _ntp["enabled"]    = false;
      _ntp["server1"]    = NTP_SERVER1;
      _ntp["server2"]    = NTP_SERVER2;
      _ntp["server3"]    = NTP_SERVER3;
      _ntp["timeoffset"] = NTP_TIME_OFFSET;
      _ntp["daylight"]   = NTP_DAY_LIGHT;
      _ntp["upinterval"] = NTP_UPDATE_INTERVAL;
      _ntp["enabled"]    = true;
    }

    return true;

}

void setup() {

  Serial.begin(SERIAL_BAUD);
  Serial.println();

  while (!Serial) {
    ; // wait for serial port to connect. Needed for Leonardo only
  }

    if(getConfig()){
        if(cfg.size()){
            Serial.println("Configuration loaded !");
        }else{
            Serial.println("Configuration not loaded !");
        }
    }

}

void loop() {

}
```

bblanchon, 2017-09-27:
Hi @zibous,

The `cfg` variable in the `getConfig()` function is shadowing the global one.
That explains why the global `cfg` is empty.

Make sure you read:

* [Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library/)
* [Can I parse data from a stream?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-data-from-a-stream/)

Regards,
Benoit

zibous, 2017-09-27:
Thanks for the reply. 

My configuration is dynamic and it is almost 
impossible to deserialize the data and store it in a structure.

As an alternative, I only can save the json data as a string and 
then retrieve the values via a get method.

```
String config;  // the current configuration

void loadConfig(){
	SPIFFS.begin();
	File file = SPIFFS.open(/"config.json", "r");
	config = file.readString();
    file.close()
}

String configGet(String key, String defaultValue) {
    String ret = "";
    DynamicJsonBuffer jsonBuffer;
    JsonObject &root = jsonBuffer.parseObject(config);
    if (root.containsKey(key)){
        const char *value = root[key];
        return String(value);
    }else {
        return String(defaultValue);
    }
}
```

Probably not very performant.

closed 2017-10-05 by bblanchon