bblanchon/ArduinoJson #584: Crash on ESP32 with invalid JSON input OR doubles

vyruz1986, 2017-10-01:
I'm trying to use this library on an ESP32 project I'm doing. My project kept crashing when trying to create JSON objects, so I started experimenting with the examples that come with this library.

First of all, when trying to decode invalid JSON data, this makes the ESP32 crash. I used example [`JsonParserExample.ino`](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonParserExample/JsonParserExample.ino) and changed [line 34](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonParserExample/JsonParserExample.ino#L34) from the valid JSON string to just `char json[] = "testing";`  
This causes the sketch to crash when calling `parseObject()` at line 41. I decoded the strack trace and this was the output:
```
0x400d2c8b: __cxa_guard_release at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc line 400
0x400d093a: ArduinoJson::JsonObject::invalid() at c:\Users\user\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp line 289
0x400d0d43: ArduinoJson::JsonObject::invalid() at c:\Users\user\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp line 289
:  (inlined by) ArduinoJson::Internals::JsonParser  ::Reader, ArduinoJson::DynamicJsonBufferBase &>::parseObject() at c:\Users\user\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParserImpl.hpp line 129
0x400d1079: ArduinoJson::JsonObject& ArduinoJson::JsonBufferBase   >::parseObject (String&, unsigned char) at c:\Users\user\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp line 98
:  (inlined by) setup() at C:\Users\user\AppData\Local\Temp\VMBuilds\JSONTest\espressif_lolin32\Debug/JSONTest.cpp line 45
0x400dad4b: loopTask(void*) at c:\Users\user\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15
```
The `parseObject()` function is supposed to handle invalid JSON correctly, right?

Regardless of the above, the `JsonParserExample.ino` example would keep crashing on me, even in its original form. I tracked this down to being related to the doubles which are in the sample JSON string to be parsed.  
If I changed line 34 from:  
`"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";`
to:  
`"{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48,2]}";`  
loosing the decimals in the GPS data, the example ran fine.

When I left the decimals in place, the crash stack trace decoded showed the following:
```
0x400d2d47: __cxa_guard_release at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc line 400
0x400d0e3d: ArduinoJson::TypeTraits::FloatTraits ::negativeBinaryPowerOfTen(int) at c:\users\10049636\onedrive\projects\arduino\libraries\arduinojson\src\arduinojson\typetraits/floattraits.hpp line 62
:  (inlined by) double ArduinoJson::TypeTraits::FloatTraits ::make_float (double, int) at c:\users\10049636\onedrive\projects\arduino\libraries\arduinojson\src\arduinojson\typetraits/floattraits.hpp line 41
0x400d0fe1: double ArduinoJson::Polyfills::parseFloat (char const*) at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/Polyfills/parseFloat.hpp line 88
0x400d102a: double ArduinoJson::JsonVariant::variantAsFloat () const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonVariantImpl.hpp line 92
:  (inlined by) ArduinoJson::TypeTraits::EnableIf  ::value, double>::type const ArduinoJson::JsonVariant::as () const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonVariant.hpp line 168
:  (inlined by) ArduinoJson::Internals::JsonVariantAs ::type ArduinoJson::JsonArray::get (unsigned int) const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonArray.hpp line 117
0x400d11a5: ArduinoJson::Internals::JsonVariantAs ::type ArduinoJson::JsonArraySubscript::as () const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonArraySubscript.hpp line 55
:  (inlined by) ArduinoJson::Internals::JsonVariantAs ::type const ArduinoJson::JsonVariantBase ::as () const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonVariantBase.hpp line 63
:  (inlined by) ArduinoJson::JsonVariantBase ::operator double () const at c:\Users\10049636\OneDrive\Projects\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonVariantBase.hpp line 58
:  (inlined by) setup() at C:\Users\10049636\AppData\Local\Temp\VMBuilds\JSONTest\espressif_lolin32\Debug/JSONTest.cpp line 62
0x400de3e3: loopTask(void*) at c:\Users\10049636\OneDrive\Projects\Arduino\hardware\espressif\esp32\cores\esp32/main.cpp line 15
```

Any ideas to what this might relate?

bblanchon, 2017-10-01:
Hi @vyruz1986,

Can you try to pass `-fno-threadsafe-statics` to the compiler command line?
This should disable `__cxa_guard_acquire`/`__cxa_guard_release`.
Let's see what happens...

Regards,
Benoit

vyruz1986, 2017-10-01:
Hi @bblanchon,

Your suggestion solved both problems, awesome!
I had read this in the [FAQ ](https://bblanchon.github.io/ArduinoJson/faq/esp32/)before I posted the issue, but I believed that problem would manifest itself in compilation errors, not runtime errors. So maybe the FAQ entry could be clarified a bit more?

Thanks for your help!

bblanchon, 2017-10-01:
I suspect there is a bug in the implementation of `__cxa_guard_acquire`/`__cxa_guard_release`.

I think that this minimal program should reproduce the bug:

```c++
const int& getAnswer() {
  static int answer = 42;
  return answer;
}

int main() {
  return getAnswer();
}
```

Can you test to see if it crashes too?

vyruz1986, 2017-10-01:
Not sure what you mean here, I'm on arduino so I don't have a `main()` loop...
I tried calling your `getAnswer()` function in arduino's `loop()` but that didn't cause any crash.

bblanchon, 2017-10-05:
If anyone else has this issue, please leave a comment.

bblanchon, 2017-12-11:
Look like a [Loch Ness Monster Bug](https://blog.codinghorror.com/new-programming-jargon/)...

[image]

closed 2017-12-11 by bblanchon