bblanchon/ArduinoJson #586: I am having problem:

Sovatna, 2017-10-03:


Sovatna, 2017-10-03:
{
  "coord": {
    "lon": -0.13,
    "lat": 51.51
  },
  "weather": [
    {
      "id": 301,
      "main": "Drizzle",
      "description": "drizzle",
      "icon": "09n"
    },
    {
      "id": 701,
      "main": "Mist",
      "description": "mist",
      "icon": "50n"
    },
    {
      "id": 741,
      "main": "Fog",
      "description": "fog",
      "icon": "50n"
    }
  ],
  "base": "stations",
  "main": {
    "temp": 281.87,
    "pressure": 1032,
    "humidity": 100,
    "temp_min": 281.15,
    "temp_max": 283.15
  },
  "visibility": 2900,
  "wind": {
    "speed": 1.5
  },
  "clouds": {
    "all": 90
  },
  "dt": 1483820400,
  "sys": {
    "type": 1,
    "id": 5091,
    "message": 0.0226,
    "country": "GB",
    "sunrise": 1483776245,
    "sunset": 1483805443
  },
  "id": 2643743,
  "name": "London",
  "cod": 200
}

This data will be obtained from  OpenWeatherMap every 10 min. May I parse this data into Json? I do not know how to do that. Please kindly help

bblanchon, 2017-10-05:
Hi @Sovatna,

They are plenty of projects using ArduinoJson for OpenWeatherMap, please make a search on [Google](http://lmgtfy.com/?q=OpenWeatherMap+ArduinoJson).

Alternatively, you can have a look at the projects listed in the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Projects-using-ArduinoJson

The [ArduinoJson Assitant](https://bblanchon.github.io/ArduinoJson/assistant/index.html) has a preset for OpenWeatherMap.

Regards,
Benoit

closed 2017-10-17 by bblanchon

Sovatna, 2017-10-17:
Dear Benoit,

I followed your recommendation, but still it does not work to obtain 5 days
data. May you please precisely guide me to the point?

Best regards,

Sovatna

Sovatna, 2017-10-17:
Dear Benoit, 

It is working right now, but it obtains only one data. I cannot work on prediction the weather. May you please help share idea on how to get 5 days weather forecast?
Best regards, 

Sovatna

bblanchon, 2017-10-17:
Hi @Sovatna,

I'm sorry but the only thing I can do is direct you to the OpenWeatherMap documentation:
https://openweathermap.org/forecast5

Here are the common pitfalls of ArduinoJson:

1. You need to make sure the `JsonBuffer` is big enough to store the complete JSON.
2. You need to make sure that your microcontroller has enough RAM to hold that data.
3. You must not reuse the `JsonBuffer`, instead create a new one each time you parse.

Every information you need is either on the [ArduinoJson website](https://bblanchon.github.io/ArduinoJson/) or on [Google](http://lmgtfy.com/?q=OpenWeatherMap+ArduinoJson).

Sovatna, 2017-10-17:
Hi Benoit,

Thanks you so much for your advise. I will try your recommendation.

Best regards,

Sovatna