bblanchon/ArduinoJson #588: float serialization

alex9x, 2017-10-14:
Hi! Please, help with float serialization.

Function to cut one decimal and part of code:
`
float Round1(float tmp) {
  return ((int)(tmp * 10)) / 10.0;
}

  DynamicJsonBuffer  jsonBuffer(200);
  JsonObject& log = jsonBuffer.createObject();
  tmp_t=24.35141;   // float value from sensor
  log["dht_t"] = Round1(tmp_t);  // cut to one decimal
`
got results in log file:

{"dht_t":24}
{"dht_t":24.1}
{"dht_t":24.2}
{"dht_t":24.**29999**}    - for 24.3
{"dht_t":24.4}
{"dht_t":24.5}
{"dht_t":24.6}
{"dht_t":24.7}
{"dht_t":24.**79999**}     - for 24.8
{"dht_t":24.9}

problem with x.3 and x.8

Any help are welcome!

alex9x, 2017-10-16:
Problems also with 2 decimal digits float values:
`
24.55999        25.1    34      24.75   29.25   24.87   24.9    36      1
24.55999        25.2    33      24.87   36.81   24.80999        24.9    36      1
24.55999        25.2    33      24.87   38.31   24.75   24.9    36      1
24.55999        25.2    33      24.80999        38.61999        24.68   24.9    36
24.5    25.1    33      24.75   38.56   24.62   24.9    36      1
24.5    25.1    33      24.62   38.43   24.55999        24.9    36      1
24.43   25.1    33      24.55999        38.36999        24.5    24.9    36      1
24.43   25.1    33      24.55999        38.31   24.5    24.9    35      1
24.43   25.1    32      24.5    38.25   24.43   24.9    35      1
24.43   25      32      24.5    38.18   24.43   24.9    35      1
24.43   25      32      24.43   38.11999        24.43   24.9    35      1
24.37   25.1    32      24.43   38.11999        24.43   24.9    35      1
24.37   25.1    32      24.37   38.11999        24.43   24.9    35      1
24.37   25      31      24.37   38.18   24.43   24.9    35      1
24.37   25.1    32      24.37   38.11999        24.43   24.9    34      1
24.43   25      31      24.37   38.25   24.37   24.9    34      1
24.43   25      31      24.30999        38.18   24.43   24.9    34      1
24.43   25.1    31      24.30999        38.18   24.43   24.9    34      1
24.43   25.1    31      24.30999        38.18   24.43   24.9    34      1
24.43   25.1    32      24.37   33.86999        24.5    24.9    34      1
24.5    25.1    32      24.37   30.12   24.5    24.9    34      1
24.5    25      33      27.37   28.05999        24.62   24.9    35      1
24.5    25.5    32      27.55999        27      24.80999        25      34      1
24.55999        25.1    32      26.62   26.18   24.80999        25      35      1
24.55999        25.1    32      26      25.68   24.80999        24.9    34      1
24.55999        25.1    32      25.55999        25.37   24.75   25      34      1
24.55999        25.1    32      25.25   25.18   24.68   25      34      1
24.55999        25.2    32      25.05999        25      24.68   25      34      1
24.55999        25.2    32      24.93   24.93   24.62   25      34      1
24.55999        25.1    32      24.80999        24.87   24.62   24.9    35      1
24.55999        25.1    32      24.75   24.80999        24.62   24.9    34      1
24.55999        25.2    32      24.68   24.80999        24.62   24.9    34      1
24      24.79999        27      22.25   23.18   22.18   24.6    29      1
22.80999        23.6    29      20.05999        21.18   19.55999        23.79999
22.18   22.79999        31      19.37   20.43   18.93   23.2    32      1
21.75   22.29999        31      18.93   19.75   18.05999        22.7    31      1
`

bblanchon, 2017-10-17:
Hi @alex9x,

This looks like a bug to me.
I'll try to fix this ASAP.

Thank you very much for reporting this 👍 

Regards,
Benoit

closed 2017-10-17 by bblanchon

bblanchon, 2017-10-17:
Demo: https://wandbox.org/permlink/s0fGZoZTIG5YP7AU

alex9x, 2017-10-17:
Yes, already test it - all ok. Thank you!

`44.9    23.6    34      26.2    24      25      23.9
44.6    23.3    34      26.1    23.8    24.8    23.8
44.5    23.6    34      26      23.7    24.7    24.1
43.9    22.8    35      24.9    23.1    24.5    23.3
43.9    22.4    36      24.1    22.6    24.3    23
44.1    18.4    35      23.7    22.3    24.2    17.8
44.3    21.8    38      23.3    22.1    24.1    22.4
44.6    21.5    39      22.8    21.8    24      22.3
44.7    21.3    39      22.7    21.7    23.9    22
44.7    21.2    40      22.5    21.6    23.8    21.9
44.5    21.1    40      22.3    21.4    23.7    21.8
44.4    21.1    40      22.2    21.3    23.5    21.6
44.6    20.8    40      22.1    21.3    23.4    21.5
44      20.9    41      22      21.3    23.3    21.4
36.6    20.8    60      22.2    21.5    23.3    21.4`

commit 591fe7e9 (2017-10-17): Fixed incorrect rounding for float values (fixes #588)