bblanchon/ArduinoJson #589: Parsing fails after a couple of reuses

Jonsen94, 2017-10-14:
Hi there,
I'm working with ArduinoJson 5.11.1 and trying to deserialize a weather String.
I'm using Visual Studio 2017 with Visual Micro to compile for the esp32.

The DynamicJson Buffer is stored as a class var and reused every time new weather data is loaded from the api. Reusage as follows:
_dynamicJsonBuffer.clear();
somevar = &_dynamicJsonBuffer.parseObject("json");

somevar is also stored as a class var.

After about 6 times calling the method there seems to be a problem parsing the json because somevar only returns 0 if I try to read the temprerature (double) while the json says it should be 15 or so.
(Accessing any value returns the values you get if it doesn't exists)

I know you shouldn't use it as a class var but that doesn't change the fact, that it shouldn't be problematic reusing the buffer.

No sure if I'm doing something else wrong.

bill-orange, 2017-10-15:
I am also parsing a weather string.  Its big the buffer has to be 46000.  I had to switch to an ESP32.  The reason for my post is that I am also failing on the sixth reuse.  Probably a coincidence but I thought I would bring it up.  My only solution so far is to force the ESP32 to restart on parsing failure.  Restart is pretty fast so it is transparent to the user.  Unsatisfying solution.

bblanchon, 2017-10-17:
Hi,

`JsonBuffer`s are not to be reused.
See: https://bblanchon.github.io/ArduinoJson/faq/how-to-reuse-a-jsonbuffer/

Regards,
Benoit

bill-orange, 2017-10-17:
I think that perhaps, I am using the term reuse incorrectly.  What I mean to say is thatin the code below,  after the function  get_forecast is called six times showWeather crashes with "jsonBuffer.parseObject() failed"  No amount of retries helps.  I have to restart.

`//----------------------------------------Get Forecast-----------------------------
void get_forecast() {
  // Open socket to WU server port 80

//------Buffer Size Critical --------
  static char respBuf [buffer_size];

  Serial.print(F("Connecting to "));
  Serial.println(WUNDERGROUND);

  // Use WiFiClient class to create TCP connections
  WiFiClient httpclient;
  const int httpPort = 80;
  if (!httpclient.connect(WUNDERGROUND, httpPort)) {
    Serial.println("connection failed");
    fflush(stdout);
    esp_restart();
    //delay(DELAY_ERROR);
    return;
  }

  // This will send the http request to the server
  Serial.print(WUNDERGROUND_REQ);
  httpclient.print(WUNDERGROUND_REQ);
  delay (50);
  httpclient.flush();
  delay (50);
  Serial.println("debug point - right after flush line 279");
  // Collect http response headers and content from Weather Underground
  // HTTP headers are discarded.
  // The content is formatted in JSON and is left in respBuf.
  long int respLen = 0;
  bool skip_headers = true;
  while (httpclient.connected() || httpclient.available()) {
    if (skip_headers) {
      String aLine = httpclient.readStringUntil('\n');
      //Serial.println(aLine);
      // Blank line denotes end of headers
      if (aLine.length() <= 1) {
        skip_headers = false;
      }
    }
    else {
      long int bytesIn;
      bytesIn = httpclient.read((uint8_t *)&respBuf[respLen], sizeof(respBuf) - respLen);
      Serial.print(F("bytesIn ")); Serial.println(bytesIn);
      delay(syncDelay);
      if (bytesIn > 0) {
        respLen += bytesIn;
        if (respLen > sizeof(respBuf)) respLen = sizeof(respBuf);
      }
      else if (bytesIn <= 0) {
        Serial.print(F("read error "));
        Serial.println(bytesIn);
        counterStop = 3 * counterStop++;
        esp_restart();
      }

    }
    delay(1);
  }
  Serial.println("debug point - headers collected  - line 312");
  Serial.println();
  httpclient.stop();

  if (respLen >= sizeof(respBuf)) {
    //Serial.print(F("respBuf overflow "));
    //Serial.println(respLen);
    Serial.println("failed on buffer overflow");
    esp_restart();
    //delay(DELAY_ERROR);
    return;
  }
  // Terminate the C string
  respBuf[respLen++] = '\0';
  //Serial.print(F("respLen "));
  //Serial.println(respLen);
  //Serial.println(respBuf);
  if (showWeather(respBuf)) {
    Serial.println("debug point - showweather true 329");
    Serial.print("Assigned size of RespBuf:    "); Serial.println(sizeof(respBuf));
    Serial.print("Actial size of RespBuf used: "); Serial.println(respLen);
    delay(DELAY_NORMAL);
  }
  else {
    Serial.println("failed on showWeather - rebooting");
    esp_restart();
    //counterStop++;
    //delay(DELAY_ERROR);
  }
}

//------------------------------------------Show Weather---------------------------
bool showWeather(char *json)
{
  Serial.print("Inside the belly of the  - line 202 ");
  //StaticJsonBuffer<73072> jsonBuffer;
  DynamicJsonBuffer jsonBuffer;
  // Skip characters until first '{' found
  // Ignore chunked length, if present
  char *jsonstart = strchr(json, '{');
  // Serial.print("jsonstart "); Serial.println(jsonstart);
  if (jsonstart == NULL) {
    Serial.println("JSON data missing");
    //fflush(stdout);
    esp_restart();
    return false;
  }
  json = jsonstart;

  // Parse JSON
  JsonObject& root = jsonBuffer.parseObject(json);
  Serial.println("debug point - root established debug point line 219");
  Serial.println (" ..");
  if (!root.success()) {
    Serial.println("jsonBuffer.parseObject() failed");
    fflush(stdout);
    esp_restart();
    jsonBuffer.clear();
    return false;
  }

  // Extract weather info from parsed JSON

  for (int hour_forecast = 0; hour_forecast < max_forecast; hour_forecast++) {

    const char *observation_time = root["hourly_forecast"][hour_forecast]["FCTTIME"]["pretty"];
    observation_time_st = reinterpret_cast<const char*>(observation_time);
    Serial.println(observation_time_st);
    temp_f = root["hourly_forecast"][hour_forecast]["temp"]["english"];
    //Serial.print(temp_f, 1); Serial.print" F, ");
    const char *humi = root["hourly_forecast"][hour_forecast]["humidity"];
    humi_st = reinterpret_cast<const char*>(humi);
    //Serial.print(humi_st);   Serial.println" RH");
    const char *weather = root["hourly_forecast"][hour_forecast]["condition"];
    weather_st = reinterpret_cast<const char*>(weather);
    //Serial.println(weather_st);
    const char *pressure_mb = root["hourly_forecast"][hour_forecast]["mslp"]["english"];
    pressure_mb_st = reinterpret_cast<const char*>(pressure_mb);
    //Serial.print(pressure_mb_st); Serial.println(" PSI ");
    unix_time = root["hourly_forecast"][hour_forecast]["FCTTIME"]["epoch"];
    //Serial.print(F(" Unix Time "));Serial.println(unix_time);
    const char *clock_time = root["hourly_forecast"][hour_forecast]["FCTTIME"]["civil"];
    clock_time_st = reinterpret_cast<const char*>(clock_time);
    //Serial.print(" Civil Time ");Serial.println(clock_time);
    Serial.println(" ..");
  }
  jsonBuffer.clear();
  return true;
}`

bill-orange, 2017-10-19:
Thank you for the fine code and for the advise.

In the future I will keep the code fragments short or use a onedrive link.  Hopefully, I will get this worked out and I will not need to post any code at all!

Regarding your second suggestion, I did not know that you could specify a buffer size with DynamicJsonBuffer in a manner similar to the static buffer.  It will be really large.  I ran the Assistant and my buffer will be:
 DynamicJsonBuffer jsonBuffer (73072);  Wow, this is a very big buffer, even for an ESP32.  Wunderground's hourlies JSON is enormous.

Regarding your first point, static char respBuf [46200] is big.  It is not the same every day and I am hoping it does not get any bigger.  Do you have any suggestions on how to improve memory usage of respBuf?

 char * respBuf; 

causes a crash right away.  So does char respBuf [46200]; (omitting static).  I don't have a very clear understating of variable scope in C or in memory allocation creation or destruction. Among ESP / Arduino programmers I fear that I am not alone.    The last class I took in C was 1994.  That doesn't help.  

Any additional guidance would be appreciated.

bblanchon, 2017-10-21:
Hi @bill-orange,

I'll try to give you some insight on what happening:

```c++
// This causes a crash because no memory is allocated and the pointer is dangling
char * respBuf;
```

```c++
// This cause a crash because it's allocated on the stack, which is probably limited to 4KB
char respBuf [46200];
```

```c++
// This doesn't crash because it's allocated in the bss segment.
// However, the memory is kept for the whole runtime, which is a huge waste.
static char respBuf [46200];
```

The best solution is to avoid this buffer and let the `jsonBuffer` absorb the input.
It's as simple as calling `jsonBuffer.parseObject(httpclient)`, just as it's done in [JsonHttpClient.ino](https://bblanchon.github.io/ArduinoJson/example/http-client/).

I've no idea of what's happening with the IDE, but the latest version is [5.11.2](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.11.2)

bill-orange, 2017-10-22:
Well, I attempted to use the example's approach with Wunderground with little result.  I have read JSON parsing failed! a dozen times this morning.

The code looks pretty straight forward but  I can not discover the reason for the failure.

I validated the Wundergound using the JSON validator web site you mentioned in another post.  It works fine.  I used JSON Assistant to get the site-specific code.

Could I still have a memory issue?  The Jason buffer is over 76000 according to JSON Assistant.

To avoid post a lot of code, I will just provide a link with my API key redacted.

https://1drv.ms/f/s!AmXKqAwyCrbxh9dKlz-41o5BMRvxaQ

Thank you.  I feel bad asking for help with this.  It should not be your job to teach me this stuff.

bblanchon, 2017-10-24:
@bill-orange Sure 76KB is massive, but the ESP32 is supposed to be able to handle that.

To eliminate the memory problem,  you can try to read to hardcode the JSON in a `const char*` (just like the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) does) and see if parsing succeeds.
If it succeeds, then it's probably a problem in extracting the JSON from the HTTP response.
If it doesn't, then it's probably a memory issue.

Another way to diagnose a memory problem is to do a `malloc(76000)` and see if it returns `NULL`.
I don't have an ESP32 so I cannot check.

Anyway, I looked at your program and it looks OK.

bill-orange, 2017-10-27:
Looks like a memory issue.  ArduinoJson Assistant's const char* string will not parse.

I think we can close out this issue.  If ESP32 can not do it, there is not much to be done on your side of things.

bblanchon, 2017-10-28:
Have a look at [json-streaming-parser](https://github.com/squix78/json-streaming-parser).
This JSON library takes a radically different approach that allows to read huge JSON document without loading them entirely in memory.

closed 2017-10-28 by bblanchon