bblanchon/ArduinoJson #591: Compile errors on Adafruit Feather M0: invalid conversion from 'ArduinoJson::Internals::JsonVariantAs<char*>::type

john3exonets, 2017-10-21:
Getting a number of errors when trying to use the library on a Adafruit Feather M0 (ATSAMD21G18 ARM Cortex M0 processor):

```
In file included from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:10:0,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:16,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:15,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:10,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:10,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:10,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\John\Documents\Arduino\HouseDisplay_v0_1\HouseDisplay_v0_1.ino:7:

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp: In instantiation of 'ArduinoJson::JsonVariantCasts<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const int&>]':

C:\Users\John\Documents\Arduino\HouseDisplay_v0_1\HouseDisplay_v0_1.ino:132:27:   required from here

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp:52:35: error: invalid conversion from 'ArduinoJson::Internals::JsonVariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]

     return impl()->template as<T>();

                                   ^

In file included from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:12:0,

                 from C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\John\Documents\Arduino\HouseDisplay_v0_1\HouseDisplay_v0_1.ino:7:

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator ArduinoJson::JsonObject::findKey(TStringRef) [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator = ArduinoJson::Internals::ListIterator<ArduinoJson::JsonPair>]':

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:305:66:   required from 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator ArduinoJson::JsonObject::findKey(TStringRef) const [with TStringRef = const int&; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::const_iterator = ArduinoJson::Internals::ListConstIterator<ArduinoJson::JsonPair>]'

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:311:48:   required from 'typename ArduinoJson::Internals::JsonVariantAs<TValue>::type ArduinoJson::JsonObject::get_impl(TStringRef) const [with TStringRef = const int&; TValue = char*; typename ArduinoJson::Internals::JsonVariantAs<TValue>::type = const char*]'

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:172:48:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type ArduinoJson::JsonObject::get(const TString&) const [with TValue = char*; TString = int; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), typename ArduinoJson::Internals::JsonVariantAs<T>::type>::type = const char*]'

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObjectSubscript.hpp:64:36:   required from 'typename ArduinoJson::Internals::JsonVariantAs<T>::type ArduinoJson::JsonObjectSubscript<TKey>::as() const [with TValue = char*; TStringRef = const int&; typename ArduinoJson::Internals::JsonVariantAs<T>::type = const char*]'

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp:52:35:   required from 'ArduinoJson::JsonVariantCasts<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const int&>]'

C:\Users\John\Documents\Arduino\HouseDisplay_v0_1\HouseDisplay_v0_1.ino:132:27:   required from here

C:\Users\John\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp:299:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const int&, void>'

       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;

                                                                   ^

exit status 1
Error compiling for board Adafruit Feather M0.
```
Since none of this actually references my code, I am assuming its a issue with the library and/or my environment:

Windows 10
Arduino IDE 1.8.3
[Adafruit Feather M0](https://www.adafruit.com/product/3010)
ArduinoJson pull was from github yesterday (11/20/17).

This issue doesn't seem to match any of the Known Issues in the FAQ from what I could see.

bblanchon, 2017-10-21:
Hi John,

It looks like the code at line 132 of `HouseDisplay_v0_1.ino` is using an `int` to index a `JsonObject`.

Indeed `JsonObject` must be indexed by a string, see [Can I access to object member by its index, instead of its key?](https://bblanchon.github.io/ArduinoJson/faq/can-i-access-to-object-member-by-its-index-instead-of-its-key/)

Regards,
Benoit

john3exonets, 2017-10-24:
Ok....Didn't see that. Sorry.

closed 2017-10-24 by john3exonets

bblanchon, 2017-11-09:
No problem John, these error messages are very hard to figure out.
In a sense, it more my fault than yours.

To help other users, I added the error message in the FAQ:
http://arduinojson.org/faq/error-equals-is-not-a-member-of/