bblanchon/ArduinoJson #592: Compile errors on NodeMcu

joingood, 2017-10-22:
Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

WARNING: Category '' in library ArduinoJson is not valid. Setting to 'Uncategorized'
In file included from C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/../include/../include/ArduinoJson/Internals/List.hpp:12:0,

                 from C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/../include/../include/ArduinoJson/JsonArray.hpp:10,

                 from C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/../include/ArduinoJson.h:7,

                 from C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:13,

                 from D:\node-core\System\Arduino\Main.lnk\Main.lnk.ino:9:

C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/../include/../include/ArduinoJson/Internals/PlacementNew.hpp: In function 'void* operator new(size_t, void*)':

C:\Users\George\Documents\Arduino\libraries\ArduinoJson\src/../include/../include/ArduinoJson/Internals/PlacementNew.hpp:13:14: error: redefinition of 'void* operator new(size_t, void*)'

 inline void *operator new(size_t, void *p) throw() { return p; }

              ^

In file included from c:\users\george\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:54:0,

                 from C:\Users\George\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27,

                 from C:\Users\George\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Users\George\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Program Files (x86)\Arduino\libraries\System/sensorUnit.h:2,

                 from D:\node-core\System\Arduino\Main.lnk\Main.lnk.ino:7:

c:\users\george\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\new:109:14: error: 'void* operator new(std::size_t, void*)' previously defined here

 inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT

              ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bblanchon, 2017-10-23:
Hi @joingood,

Please check the version or ArduinoJson.
It seems that an old one is installed.

Regards,
Benoit

joingood, 2017-10-23:
I'm installing it from arduino manager

bblanchon, 2017-10-24:
@joingood Can you try to remove the library (make sure none of the files remain) and install it again?

closed 2017-11-05 by bblanchon