bblanchon/ArduinoJson #593: JSON gets corrupted when reading from file and modifying it

Anton-V-K, 2017-10-24:
When JSON is read from file (SPIFFS is used) into `DynamicJsonBuffer` and some values are modified, it gets corrupted.

Sample code to illustrate the idea:
```
#include <ArduinoJson.h>
#include <FS.h>

void setup() {

    Serial.begin(115200); // Init console
  
    SPIFFS.begin();
  
    DynamicJsonBuffer jsonBuffer;
    
    JsonObject* root = NULL;
    
    if (File file = SPIFFS.open("Test.json", "r"))
    {
      if (const size_t size = file.size())
      {
        std::unique_ptr<char[]> buf (new char[size]);
        file.readBytes(buf.get(), size);
        root = &jsonBuffer.parseObject(buf.get());
      }
      file.close();
    }
    if (File file = SPIFFS.open("Test.json", "w"))
    {
      if (!root)
        root = &jsonBuffer.createObject();
        
      (*root)["DEVICE_ID"] = 1;
      (*root)["millis"]    = millis();
      
      root->prettyPrintTo(Serial); // DEBUG
      root->printTo(file);
      file.close();
      Serial.println("\nSaved!");
    }
    
}

void loop() {
  // not used in this example
}
```
The output when ran 2nd time:
```
{
  "\t": 1,
  "millis": 320,
  "DEVICE_ID": 1
}
Saved!
```
Expected output when ran 2nd time:
```
{
  "DEVICE_ID": 1
  "millis": 320,
}
```
Target platform: ESP8266 (WeMos D1 R2)
IDE: Arduino IDE 1.6.13
LIB version: 5.11.2

bblanchon, 2017-10-26:
Hi @Anton-V-K

As I understand, you found the solution by yourself. 👍 

Since the input of `parseObject()` was a `char*`, the parser assumed it to be persistent; and therefore stored a pointer to a temporary buffer.

Now, as you found, you can avoid this problem by just calling `parseObject(file)`.
In that case, the parser knows that the input is volatile and will make the copy in the `JsonBuffer`.

Is there anything else I can do?

Regards,
Benoit

Anton-V-K, 2017-11-04:
Well, I guess the issue may be closed. Though I think the interface of `DynamicJsonBuffer` is somewhat confusing with overloaded method `parseObject` - the specifics of handling const/non-const buffer is easily overlooked...

bblanchon, 2017-11-05:
Agreed, @Anton-V-K.
Unfortunately, this came organically with the evolution of the library.
Do you have any suggestion for futures versions?

closed 2017-11-05 by bblanchon