bblanchon/ArduinoJson #594: How can i call json object multiple times in a program?

parmarravi, 2017-10-24:
**_I have this json code that i want to parse in switch case continuously but inorder to do that i need
 jsonObject root. and i cant declare it outside as its out of scope already .whats the best way to
 make it work._**
**Once i extract the data from the json i need that to be pass in various functions with different number of  parameters,.** For ex: f1(data0_f1_Message,data0_f1_Length,data0_f1_font);

```
void loop() {
  // put your main code here, to run repeatedly:
  /*{
     //this is what my json structure looks like
    "animate": [ 1,2,5,2,1],
    "data":[{
    "f1":{ "Message": "hello world",
        "Length": 5,
        "font":"Arial"},
    "f2":{ "Message": "hello world",
        "Length": 5,
        "font":"Arial"},
    "f5":{ "Message": "hello world",
        "Length": 5,
        "font":"Arial"},
    "f2":{ "Message": "hello world",
        "Length": 5,
        "font":"Arial"},
    "f1":{ "Message": "hello world",
        "Length": 5,
        "font":"Arial"}
    }]
    }*/
  if (Serial.available() > 0) {
    Jsondata = Serial.readStringUntil('\n');
    Serial.println(Jsondata);
    Jsondata.trim();
    JsonObject& root = jsonBuffer.parseObject(Jsondata);
    JsonArray& animates = root["animate"];

    for (int i = 0; i < animates.size(); i++) {
      animatesJson[i] = animates[i];
    }

    if (Jsondata == "OK") {
      animate = true;
      Serial.println("TRUE");
    }
    if (Jsondata == "DEFAULT") {
      animate = false;
      Serial.println("default");
    }
  }
  if (animate) {
    for (int i = 0; i < (sizeof(animatesJson) / sizeof(int)); i++) {
       /* HOW can i call root here, so that i can get the data from array in loop?
             JsonObject& data0 = root["data"][0];
      switch (animatesJson[i]) {
        case 1:
          //do scrollSingleMessage
            JsonObject& data0_f1 = data0["f1"];
            const char* data0_f1_Message = data0_f1["Message"]; // "hello world"
            int data0_f1_Length = data0_f1["Length"]; // 5
            const char* data0_f1_font = data0_f1["font"]; // "Arial"
            .....f1(data0_f1_Message,data0_f1_Length,data0_f1_font);........
          */
          f1();
          break;
        case 2:
          //do staticSingleMessage
          /*
             JsonObject& data0_f2 = data0["f2"];
            const char* data0_f2_Message = data0_f2["Message"]; // "hello world"
            int data0_f2_Length = data0_f2["Length"]; // 5
            const char* data0_f2_font = data0_f2["font"]; // "Arial"
             .....f2(data0_f2_Message,data0_f2_Length,data0_f2_font);........
          */
          f2();
          break;
        case 3:
          //do splitDualMessage
          f3();
          break;
        case 4:
          //do BLinkSingleMessage
          f4();
          break;
        case 5:
          //do BLinkDoubleMessage
          f5();
          break;
        case 6:
          //do temp
          f6();
          break;
      }
    }
  }
  else
  {
    DM();//DEFAULT  MODE..
  }
}

```

parmarravi, 2017-10-25:
I am now able to receive data but there is still some issues 
When passing  **"{"animate": [ 1,2,5,2,1]}"** i am able to receive data but  with more complex array for example **" {"animate": [ 1,2,5,2,1],
     "data":[{"f1":{ "Message": "hello world", "Length": 5, "font":"Arial"}, "f2":{ "Message": "hello world", "Length": 5,"font":"Arial"},
     "f5":{ "Message": "hello world", "Length": 5,"font":"Arial"},"f2":{ "Message": "hello world", "Length": 5,"font":"Arial"},
     "f1":{ "Message": "hello world", "Length": 5,"font":"Arial"}}]}"**
I am not able to get the complete array _.please help me out .._
```
#include <ArduinoJson.h>
boolean animate = false;
int animatesJson[10];
String  Jsondata;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
  /*
     {"animate": [ 1,2,5,2,1],
     "data":[{"f1":{ "Message": "hello world", "Length": 5, "font":"Arial"}, "f2":{ "Message": "hello world", "Length": 5,"font":"Arial"},
     "f5":{ "Message": "hello world", "Length": 5,"font":"Arial"},"f2":{ "Message": "hello world", "Length": 5,"font":"Arial"},
     "f1":{ "Message": "hello world", "Length": 5,"font":"Arial"}}]}
  */
  /* {
    "Settings":[{ "default": true}]}
  */
  if (Serial.available() > 0) {
    Jsondata = Serial.readStringUntil('\n');
    Serial.println(Jsondata);
    JsonObject& roots = getJsonroot(Jsondata);
    JsonArray& animates = roots["animate"];
    bool Settings0_default = roots["Settings"][0]["default"]; // true
    for (int i = 0; i < animates.size(); i++) {
      animatesJson[i] = animates[i];
      Serial.println(animatesJson[i]);
      animate = true;
    }
    if (Settings0_default) {
      animate = false;
      Serial.println("default");
    }
  }
  if (animate) {
    for (int i = 0; i < (sizeof(animatesJson) / sizeof(int)); i++) {
      if (animatesJson[i] == 1)
      {
        JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f1 = data0["f1"];
        const char* data0_f1_Message = data0_f1["Message"]; // "hello world"
        int data0_f1_Length = data0_f1["Length"]; // 5
        const char* data0_f1_font = data0_f1["font"]; // "Arial"
        f1();
      }
      if (animatesJson[i] == 2)
      { JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f2 = data0["f2"];
        const char* data0_f2_Message = data0_f2["Message"]; // "hello world"
        int data0_f2_Length = data0_f2["Length"]; // 5
        const char* data0_f2_font = data0_f2["font"]; // "Arial"
        f2();
      }
      if (animatesJson[i] == 3)
      { JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f3 = data0["f3"];
        const char* data0_f3_Message = data0_f3["Message"]; // "hello world"
        int data0_f3_Length = data0_f3["Length"]; // 5
        const char* data0_f3_font = data0_f3["font"]; // "Arial"
        f3();
      }
      if (animatesJson[i] == 4)
      { JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f4 = data0["f4"];
        const char* data0_f4_Message = data0_f4["Message"]; // "hello world"
        int data0_f4_Length = data0_f4["Length"]; // 5
        const char* data0_f4_font = data0_f4["font"]; // "Arial"
        f4();
      }
      if (animatesJson[i] == 5)
      { JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f5 = data0["f5"];
        const char* data0_f5_Message = data0_f5["Message"]; // "hello world"
        int data0_f5_Length = data0_f5["Length"]; // 5
        const char* data0_f5_font = data0_f5["font"]; // "Arial"
        f5();
      }
      if (animatesJson[i] == 6)
      { JsonObject& rootsd = getJsonroot(Jsondata);
        JsonObject& data0 = rootsd["data"][0];
        delay(1000);
        JsonObject& data0_f6 = data0["f6"];
        const char* data0_f6_Message = data0_f6["Message"]; // "hello world"
        int data0_f6_Length = data0_f6["Length"]; // 5
        const char* data0_f6_font = data0_f6["font"]; // "Arial"
        f6();
      }
    }
  }
  else
  {
    DM();//DEFAULT  MODE..
    delay(1000);
  }
}

JsonObject& getJsonroot(String jsond) {
  StaticJsonBuffer<1000> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(jsond);
  if (animate == false) {
    if (root.success()) {
      Serial.println("parseObject() succeeded");
    } else {
      Serial.println("parseObject() failed!");
    }
  }
  return root;
}
//functionss
void f1() {
  Serial.println("f1");
  delay(1000);
}
void f2() {
  Serial.println("f2");
  delay(1000);
}
void f3() {
  Serial.println("f3");
  delay(1000);
}
void f4() {
  Serial.println("f4");
  delay(1000);
}
void f5() {
  Serial.println("f5");
  delay(1000);
}
void f6() {
  Serial.println("f6");
  delay(1000);
}
void DM() {
  Serial.println("default");
}
```

bblanchon, 2017-10-26:
Hi @parmarravi,

It seems that the `JsonBuffer` is too small.
Please read:

* [Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)
* [How to determine the buffer size?](https://bblanchon.github.io/ArduinoJson/faq/how-to-determine-the-buffer-size/)

Regards,
Benoit

parmarravi, 2017-10-30:
still no luck.Do you think memory of arduino is falling short?right now only 20 % of memory is been use.Or my code is running in a loop and its taking time to process json in short time?

bblanchon, 2017-10-31:
I don't know.
What kind of Arduino is it?
What is the size computed by the assistant?

closed 2017-11-09 by bblanchon