bblanchon/ArduinoJson #595: Question on automatic population

mnelson920, 2017-10-27:
Hi,

This is not a problem but a question on how something can be down with your library. 

I want to create a JSON string that holds information of multiple sensors, so for example:
`{
  "Sensor": [{
    "Unique ID": "Some ID",
    "Type": "Temp",
    "Sensor Output": 25
  },
  {
    "Unique ID": "Some ID",
    "Type": "Temp",
    "Sensor Output": 30
  }],
"Time": "some time"
}`
Where each sensor connected to the system will be under the Sensor tag. My question is how can I use your library to create a JSON string with multiple sensor under the Sensor tag, where the code will automatically adjust the JSON string depending on how many sensors are attached? There is no predetermine amount of sensors. Could be one it also could be 20. 

I am programming this for a Particle Electron. It uses most if not all the Arduino Libraries. 
Any help you can provide would be great. Thanks

bblanchon, 2017-10-29:
Hi @mnelson920,

ArduinoJson seems to be a good fit to your project, why don't you give it a try?
Start from the [examples](https://bblanchon.github.io/ArduinoJson/example/) or from the [ArduinoJson assistant](https://bblanchon.github.io/ArduinoJson/assistant/).

Regards,
Benoit

mnelson920, 2017-10-31:
Thank you for the response @bblanchon. I have tried it but I can't quite figure out how to add nestedObjects in a nested array for an undetermined amount of sensors.

bblanchon, 2017-11-01:
I don't understand.
Is it a question related to ArduinoJson or Arduino programming in general?

mnelson920, 2017-11-01:
Its related to ArduinoJson. How can use this library to do this:

`{
  "Sensor": [{
      "ROM": "dljfl",
      "Type": "DS18B20",
      "Celsius": 25.45
  },
  {
    "ROM": "456464564",
    "Type": "DS18B20",
    "Celsius": 30.45
  }],
Theres other stuff but I got that figured out`

The AdruinoJson assistant says to create nestedObject (ex Sensor_0, Sensor_1) for each sensor but you need to have a variable for that. How can use your library to create nestedObjects for an unknown amount of sensors? Is there a way to increment them? Is it possible to use a for... loop to create them? Pretty much I have a temp monitor but the monitor can have any amount of sensors attach. Each sensor will have same type of data like you see above. I want to adjust the JSON string accordingly. All I have seen so far from your library is everything has to be known ahead of time to create the proper JSON string. I want to know if there is a way to use your library dynamically.

bblanchon, 2017-11-05:
@mnelson920, Yes you can call `JsonArray::createNestedObject()` in a loop.
Can you share a short code sample so we can see if there is something wrong?

mnelson920, 2017-11-07:
@bblanchon, Sorry for the delay I didn't see your reply.

Here is my code so far:

`void createJSON(){
    FuelGauge fuel;

    String senType, senROM, currentTime;
   // unsigned short int index = 0;

    // Allocate the size of the JSON buffer
    StaticJsonBuffer<1000> jsonBuffer; // 1KB jsonBuffer;

    // Create Json Object
    JsonObject& root = jsonBuffer.createObject();
    // Create Nested Array "Sensor"
    JsonArray& sensorJSON = root.createNestedArray("Sensors");
    
    JsonObject& GPS = root.createNestedObject("GPS");
    
    GPS["Latitude"] = 0;
    GPS["Longitude"] = 0;
    
    root["Voltage"] = fuel.getVCell();
    root["Time"] = currentTime;
    
    unsigned char index = '0';/ Read the next available 1-Wire temp sensor
    while(!sensor.searchDone()){
        Serial.printf("Im in the loop");
        if (sensor.read()){
            JsonObject& sensorData = sensorJSON.createNestedObject();
            sensorType(senType);
            sensorROM(senROM);
            getCurrentTime(currentTime);
            sensorData["Unique ID"] = senROM;
            sensorData["Type"] = senType; 
            sensorData["Celsius"] = sensor.celsius();
            index++;
        }
    }
    
    root.prettyPrintTo(Serial);
    delay(2000);
}`

How can I increment ` JsonObject& sensorData = sensorJSON.createNestedObject();` to move to the next Object to collect the sensor information

mnelson920, 2017-11-07:
Disregard that `index++;` I forgot to remove it before I pasted it in.

mnelson920, 2017-11-07:
I think I got it now. Pretty much I did the exact same thing but I change this line 
`JsonObject& sensorData = sensorJSON.createNestedObject();`
to:
`JsonObject &sensorData = sensorJSON.createNestedObject();`

Not sure exactly why this changed work. But I seen it on early question you answered (#165).

bblanchon, 2017-11-08:
Hi @mnelson920,

The location of the ampersand (`&`) doesn't affect the compiled code.
Your code looks fine.
What is the problem exactly?

* If the code crash, you probably have a stack overflow, try a DynamicJsonBuffer
* If the JSON is incomplete, try to increase the size of the buffer

See also:

* [ArduinoJson Assistant](https://arduinojson.org/assistant/) to compute the size of the buffer
* [Why does my device crash or reboot?](https://arduinojson.org/faq/why-does-my-device-crash-or-reboot/)
* [FAQ; Why some parts are missing?](https://arduinojson.org/faq/why-some-parts-are-missing/)

mnelson920, 2017-11-08:
My problem was each new sensorData object would create another Sensor array. But after looking at a question you answered previously I figured out what the problem was. 

I didn't think the location of the '&' should matter either but when I moved it, The JSON string started to output like how I wanted. In the original location it would just output the JSON string with just the last sensor it seen and its corresponding data. But anyway sorry for taking up so much of your time. I appreciate the help.

closed 2017-11-08 by mnelson920