bblanchon/ArduinoJson #597: latest jsonarduino - fails to output json for some keys  despite 500 bytes of free memory.

geetee24, 2017-10-31:
tried both static and dynamic.    is there any debugging options in your library to tell me why your code is not outputting its data 1/2 way through the objecT?

geetee24, 2017-10-31:
more info.  with 800 bytes free ram at time of failure, this is output

{"bSuccess":true,"message":"","millis":43357,"reqNum":2,"uiControl":{"uiRD1":7,"uiPCC1":2000,null:}}

not the errror in format at end:   null:

The key is null and the value is missing........

geetee24, 2017-10-31:
here is pseudo coded parts:

	DynamicJsonBuffer jsonBuffer;

	JsonObject& root = jsonBuffer.createObject ();

	unsigned long ulRet = millis ();

	root [ F("bSuc") ] = bSuccess;
	root [ F("msg") ] = pszMessage;
	root [ F("ulMillis") ] = ulRet;
	root [ F("uiReqNum") ] = uiRequestCounter;

		
		JsonObject& nestedObjUiControl = jsonBuffer.createObject();

		nestedObjUiControl [ F("foo1") ] = 5;
		nestedObjUiControl [ F("foo2") ] = 1000;

		root [ F("objUiControl") ] = nestedObjUiControl;

		JsonObject& nestedObjStatus = jsonBuffer.createObject();

		nestedObjStatus [ F("bAZoneIsOn") ] = isAnyValveTurnedOn ();

		//---------------------------------------

		JsonArray& arrayZoneStatus = jsonBuffer.createArray();

		for ( unsigned int uiDisplayAt0 = 0; uiDisplayAt0 < 9; uiDisplayAt0++ ) {
			// each zone
			// return in order of display

			for ( unsigned int uiAt = 1; uiAt <= 9; uiAt++ ) {

				if ( uiDisplayAt0 == arrayFooIndex0 [ uiAt - 1 ] ) {
					// match

					JsonObject& forNestedZoneStatus = jsonBuffer.createObject();

					forNestedZoneStatus [ F("uiZoneIndex1") ] = uiAt;
					forNestedZoneStatus [ F("name") ] = "foo argh";
					forNestedZoneStatus [ F("bOn") ] = isValveOnN ( uiAt );

					forNestedZoneStatus [ F("bEnabled") ] = isFoo ( uiAt );
					forNestedZoneStatus [ F("index") ] = uiDisplayAt0;

					arrayZoneStatus.add ( forNestedZoneStatus );
					
					break;
				};
			}; // for
		}; // for

		nestedObjStatus [ F("objZs") ] = arrayZoneStatus;

		root [ F("objStatus") ] = nestedObjStatus;
	};
	root.printTo ( client );

bblanchon, 2017-11-01:
Hi @geetee24,

I believe the problem comes from the Flash strings.
Indeed, when you use a Flash string to add an element in a `JsonObject`, a copy of the string is made in the `JsonBuffer`.
This is usually not a problem, but in your case, it happens in a loop.
Your program will probably work fine if you remove the `F()` macros in the loop.

Now, is it a good idea for ArduinoJson to copy the content of Flash string in RAM is a good question.
Up till today, one year after introducing this feature, this is the first time someone opens an issue on that topic, so we need to see if more users are impacted, before changing the behavior.
The decision to make a copy of Flash string is motivated by:

1. Keeping the code smaller when Flash strings are not used.
    You don't want to link your program to `strcmp_P()` if you don't use Flash strings.
2. Keeping the `JsonPair` key type to `const char*`.
    It is used when you iterate over the key-value pairs of an object. Changing the type would break a lot of code.
3. Being able to call `myObject["key"].as<char*>()` regardless of how the value was inserted.

So, I'm not saying the behavior is good and should not be changed. Instead, I'm saying that I need to think it through to make sure we go in the right direction.
If you confirm that this is a real concern for you, I'll create a new issue as a reminder.

By the way, I found a few unrelated things that could be improved in your program:

1. It is strongly recommended to provide a size to the constructor of `DynamicJsonBuffer`, so that only one allocation is made.
2. If you're running on a small microcontroller like an ATmega, it is strongly recommended to use a `StaticJsonBuffer` instead.
3. You can use [JsonObject::createNestedObject()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/createnestedobject/) and [JsonObject::createNestedArray()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/createnestedarray/) to simplify your program.

geetee24, 2017-11-01:
Many thanks.  But the same issue (bug) is present when I dont use flash strings as keys, use dynamic with a loarge buffer in the constructor, use a static buffer.......

To me the issue is that the library offers NO clue as to why it is failing.  Can you enhance the lib to explain why it is failing?

bblanchon, 2017-11-01:
You can check the return values of `JsonObject::set()` and `JsonArrary::add()`, they tell if the insertion was successful or not.
In any case, this is failing because there was not enough space in the `JsonBuffer`.

Are you sure you have enough RAM?
What does the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) say?

bblanchon, 2017-11-10:
@geetee24 any progress?

geetee24, 2017-11-10:
I abandoned the library due to the bug.

bblanchon, 2017-11-13:
I'm sorry to hear that 😢 
I hope to see you back someday...

closed 2017-11-13 by bblanchon