bblanchon/ArduinoJson #598: Arduino: parser replaces characters with "null values" and fails

desteaquino, 2017-11-04:
First of all thankyou for your code.
Second, please excuse my poor english.

Im programming an arduino Yun using your Library. My code is as follows:

```
char json[] = "{\"Blinds\":[{\"Id\":1,\"Name\":\"VERTICAL_1\",\"PRPin\":1,\"ARPin\":2,\"TTOpen\":1,\"TTClose\":2},{\"Id\":2,\"Name\":\"VERTICAL_2\",\"PRPin\":3,\"ARPin\":4,\"TTOpen\":1,\"TTClose\":2},{\"Id\":3,\"Name\":\"VERTICAL_3\",\"PRPin\":5,\"ARPin\":6,\"TTOpen\":1,\"TTClose\":2},{\"Id\":4,\"Name\":\"VERTICAL_4\",\"PRPin\":7,\"ARPin\":8,\"TTOpen\":1,\"TTClose\":2},{\"Id\":5,\"Name\":\"VERTICAL_5\",\"PRPin\":9,\"ARPin\":10,\"TTOpen\":1,\"TTClose\":2},{\"Id\":6,\"Name\":\"VERTICAL_6\",\"PRPin\":11,\"ARPin\":12,\"TTOpen\":1,\"TTClose\":2}]}";

DynamicJsonBuffer jsonBuffer(1800);
//at` this point printing json begins as expected: "{"Blinds":[{"Id":1,"Name":"VERTICAL_1",..."
JsonObject& root = jsonBuffer.parseObject(json);
// but when json is parsed, printing json shows: "Blinds Id 1 "Id":1,"Name":"VERTICAL_1",..."
//after parsing the first " of "Blinds" dissapears, the "{"also desapeared... The rest of the array seems ok.

if (!root.success()) {
  blindsQty = -3; //-3 means parsing error, the 
}
else
   {
```
I have tested the code using The wandbox link you added and works ok
I also have tested the json format using your assistant and everything seems also ok.

I dont know if i am doing something wrong, could you please help me?

Thanks in Advance.

Juanma

bblanchon, 2017-11-05:
Hi @desteaquino,

There are two questions here:

1. Why the input is modified? => please see FAQ
2. Why parsing fails? => please read bellow

As far as I know, the ATmega32U4 only has 2.5KB or RAM.
So 1800 bytes of heap memory is probably too much.

Try to switch to a `StaticJsonBuffer` and shrink it to the maximum.
Use the expression given by the Assistant, but don't include the "Additional bytes for input duplication" as you are in the zero-copy mode.

```c++
const size_t SIZE = JSON_ARRAY_SIZE(6) + JSON_OBJECT_SIZE(1) + 6*JSON_OBJECT_SIZE(6);
StaticJsonBuffer<SIZE> jsonBuffer;

JsonObject& root = jsonBuffer.parseObject(json);
```

See:

* [Why is the input modified?](https://bblanchon.github.io/ArduinoJson/faq/why-is-the-input-modifed/)
* [Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)

closed 2017-11-13 by bblanchon

desteaquino, 2017-11-22:
Thankyou for your answer. You were right. I optimized the json format and i got it right.

Thansks again. Regards:

Juanma

bblanchon, 2017-11-22:
I'm glad to read that 😄 
You're welcome Juanma, thank you for using ArduinoJson.