bblanchon/ArduinoJson #599: Loop only shows last values

ersinpw, 2017-11-04:
I am looping with my esp8266 through all WiFi hotspots, encode these in json and send these to my webbrowser. I have read lots of examples, but the problem I have couldn't fix my problem.

```
DynamicJsonBuffer JSONbuffer;
JsonObject& JSONencoder = JSONbuffer.createObject();
JsonArray& ap           = JSONencoder.createNestedArray("app");

for(int j=0; j<hotspots_known_count; j++) {
  JsonObject& app = ap.createNestedObject();
  hotspotinfo h_i   = hotspots_known[j];

  int i =0;
  char t[2*6] = { 0 };

  for(int i=0; i<6; i++) {
    snprintf(t+2*i, sizeof(t)-2*(i), "%02x", h_i.station[i]);
  }
  app["id"] = j;
  app["mac_address"] = t;
}
```

Suppose there are 2 hotspots found with the following mac addresses:
 - **2EB8A0B0437E**
 - **EAEE3CD2CB7B**

Then the json will look like:
`{"app":[{"id": 0, "mac_address": "EAEE3CD2CB7B"}, {"id": 1, "mac_address": "EAEE3CD2CB7B"}]}`

Note mac_address [0] and [1] has the same mac address. Because the id key works properly I think app["mac_address"] is a pointer to char t, however char t changes each loop. So the last loop determines which mac_address t will become?

I have tried things like
`strcpy(t, app["tomac"]);`
But unfortunately this resulted in an error.

ersinpw, 2017-11-04:
Found the solution in section 6!
https://bblanchon.github.io/ArduinoJson/doc/pitfalls/

This did the trick:
`app["mac_address"] = JSONbuffer.strdup(t);`

bblanchon, 2017-11-05:
I'm glad you found it @ersinpw.
I added [this question to the FAQ](https://bblanchon.github.io/ArduinoJson/faq/using-a-loop-why-are-values-all-the-same/), it's going to be easier to find.

closed 2017-11-05 by bblanchon