bblanchon/ArduinoJson #600: JSON parsing failed!

mjoelmendes, 2017-11-08:
hello 
this and my code:

//https://bblanchon.github.io/ArduinoJson/example/http-client/

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
#include <ESP8266HTTPClient.h>
#include <ArduinoJson.h>
#include <EEPROM.h>

WiFiManager wifiManager;
WiFiClient client;

#define Input_1_pin  12 // GPIO
#define Input_2_pin  13 // GPIO
#define output_1_pin 5  // GPIO
#define output_2_pin 4  // GPIO

//********* CONSTANTES ************
const int size_eeprom = 512;

const char* host = "jsonplaceholder.typicode.com"; // nao colocar o HTTP:// 
const char* resource = "/users/1";                    // http resource

const unsigned long HTTP_TIMEOUT = 10000;  // max respone time from server
const size_t MAX_CONTENT_SIZE = 512;       // max size of the HTTP response

//********* VARIAVEIS ************
String ee_data_read; // buffer que returna os dados lidos da eeprom

// The type of data that we want to extract from the page
struct UserData {
  char name[32];
  char company[32];
};

//******** EEPROM READ *************
void eeprom_read_all() {
  unsigned int i;
  char achar;

  ee_data_read = "";
  i = 0;
  while ( i < size_eeprom) {
    achar = char(EEPROM.read(i));

    if (achar != char(255)) {
      ee_data_read += achar;
      i++;
    }
    else break;
  }

}

//******** EEPROM WRITE *************
void eeprom_write(String data) {

  unsigned int i;
  for (i = 0; i < sizeof(data); i++) {
    EEPROM.write(i, data[i]);
  }

  EEPROM.commit();

}

//******** EEPROM CLEAR ALL *************
void eeprom_clear_all() {

  for (int i = 0; i < size_eeprom; ++i) {

    EEPROM.write(i, 255);

  }
  EEPROM.commit();

}

//********* POST in Server**************
void post(String Adata) {

  HTTPClient http;
  http.begin("http://arodrigues.esy.es/teste.php HTTP/1.1");
  http.addHeader("Content-Type", "application/x-www-form-urlencoded");
  //int httpCode = http.POST("id=esp&value=POSTesp");

  int httpCode = http.POST(Adata);

  String payload = http.getString();

  Serial.print("HTTP CODE:");
  Serial.println(httpCode);   //Print HTTP return code
  Serial.print("RESPONSE REQUEST: ");
  Serial.println(payload);    //Print request response payload

  http.end();  //Close connection

}

// Send the HTTP GET request to the server
bool sendRequest(const char* host, const char* resource) {
  Serial.print("GET ");
  Serial.println(resource);

  client.print("GET ");
  client.print(resource);
  client.println(" HTTP/1.0");
  client.print("Host: ");
  client.println(host);
  client.println("Connection: close");
  client.println();

  return true;
}

// Skip HTTP headers so that we are at the beginning of the response's body
bool skipResponseHeaders() {
  // HTTP headers end with an empty line
  char endOfHeaders[] = "\r\n\r\n";

  client.setTimeout(HTTP_TIMEOUT);
  bool ok = client.find(endOfHeaders);

  if (!ok) {
    Serial.println("No response or invalid response!");
  }

  return ok;
}

//********* READ RESPONSE JSON**************
bool readReponseContent(struct UserData* userData) {
  // Compute optimal size of the JSON buffer according to what we need to parse.
  // See https://bblanchon.github.io/ArduinoJson/assistant/
  const size_t BUFFER_SIZE =
    JSON_OBJECT_SIZE(8)    // the root object has 8 elements
    + JSON_OBJECT_SIZE(5)  // the "address" object has 5 elements
    + JSON_OBJECT_SIZE(2)  // the "geo" object has 2 elements
    + JSON_OBJECT_SIZE(3)  // the "company" object has 3 elements
    + MAX_CONTENT_SIZE;    // additional space for strings

  // Allocate a temporary memory pool
  DynamicJsonBuffer jsonBuffer(BUFFER_SIZE);

  JsonObject& root = jsonBuffer.parseObject(client);

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  }

  // Here were copy the strings we're interested in
  strcpy(userData->name, root["name"]);
  strcpy(userData->company, root["company"]["name"]);
  // It's not mandatory to make a copy, you could just use the pointers
  // Since, they are pointing inside the "content" buffer, so you need to make
  // sure it's still in memory when you read the string

  return true;
}

// Print the data extracted from the JSON
void printUserData(const struct UserData* userData) {
  Serial.print("Name = ");
  Serial.println(userData->name);
  Serial.print("Company = ");
  Serial.println(userData->company);
}

// Close the connection with the HTTP server
void disconnect() {
  Serial.println("Disconnect");
  client.stop();
}

// Pause for a 1 minute
void wait() {
  Serial.println("Wait 60 seconds");
  delay(60000);
}

// Open connection to the HTTP server
bool connect(const char* hostName) {
  Serial.print("Connect to ");
  Serial.println(host);

  bool ok = client.connect(host, 8080);

  Serial.println(ok ? "Connected" : "Connection Failed!");
  return ok;
}

void get_json() {

  if (connect(host)) {
    if (sendRequest(host, resource) && skipResponseHeaders()) {
      UserData userData;
      if (readReponseContent(&userData)) {
        printUserData(&userData);
      }
    }
  }
  disconnect();
  wait();

}

void int_var() {

  wifiManager.autoConnect("JSON");
  ee_data_read = "";
}

void setup() {
  pinMode(output_2_pin, OUTPUT);
  pinMode(output_1_pin, OUTPUT);

  int_var();

  Serial.begin(115200); // init serial
  EEPROM.begin(512);
  delay(10);

  wifiManager.autoConnect("IOT4YOU");

}

int value = 0;
void loop() {

  ++value;

  get_json();

}

This will give me this output:

Connect to jsonplaceholder.typicode.com
Connected
GET /users/1
JSON parsing failed!
Disconnect
Wait 60 seconds

someone can help me

bblanchon, 2017-11-10:
Does the unmodified example work?

MFreitasX7, 2017-11-11:
Hello, I'm using the same library but I stumbled on the same problem using the same basic Json client test code.

Would you help me?

bblanchon, 2017-11-11:
Parsing fails because the response is empty.

Something changed in typicode.com, they are now redirecting to HTTPS.
When you go to http://jsonplaceholder.typicode.com/users/1, you get a 301 redirect to https://jsonplaceholder.typicode.com/users/1.
We need to find another site for the example 😢 

Any idea?

Maxwe1, 2017-11-11:
yes i faced this problem then found that the site used in the example use now HTTPs than HTTP which will need ya to use SSL supported lib. there are many out there for the esp8266

commit 2eab5b01 (2017-11-13): Rewrote example JsonHttpClient.ino (fixes #600)
commit 126f7ab8 (2017-11-13): Rewrote example JsonHttpClient.ino (fixes #600)
closed 2017-11-13 by bblanchon