bblanchon/ArduinoJson #601: Cann purchase e-book

heritageorchard, 2017-11-09:
Hi,

Fantastic library !!! Thank You for your work.

The link: https://ebook.benoitblanchon.fr/buy

appears to be broken.

Take Care,
Mike

bblanchon, 2017-11-09:
Not yet, I'm in the process of writing it and should be ready by the end of this month.
Sorry about that 😃

heritageorchard, 2017-11-09:
No worries at all my friend. 

Would you consider some way to identify c++ knowledge versus Json?

I have 30 years(!!!) c++ experience so need the Json side more. 

As an aside, I want to use Json as the persistence. Should my ctor take in a JsonObject or varient!

Once again I really appreciate your work. I had open source code and a few textbooks for the verification world and know the effort involved. 

Typos due to small phone and big fingers.

bblanchon, 2017-11-10:
> As an aside, I want to use Json as the persistence. Should my ctor take in a JsonObject or varient!

If you want to hydrate an object from a JSON object, you most likely want its constructor to take a `JsonObject&`.
A `JsonVariant` would just add a layer of indirection that is probably not necessary.

heritageorchard, 2017-11-10:
Yes that makes perfect sense. Sorry for being dense.

closed 2017-11-10 by heritageorchard

bblanchon, 2017-12-08:
# [The book is out!!!!](https://leanpub.com/arduinojson) 🎉