bblanchon/ArduinoJson #602: parseObject nesting limit does not seem to work properly

boblemaire, 2017-11-09:
Two level Json string parses to {} with nestlevel set to  <=2, and parses both levels when set to 3=>.
ESP8266/arduino, staged core, latest platformIO toolchain, ArduinoJson-ID64 (latest release)

MVCE - Would need a day to write a new program to reproduce.  Here's what I've got:

Json string:
`data={"name":"abc","start":1510172880000,"end":1510259400000,"interval":120,"fixinterval":false,"floatingtime":1,"yaxismin":"auto","yaxismax":"auto","showmissing":false,"showtag":true,"showlegend":true,"feedlist":[{"id":"IPInput_1","name":"Input_1","tag":"Power","yaxis":1,"fill":0,"scale":"1","delta":false,"getaverage":false,"dp":1,"plottype":"lines"},{"id":"IPInput_2","name":"Input_2","tag":"Power","yaxis":1,"fill":0,"scale":"1","delta":false,"getaverage":false,"dp":1,"plottype":"lines"}]}`

Program code:
```
void handleGraphCreate(){
  trace(T_uploadGraph,11);
  DynamicJsonBuffer Json;
  JsonObject& graph = Json.parseObject(server.arg("data"), 3);
  graph.prettyPrintTo(Serial);       
  returnOK();
  return;
```
Output to Serial when nestlevel <= 2:
`{}`

Output to Serial; when nestlevel >=3:
```
{
  "name": "abc",
  "start": 1510172880000,
  "end": 1510259400000,
  "interval": 120,
  "fixinterval": false,
  "floatingtime": 1,
  "yaxismin": "auto",
  "yaxismax": "auto",
  "showmissing": false,
  "showtag": true,
  "showlegend": true,
  "feedlist": [
    {
      "id": "IPInput_1",
      "name": "Input_1",
      "tag": "Power",
      "yaxis": 1,
      "fill": 0,
      "scale": "1",
      "delta": false,
      "getaverage": false,
      "dp": 1,
      "plottype": "lines"
    },
    {
      "id": "IPInput_2",
      "name": "Input_2",
      "tag": "Power",
      "yaxis": 1,
      "fill": 0,
      "scale": "1",
      "delta": false,
      "getaverage": false,
      "dp": 1,
      "plottype": "lines"
    }
  ]
}
```

bblanchon, 2017-11-10:
Hi @boblemaire,

Indeed, parsing fails when `nestingLimit` is too low.
In that case `parseObject()` returns `JsonObject::invalid()`.
As a matter of fact, `JsonObject::invalid()::prettyPrintTo()` generates an empty object.

I don't see any problem.
Can you elaborate?

Regards,
Benoit

boblemaire, 2017-11-10:
Sorry,
I was under the mistaken impression that nesting limit would enable parsing for only the elements at or below the specified limit.  I now see that it simply doesn't parse an element whose content exceeds the specified limit.

closed 2017-11-10 by boblemaire

bblanchon, 2017-11-10:
It is a safety measure to protect your program from stack overflow.
Indeed if a malicious user sends a deeply nested array like `[[[[[[[[[[[[[[[[`, it would cause a stack overflow.

I'm afraid I cannot implement the feature you expect as it would defeat the purpose of this protection.

Sorry.

boblemaire, 2017-11-10:
I'm trying to understand this.  Are you saying that the nesting levels specifically cause the stack overflow?  Possibly because of recursive use of the parsing code?  And is that different from protection against thousands of elements in a single level?  I can see a difference, where the stack might not be under your control and heap is, but if that is the case, why would the functionality that I assumed not afford the same protection if it suppressed the recursive calls?
Once again, this is academic for me, because I can get by without the limit, but I'm trying to understand the rational.  Thanks.

bblanchon, 2017-11-10:
There is indeed a recursive call in the parser.
I'm not sure it would be possible to remove it without any allocation on the heap.

boblemaire, 2017-11-10:
I guess what I'm asking is if levels beyond nestinglevel are ignored, wouldn't that eliminate the recursive calls?

bblanchon, 2017-11-10:
But how to find the matching `}` or `]`?
On the one hand, we need to stop the recursion to avoid the stack overflow.
On the other hand, we need to count the opening and closing braces.

boblemaire, 2017-11-10:
I see.  As is always the case, the plot thickens.  But as a final word, Json syntax is pretty simple, and a non-recursive function to rip through the text until the nesting level is restored, would be fairly simple affair.  Anyway, thanks for the insights.