bblanchon/ArduinoJson #603: Help, memory leak

mcmega, 2017-11-11:
Hi!
Thank you for your work!

Prompt, how correctly to work with library to avoid memory leak?
I request the data in a loop (findDevices) and noticed that the memory decreases with each time.

```
AsyncUDP udp;
DynamicJsonBuffer devListBuffer;
JsonArray &devsList = devListBuffer.createArray();

void findDevices() {
  digitalWrite(STATUS_LED_PIN, LOW);
  if (WiFi.status() != WL_CONNECTED)
    return;
  while (devsList.size())
    devsList.remove(0);
  JsonObject &locDev = devsList.createNestedObject();
  locDev["ip"] = devListBuffer.strdup(WiFi.localIP().toString().c_str());
  locDev["da"] = da;
  locDev["dt"] = DEVICE_TYPE;
  locDev["dm"] = DEVICE_MODEL;
  locDev["dn"] = dn;
  locDev["dc"] = dc;
  udp.broadcastTo(
    "M-SEARCH * HTTP/1.1\r\n"
    "Host:239.255.255.250:1900\r\n"
    "ST:upnp:rootdevice\r\n"
    "Man:\"ssdp:discover\"\r\n"
    "MX:3\r\n\r\n", 1900
  );
}

Heap:
25120 .. 21008 .. 12800 .. 8250 .. Crash
```
As a result, the ESP does not work after a while.

bblanchon, 2017-11-13:
Hi @mcmega,

I know that this is very tempting to use a JsonObject as a property tree to store the state of your application.
Unfortunately, ArduinoJson is poorly adapted to this task; it is optimized for the JSON serialization only.

Instead of a `JsonObject`, I recommend that you use an array of custom struct like this:

```c++
struct DeviceInfo {
   char ip[16];
   char da[64];
};

const size_t MAX_DEVICES = 10;
DeviceInfo devices[MAX_DEVICES];
int deviceCount = 0;
```

Then use ArduinoJson only to serialize/deserialize this collection to/from JSON.
Not only the code will be simpler, but it will also be much faster!

Regards,
Benoit

mcmega, 2017-11-13:
Thank you!
A good idea!

closed 2017-11-13 by mcmega