bblanchon/ArduinoJson #606: Cannot recover string or char from json. Numbers are OK

MikeyMoMo, 2017-11-13:
Running on a WeMos Mini D1.  

Here's the input line from api.openweathermap.org:

```{"coord":{"lon":-122.26,"lat":38.1},"weather":[{"id":701,"main":"Mist","description":"mist","icon":"50d"}],"base":"stations","main":{"temp":13.33,"pressure":1022,"humidity":93,"temp_min":12,"temp_max":14},"visibility":16093,"wind":{"speed":1.5,"deg":150},"clouds":{"all":90},"dt":1510587420,"sys":{"type":1,"id":438,"message":0.1695,"country":"US","sunrise":1510584534,"sunset":1510621058},"id":5405380,"name":"Vallejo","cod":200}```

Tried all of the following.  All print nothing except the text header.  Numbers recovered from the same json packet are just fine.
```
  const char* WXMain = root["weather"]["main"];
  Serial.print("WXMain: "); Serial.println(WXMain);
  
  String weatherMain = "";
  weatherMain = root["weather"]["main"].as<String>();
  Serial.print("WXMain: "); Serial.println(weatherMain);
  
  const char* City = root["sys"]["name"];
  Serial.print("City:   "); Serial.println(City);

```
The following returns the sunrise time, also from the "sys" portion:

```
  unsigned long int sunrise, sunset;
  sunrise = root["sys"]["sunrise"];

```
Looks good (converted from the unsigned long int to human readable format):

```
Sunrise: 2017/11/13 6:48:54
```

What must I do to retrieve textual elements?  I tried following the examples I found but none are working for me.

Thanks,
Mike

MikeyMoMo, 2017-11-13:
OK.  Never mind.  I was badly misreading the levels of the json packet.  Here's the corrected statements to get what I wanted.  Part of the problem is that the code I originally got had a bug in it having to do with the [ and ] characters.  It was removing them when it should not have been so it was causing a problem right there.  Then when I figured out that that was a problem and put them back, I was not seeing them as applying because, in my mind by that time, they seemed irrelevant.  They were not!  

So all is well.  Now that I got through that little bug and then got my head screwed back on right, I am seeing the data I want to see.  Sorry for the false alarm.  Maybe this will help someone not get hung up on that original bug in the code.

  ```
  const char* WXMain = root["weather"][0]["main"];
  Serial.print("WXMain: "); Serial.println(WXMain);
  
  weatherMain = root["weather"][1]["main"].as<String>();
  Serial.print("WXMain: "); Serial.println(weatherMain);
  
  const char* City = root["name"].asString();
  Serial.print("City:   "); Serial.println(City);

```
And here is the entire, corrected and working demonstration code.  It's a little ugly but it is just first day demo code.  I did not write it.  Copied and modified heavily from someone else's demo code for scrolling LED matrix.  It will get prettier for the production version.  You will still have to customize it with your api key and city name and code.

```
#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <Time.h>                 // Time functions
#include <TimeLib.h>            // Time functions
WiFiClient client;

// =======================================================================
// CHANGE YOUR CONFIG HERE:
// =======================================================================
const char* ssid     = "your SSID";     // SSID of local network
const char* password = "your PW";   // Password on network
String weatherKey = "your api key here"; // For Me.  Get yours at openweathermap.org
String weatherLang = "&lang=en";
String cityID = "5405380"; // Vallejo, CA  Change to your city code.  Can be hard to find...

String weatherDescription = "";
String weatherMain = "";
int humidity;
int pressure;
float temp;
float tempMin, tempMax;
int clouds;
unsigned long int sunrise, sunset;
float windSpeed;
String myDate;
String myTime;
String weatherString;

const char *weatherHost = "api.openweathermap.org";

float utcOffset = -8;  // PST
long localEpoc = 0;
long localMillisAtUpdate = 0;

IPAddress      ip( 10,  10,   0,  24);  // IP My static address
IPAddress gateway( 10,  10,   0,   1);  // Where to go when in doubt or just plain lost.
IPAddress  subnet(255, 255, 255,   0);  // The usual...
// read OpenWeather api description for more info

int WX_UpdateCycles = 600; // Update the WX info each 10 minutes.
int h,m,s;
/****************************************************************************************/
void setup() 
/****************************************************************************************/
{
  Serial.begin(115200);
  Serial.print("\n\nConnecting WiFi ");
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
[... 133 lines cut ...]
      s = line.substring(29, 31).toInt();
      localMillisAtUpdate = millis();
      localEpoc = (h * 60 * 60 + m * 60 + s);
    }
  }
  client.stop();
}
/****************************************************************************************/
void updateTime()
/****************************************************************************************/
{
  long curEpoch = localEpoc + ((millis() - localMillisAtUpdate) / 1000);
  long epoch = round(curEpoch + 3600 * utcOffset + 86400L) % 86400L;
  h = ((epoch  % 86400L) / 3600) % 24;
  m = (epoch % 3600) / 60;
  s = epoch % 60;
  myTime = "";
  if (h < 10) myTime += "0"; myTime += String(h); myTime += ":"; 
  if (m < 10) myTime += "0"; myTime += String(m); myTime += ":"; 
  if (s < 10) myTime += "0"; myTime += String(s);
}
/****************************************************************************************/
float CtoF (float degC) { return ((9.0 / 5.0) * degC) + 32.0; }
/****************************************************************************************/
```

bblanchon, 2017-11-14:
Hi @MikeyMoMo,

Thanks for posting the solution.
Did you try the [ArduinoJson Assistant](http://arduinojson.org/assistant/)?

Regards,
Benoit

closed 2017-11-14 by bblanchon