bblanchon/ArduinoJson #607: Parsing JSON File fails in Class

merlinschumacher, 2017-11-13:
Im trying to read a JSON File inside a class to provide a configuration management. I run in to the "use of deleted function 'ArduinoJson::JsonObject&" issue but can't solve it with the help of the FAQ.
Here is my code:

## library.cpp

```cpp
#include "library.h"

Configuration::Configuration(const char* filename)
{
  jsonFile = filename;
}

bool Configuration::load() {
  SPIFFS.begin(true);

  DynamicJsonBuffer _jsonBuffer;
  File configFile = SPIFFS.open(jsonFile, "r");
  if (!configFile) {
    Serial.println("Failed to open config file");
  }

  size_t size = configFile.size();

  char buf[size];

  configFile.readBytes(buf, size);

  Serial.println(buf);
  jsonConfig = _jsonBuffer.parseObject(buf);
  if (!jsonConfig.success()) {
    Serial.println("Failed to parse config file");
  }
  jsonConfig.printTo(Serial);
  configFile.close();
}

bool Configuration::save() {
  Serial.println("Saving config");
  File configFile = SPIFFS.open(jsonFile, "w");
  if (!configFile) {
    Serial.println("Failed to open config file for writing");
    return false;
  }

  jsonConfig.printTo(configFile);
  jsonConfig.printTo(Serial);
  configFile.close();
  return true;
}

bool Configuration::put(const char* key, const char* value) {
  Serial.println("Storing");
  jsonConfig[key] = value;
  jsonConfig.printTo(Serial);
}

```

## library.h

```cpp

#ifndef library_h
#define library_h
#include "Arduino.h"
#include <ArduinoJson.h>
#include "FS.h"
#include "SPIFFS.h"

class Configuration {
  public:
    Configuration(const char* filename);
    bool load();
    bool save();
    const JsonObject* get(const char*);
    bool put(const char* key,const char* value);
  private:
    const char* jsonFile;
    DynamicJsonBuffer jsonBuffer;
    JsonObject& jsonConfig = jsonBuffer.createObject();
};

#endif
```

## library.ino

```cpp
#include "libctiot.h"

Configuration configuration("/config.json");

void setup() {
      
  Serial.begin(115200);
  
  configuration.load();
  configuration.put("test","123");
  configuration.save();
}

void loop() {
}

```

bblanchon, 2017-11-14:
Hi @merlinschumacher,

I cannot find the problem in the code.
Can you please copy paste the exact compiler error?

Also, I don't recommend using a global `JsonBuffer` as in this program.
The fact that it's hidden in a `Configuration` class doesn't change the problem.

Please read:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://arduinojson.org/faq/why-shouldnt-i-use-a-global-jsonbuffer/)

Regards,
Benoit

merlinschumacher, 2017-11-14:
The compiler error i get is 
```
use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
```

Essentially it boils down to creating a variable that can hold a JSON structure and can be globally manipulated inside the class.

bblanchon, 2017-11-14:
Indeed, you cannot copy a `JsonObject`.
Here is the line causing the error:

```c++
jsonConfig = _jsonBuffer.parseObject(buf);
```

You cannot reassign a C++ reference either, so you have to replace with a pointer or `JsonVariant`.

Again, you're fighting against the library here, ArduinoJson is probably not the right tool for this job.

merlinschumacher, 2017-11-15:
So i tried it another way and found this solution to store config data in a json file and to read them back into a map:

## test.ino

```cpp
#include "test.h"
Configuration configuration("/config.json");

void setup() {
  Serial.begin(115200);
  configuration.load();
  configuration.put("foo", "peng");
  configuration.put("bar", "foo");
  configuration.put("foo", "json");
  configuration.dump();
  configuration.save();
}

void loop() {
}
```
## test.cpp
```cpp
#include "test.h"
Configuration::Configuration(const char* filename)
{
  jsonFile = filename;
}

bool Configuration::load() {
  SPIFFS.begin();
  File configFile = SPIFFS.open(jsonFile, "r");
  if (!configFile) {
    Serial.println("Failed to open config file");
  }

  size_t size = configFile.size();
  char buf[size + 1];
  configFile.readBytes(buf, size);

  DynamicJsonBuffer _jsonBuffer;
  JsonObject& _jsonData = _jsonBuffer.parseObject(buf);

  if (!_jsonData.success()) {
    Serial.println("Failed to parse config file");
  };

  for (JsonObject::iterator it = _jsonData.begin(); it != _jsonData.end(); ++it)
  {
    const char* key = it->key;
    auto value = it->value.as<char*>();
    char* cfgkey = _jsonBuffer.strdup(key);
    char* cfgvalue = _jsonBuffer.strdup(value);
    this->put(cfgkey, cfgvalue);
  }

  configFile.close();
}

bool Configuration::save() {
  File configFile = SPIFFS.open(jsonFile, "w");
  if (!configFile) {
    Serial.println("Failed to open config file for writing");
    return false;
  }
  if (configuration.empty())
  {
    Serial.println("Configuration empty");
  };
  DynamicJsonBuffer _jsonBuffer;
  JsonObject& _jsonData = _jsonBuffer.createObject();

  for (auto const& x : configuration)
  {
    _jsonData.set(_jsonBuffer.strdup(x.first), _jsonBuffer.strdup(x.second));
  }

  _jsonData.printTo(configFile);
  _jsonData.prettyPrintTo(Serial);
  configFile.close();
  return true;
}

bool Configuration::put(const char* key, char* value) {
  configuration[key] = value;
}

bool Configuration::dump() {
  for (const auto &p : configuration) {
    Serial.print( "configuration[");
    Serial.print(p.first);
    Serial.print("] = ");
    Serial.println(p.second);
  };
}
```

## test.h
```cpp
#ifndef test_h
#define test_h

#include "Arduino.h"
#include <ArduinoJson.h>
#include "FS.h"
#include "SPIFFS.h"
#include "map"
class Configuration {
  public:
    Configuration(const char* filename);
    bool load();
    bool save();
    bool dump();
    bool put(const char* key, char* value);
    struct cmp_str
    {
      bool operator()(char const *a, char const *b)
      {
        return strcmp(a, b) < 0;
      }
    };
    std::map<const char*, const char*, cmp_str> configuration;

  private:
    const char* jsonFile;

};
#endif
```

bblanchon, 2017-11-15:
On the one hand, it is better because the `JsonBuffer` is short-lived; so the memory leak will not appear.

However, the `std::map` stores pointers to strings stored in the (destructed) `JsonBuffer`.
(see [JsonBuffer::strdup](http://arduinojson.org/api/jsonbuffer/strdup/).
This cannot work.
Instead, you need `std::map<std::string,std::string>`.

And now, you're limited to storing strings in the configuration.
So you'd quickly need a `std::map<std::string,std::variant<std::string,int,float...>>`.

I understand the motivation, but I don't think that this is an appropriate pattern for embedded development: too much memory wasted in dynamic allocation (`std::map` is a BST), too much CPU cycles wasted in duplicating and searching strings, and an explosion of the size of the firmware.

You're not the first to come with the exact same idea. From what I've seen, it's rooted in the user experience with a dynamically-typed language like Python, Javascript or Ruby.

My advice: if you need to store a configuration with 2 strings `foo` and `bar`, here is how to do it:

```c++
struct Config {
    char foo[32];
    char bar[32];
};
```

BTW, the `buf` array in `load()` is not required, see [Can I parse from a stream?](http://arduinojson.org/faq/can-i-parse-data-from-a-stream/)

closed 2017-11-24 by bblanchon