bblanchon/ArduinoJson #608: Compile problem on NodeMCU 0.9

lesept777, 2017-11-14:
Hi
I have a compilation error while trying to use ArduinoJson on a NodeMCU 0.9
Here is the error message

``

In file included from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:10:0,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:16,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:15,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:10,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:10,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:10,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10,

                 from C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:10,

                 from C:\Users\Chuwi\Documents\Arduino\Test_Json\Test_Json.ino:3:

C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp: In instantiation of 'ArduinoJson::JsonVariantCasts<TImpl>::operator T() const [with T = char*; TImpl = ArduinoJson::JsonObjectSubscript<const char*>]':

C:\Users\Chuwi\Documents\Arduino\Test_Json\Test_Json.ino:42:31:   required from here

C:\Users\Chuwi\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantCasts.hpp:52:35: error: invalid conversion from 'ArduinoJson::Internals::JsonVariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]

     return impl()->template as<T>();

                                   ^

exit status 1
Erreur de compilation pour la carte NodeMCU 0.9 (ESP-12 Module)``

I am very new to Json, and I don't understand the meaning of the error. I installed the latest version of the library, using the library manager of the IDE (version 1.8.2). I attach my sketch, I hope you can explain my mistake...

[Test_Json.zip](https://github.com/bblanchon/ArduinoJson/files/1472407/Test_Json.zip)

Thanks for your help

bblanchon, 2017-11-15:
Hi,

I bet that a `const` keyword is missing in line 42 of `Test_Json.ino`.

Please see: 

* [FAQ: Error: invalid conversion from 'const char*' to 'char*' [-fpermissive]](http://arduinojson.org/faq/invalid-conversion-from-const-char-to-char/)

Regards,
Benoit

lesept777, 2017-11-15:
Thanks, it works.. but does not provide the expected result. I get the payload but cannot parse it. As I said, it's my first test, can you help me and tell me why it's not parsing?

[Test_Json.zip](https://github.com/bblanchon/ArduinoJson/files/1475758/Test_Json.zip)

lesept777, 2017-11-17:
OK, I found the assistant, which fully answers my question! Breat idea! Thanks

closed 2017-11-17 by lesept777