bblanchon/ArduinoJson #609: value returned for operator[] is diferent to JsonObject::set()

ferchinas, 2017-11-15:
hello: I'm trying to verify the correct assignment of an object and I find this difference.
```
success = (jsonobjet["Ty"] = Type);			// ok success :1
success = (jsonobjet["Nm"] = Value->Name);	// FAIL success :0
success = jsonobjet.set("Nm2", Value->Name); // ok success :1
```
In line 2 of the example, it returns 0, but the object "Nm" is added correctly to the json.
Could you help me find out what I'm wrong about?
Thank you very much.

bblanchon, 2017-11-16:
Hello @ferchinas,

Indeed, [`JsonObject::set()`](http://arduinojson.org/api/jsonobject/set/) returns a boolean indicating the success; whereas [`JsonObject::operator[]`](http://arduinojson.org/api/jsonobject/subscript/) returns a reference to the `JsonVariant`.

Regards,
Benoit

ferchinas, 2017-11-16:
Thank you very much! sorry for the basic question, long days of code ...

closed 2017-11-16 by ferchinas