bblanchon/ArduinoJson #610: How can I use a Struct to return JsonArray from a function ?

gusantor, 2017-11-16:
Thank you for your library

I'm trying to bring data from json files, repeteadly. To do so, I want to build a function which receive a file name and return a Struct, with a jsonArray among others

At jsonStruct, if I not declare puntos as reference, receive this error message: use of deleted function 'jsonStruct::jsonStruct()', which leads me to #290, but this conversation surpases my knowledge

if I declare puntos (inside struct) as reference, I receive error message: error: invalid conversion from 'ArduinoJson::Internals::JsonVariantAs<char*>::type {aka const char*}' to 'char*' [-fpermissive]
     return as<T>();

So my question is how can I use a Struct to return JsonArray from a function ?, thank you in advance

    //the struct declared globally
    struct jsonStruct {
      char *nombre;
      char *titulo;
      char *descripcion;
      JsonArray *puntos;//<-- not declared as reference leads me to #290
    };

    //Then I call the function from here
    void handleLista(AsyncWebServerRequest *request){

      jsonStruct jsonData;
      jsonData = traeJsonFile("/lista-pares-biomagneticos.json");

      request->send(200, "text/html", jsonData.nombre);

    }

    //and the function to be called many times (for each file)
    jsonStruct traeJsonFile(char *fileName){
      File file = SPIFFS.open(fileName, "r");

      jsonStruct fileData;

      if(file){
        size_t size = file.size();
        std::unique_ptr<char[]> buf (new char[size]);
    
        file.readBytes(buf.get(), size);
    
        JsonObject& root = jsonBuffer.parseObject(buf.get());

        if (root.success()) {
            fileData.nombre = root["nombre"];//<-- error is generated here
                //fileData.nombre = "test";//<-- no error
                .
                .
                .I should continue populating fileData with JsonArray and others
	    }
        }
        return fileData;
    }

bblanchon, 2017-11-16:
Hi @gusantor,

It is a good idea to return a `struct` like this :+1:
The only problem is that this `struct` contains pointers, instead of actual values.

Here is what I would do:

```c++
const int MAX_PUNTOS = 30;

struct jsonPunto {
  // insert whatever you need here
};

struct jsonStruct {
  char nombre[16];
  char titulo[16];
  char descripcion[64];
  jsonPunto puntos[MAX_PUNTOS];
  int puntosCount;
};

// and the function to be called many times (for each file)
jsonStruct traeJsonFile(char *fileName) {
  File file = SPIFFS.open(fileName, "r");

  jsonStruct fileData;
  DynamicJsonBuffer jsonBuffer(/*use arduinojson.org/assistant */);
  JsonObject &root = jsonBuffer.parseObject(file);

  if (root.success()) {
    strlcpy(fileData.nombre, root["nombre"], sizeof(fileData.nombre));
    strlcpy(fileData.titulo, root["titulo"], sizeof(fileData.titulo));
    strlcpy(fileData.descripcion, root["descripcion"], sizeof(fileData.descripcion));

    fileData.puntosCount = 0;
    for( JsonObject& punto : fileData["puntos"].as<JsonArray>()) {
      // set the value of fileData.punto[fileData.puntosCount]
      fileData.puntosCount++;
    }
  }

  return fileData;
}
```

Now, the `struct` **owns** the memory, it doesn't store pointers.

Regards,
Benoit

gusantor, 2017-11-16:
Thanks, I'll give it a tray and let you know

closed 2017-11-24 by bblanchon