bblanchon/ArduinoJson #612: Changing the value of a key in an object

ferchinas, 2017-11-16:
oldValue =  jsonObject[key];  // value = 0
newValue = 500;
Hello again:
I have my configuration file in the file system of esp8266. What is the best way to replace a value of a key?
I am trying this way but and work fine, but I have problems with numbers greater than 255 when the previous value was less.

And I do not find that I'm doing wrong.
There is enough space in the json buffer.

```

if (jsonObject.containsKey(key)) {

	jsonObject.remove(key);		       //  I tried with and without this line		

	if (!jsonObject.set(key, newValue )) // this pass ok, but it does not update the value
		return (ERROR);
					
	if (newValue != jsonObject[key])     //here i detect the error, 
		return (ERROR);
}	
	
```

bblanchon, 2017-11-17:
Hi @ferchinas,

Here is an online demo:
https://wandbox.org/permlink/LYkuILoDnrxPD0wK

The issue is probably somewhere else in the program.
Look for type overflow (btw what is the type of newValue?) or undefined behavior (such as destructed buffers).

Regards,
Benoit

ferchinas, 2017-11-17:
Hi @bblanchon 

Muchas gracias por la respuesta y gracias a Wandbox (no lo conocia) encontre el error. 
De a poco voy descubriendo la enorme potencia de ArduinoJson

Regards,
Fernando

closed 2017-11-17 by ferchinas