bblanchon/ArduinoJson #613: Parsing failed. I dont know why

hansfeenstra1997, 2017-11-16:
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

//JSON PART
#include <ArduinoJson.h>

#define USE_SERIAL Serial

ESP8266WiFiMulti WiFiMulti;

void setup() {
 
    USE_SERIAL.begin(115200);
    
    USE_SERIAL.println();
    USE_SERIAL.println();
    USE_SERIAL.println();
    
    pinMode(2, OUTPUT);
    digitalWrite(2, LOW);    

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("PIER_ASUS", "peterjanlammiehans");

}

void loop() {
  

    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;        
        http.begin("http://192.168.1.72/test.html"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        int httpCode = http.GET();

        if(httpCode > 0) {
          
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);
            
            if(httpCode == HTTP_CODE_OK) {
                //String payload = http.getString();

                const size_t BUFFER_SIZE = JSON_OBJECT_SIZE(3) + JSON_ARRAY_SIZE(2) + 60;
                DynamicJsonBuffer jsonBuffer(BUFFER_SIZE);

                Serial.println(http.getString());
                JsonObject& root = jsonBuffer.parseObject(http.getString());
                
                //json(payload);
                //StaticJsonBuffer<500> jsonStringBuffer;
                //JsonObject& root = jsonStringBuffer.parseObject(http.getString());

                if (!root.success()) {
                Serial.println("parseObject() failed");
              } else{
                int switch_id = root["switch_id"];
                int status = root["status"];
                USE_SERIAL.printf("Switch ID: %d\n", switch_id);
                USE_SERIAL.printf("Status: %d\n", status);
              }
                
                //USE_SERIAL.println(payload);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }

        http.end();
    }

    delay(10000);
}

bblanchon, 2017-11-17:
Hi,

It is very likely to be caused by a wrong input.

I'm not familiar with `ESP8266HTTPClient`.
Does the result of `http.getString()`  include the headers?
Are you sure that [Chunked Transfer Encoding](https://fr.wikipedia.org/wiki/Chunked_transfer_encoding) is disabled or handled by the library?

Please check [Why parsing fails?](http://arduinojson.org/faq/why-parsing-fails/), your answer is there.

Regards,
Benoit

hansfeenstra1997, 2017-11-21:
Hello,

The result of http.getString() does not include headers. It contains the result from the get request: "{switch_id":1, "status":1}".

Chunked transfer encoding is handled by the library. I searched on the "Why parsing fails" page. And I can't find the solution for this.

Regards,
Hans

bblanchon, 2017-11-21:
This is not a valid JSON document, a quote is missing

```json
{switch_id":1, "status":1}"
```

The code works after adding the missing quote.
Demo: https://wandbox.org/permlink/jjgyAWUEZwcgseGJ

hansfeenstra1997, 2017-11-21:
I see, thank you!

Thanks for your support.

Can i put my code as an example on your page. I work with a WiFi module, maybe other people also want to try it. For a combination for IOT.

bblanchon, 2017-11-22:
@hansfeenstra1997, feel free to add a link to your project [in the wiki](https://github.com/bblanchon/ArduinoJson/wiki/Projects-using-ArduinoJson)

closed 2017-11-22 by bblanchon