bblanchon/ArduinoJson #615: Creating a JSON out of an array of unsigned char

Chinoforest, 2017-11-17:
I know this is not the place to ask this, but I didn't know where to ask. The question is too specific.

I have a class that is basically a circular buffer using a fixed length unsigned char array. I can null terminate it if neccesary (it's my code, so I can modify it).

I need to send this array of bytes to the client (via ESPAsyncWebServer library) but it's a large array so I want to be 100% sure there's zero copy of the data in the process. 

What I'm wondering, if I create a json object and add my unsigned char array, will your library copy it or just use the same space of memory? (without copying it). 

Example:

unsigned char data[FIXED_LENGHT]; 
// fill the data array here and null terminate it
DynamicJsonBuffer jsonBuffer;
 JsonObject& root = jsonBuffer.createObject();
 root["bytesarray"] = data;

Does this allocates new memory for the data array or it just uses the same memory.

bblanchon, 2017-11-17:
Hi @Chinoforest,

It seems to be the right place for this question 😄 

The following statement doesn't make a copy:

```c++
root["bytesarray"] = data;
```

but this one does:

```c++
root["bytesarray"] = jsonBuffer.strdup(data);
```

But I'm not sure that's what you want...

This program produces:

```json
{"bytesarray":"abc"}
```

But maybe you're looking for:

```json
{"bytesarray":[97,98,99,0]}
```

In that case, you need to call [JsonObject::createNestedArray()](http://arduinojson.org/api/jsonarray/createnestedarray/) and [JsonArray::copyFrom()](http://arduinojson.org/api/jsonarray/copyfrom/).

Regards,
Benoit

Chinoforest, 2017-11-17:
Thanks for the answer. I will reformulate the question.

Suppose I have this array (always null terminated).
`foo [5] = { 0x16, 0x02, 0x77, 0x40, 0x00};`
If I do:
`DynamicJsonBuffer jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
root["bytesarray"] = foo;`

I will get this json? (an there will be not data duplication... right?)
`{"bytesarray": 0x16027740}`

Don't worry about the data itself because I will parse the json as binary on the other end.

bblanchon, 2017-11-20:
The following is not a valid JSON document:

```json
{"bytesarray": 0x16027740}
```

The program above would generate a string with non-printable characters.
And there will be no data duplication.

Chinoforest, 2017-11-21:
Yes I know, I didn't care if the characters were not printable. 

Anyway, I ended up doing it using another way (without your library). 

Nevertheless, I'm using your library for other stuff and it works perfectly! Great work!! Thanks!

bblanchon, 2017-11-21:
I'm glad to read that. 

Thank you for using ArduinoJson.

Don't hesitate to add a link to your project in the [wiki](https://github.com/bblanchon/ArduinoJson/wiki/Projects-using-ArduinoJson)

closed 2017-11-21 by bblanchon