bblanchon/ArduinoJson #619: Exception on containsKey, in ESPAsyncWebserver handler

klaasdc, 2017-11-26:
I'm using a NodeMCU/ESP8266 with the 2.4.0-rc2 Arduino library.

In my sketch, I have an ESPAsyncWebserver that has to process a small POST with JSON values:

```arduino
void onResetConfig(AsyncWebServerRequest *request){
  if (request->hasParam("body", true)){
    AsyncWebParameter* p = request->getParam("body", true);
    String json = p->value();
    Serial.println(json);
    DynamicJsonBuffer locJsonBuffer;
    JsonObject& root = locJsonBuffer.parseObject(json);
    if (root.containsKey("reset")){
      Serial.println("Found reset key");
      loadDefaultConfig();
      writeSettingsToFile();
      Serial.println("Settings reset!");
      request->send(202); return;
    }
  }
  //No post data specified
  request->send(405);
}
```

When running, the first Serialprint gives `{"reset":true,"type":1}`, as expected. But then I get a stack trace, that refers to calling `root.containsKey("reset")`. 
```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 26 results
0x40215497: strcmp at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strcmp.S line 471
0x40209bdd: set_impl  at C:\Users\klaasdc\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp line 319
:  (inlined by) set  at C:\Users\klaasdc\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonObject.hpp line 124
:  (inlined by) ArduinoJson::Internals::JsonParser  ::Reader, ArduinoJson::DynamicJsonBufferBase &>::parseObject() at C:\Users\klaasdc\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/JsonParserImpl.hpp line 112
0x4021153d: Print::write(char const*) at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/Print.cpp line 87
0x40209c42: ArduinoJson::JsonObject& ArduinoJson::JsonBufferBase   >::parseObject (String&, unsigned char) at C:\Users\klaasdc\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp line 89
0x4021160a: Print::printf(char const*, ...) at ?? line ?
0x40209f5e: onResetConfig(AsyncWebServerRequest*) at C:\Users\klaasdc\AppData\Local\Temp\arduino_build_124071\sketch/webhandles.cpp line 241
0x4010020c: _umm_free at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1291
0x40211954: String::~String() at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 719
0x40213574: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*) at c:\users\klaasdc\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40210084: std::function ::operator()(AsyncWebServerRequest*) const at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebServer.cpp line 79
:  (inlined by) AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebHandlerImpl.h line 95
0x40211954: String::~String() at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/WString.cpp line 719
0x4020e4a8: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 777
0x40101f88: wDev_ProcessFiq at ?? line ?
0x4020e505: _M_invoke at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncWebServer\src/WebRequest.cpp line 777
0x40210b9d: AsyncClient::_recv(tcp_pcb*, pbuf*, long) at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x402125cc: loop_wrapper at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/core_esp8266_main.cpp line 57
0x40210be8: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at C:\Users\klaasdc\Documents\Arduino\libraries\ESPAsyncTCP\src/ESPAsyncTCP.cpp line 650
0x402382e0: tcp_input at core/tcp_in.c line 394 (discriminator 1)
0x40106ed8: pvPortMalloc at C:\Users\klaasdc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0-rc2\cores\esp8266/heap.c line 40
0x402366fd: ip_input at core/ipv4/ip.c line 559
0x40235b01: ethernet_input at netif/etharp.c line 1379
0x4022ba4b: pp_tx_idle_timeout at ?? line ?
0x4022b9be: pp_tx_idle_timeout at ?? line ?
0x40239767: ets_snprintf at ?? line ?
```

I tried some different approaches, such as using a static buffer, bigger buffer, checking for NULL instead of using containsKey,... But I don't find why it is crashing there. I'm using the ArduinoJSON library also for parsing a configuration file in SPIFFS at startup, where it works perfectly.
BUT, if I disable the config parsing part of the code, the handler to parse the POST works... What could be wrong? Could I be running out of memory? I suppose the buffer is released when it goes out of scope?

bblanchon, 2017-11-27:
Hi @klaasdc,

It seems that the stack trace refers to `parseObject()` but I don't know how much accurate this is because it also contains `~String()` that has no reason to be there.
Can you check that it's actually `parseObject()` and not `containsKey()` that is causing the error?

>  if I disable the config parsing part of the code, the handler to parse the POST works...

I'm very intrigued by this part, there may be something fishy in the config parsing.
Is it possible that the config parsing saves pointers to a temporary memory location; for instance does it saves a `const char*` to a string stored in the `JsonBuffer`?

Regards,
Benoit

klaasdc, 2017-11-27:
Ah, you were right, it refers to parseObject().

Because of your remark, I started to disable part by part of the config parser, until the crashing stopped. After commenting the following lines, the crashes stop:
```arduino
strcpy(ap_ssid, root["network"]["ssid"]);
strcpy(ap_pass, root["network"]["pass"]);
```

Those strings are defined global as:
```arduino
char ap_ssid[SSID_LEN] = "";
char ap_pass[PASS_LEN] = "";
```

I specifically used strcpy to avoid dangling pointers, but I did it wrong?

bblanchon, 2017-11-27:
This looks correct unless there is a length issue.
First, we could replace with `strlcpy()` to make sure there is no buffer overrun:

```c++
strlcpy(ap_ssid, root["network"]["ssid"], sizeof(ap_ssid));
strlcpy(ap_pass, root["network"]["pass"], sizeof(ap_pass));
```

Also, I'm pretty sure that it's an undefined behavior to call `strcpy()` with a null pointer, so it would be a good idea to check:

```c++
const char* ssid = root["network"]["ssid"]
if(ssid)
  strlcpy(ap_ssid, ssid, sizeof(ap_ssid));

const char* pass = root["network"]["pass"];
if(pass)
  strlcpy(ap_pass, pass, sizeof(ap_pass));
```

klaasdc, 2017-11-27:
Inserted your code, but still crashing. The crashing happens in every webserver handler that parses JSON from a POST. I discovered that commenting the ssid/pass parsing but uncommenting further parts of the config parser also gives crashes. It appears that the mere fact of accessing the root object once in the config parser is enough to crash later inside the webserver handler.

I will try to make a minimal example. The sketch now is quite big and I'm also using SPIFFS for serving the webpages.

bblanchon, 2017-11-30:
Any progress?
Could this be a stack corruption or overflow?

klaasdc, 2017-11-30:
I removed a lot of the code until I had a bare minimum, which then magically worked. So I'm adding again piece by piece. I didn't have so much time to work on it yet, but I'll update here as soon as possible :)

klaasdc, 2017-11-30:
Ok, I don't know why this is happening.

My handlers for the webserver are defined in webhandles.h and implemented in webhandles.cpp
The .h is imported in the 'main' sketch file.

When I move the handler(s) to the main sketch file (and thus remove the .h/.cpp), it works perfectly. Moving them back to the .h/.cpp and calling parseJSON will give a crash in the handler.

bblanchon, 2017-12-01:
This sure sounds like a pointer to a temporary.
Can you share the code?

klaasdc, 2017-12-04:
I toyed some more with different body handler methods in the ESPAsyncWebserver, but to no avail.

The following is an example that gives my problem. Moving the 'onResetConfig' handler to the main ino will make it work.

mainTest.ino
========
```arduino
#include "ESP8266WiFi.h"
#include "ESPAsyncWebServer.h"
#include "FS.h"

#include "webhandles.h"

#define ARDUINOJSON_USE_LONG_LONG 1 
#include <ArduinoJson.h>

#include <IPAddress.h>

AsyncWebServer server(80);

IPAddress captiveIP(192,168,4,1);
IPAddress subnet(255,255,255,0);

int CONFIG_BUTTON_PIN = D5;

bool reconnectWifi = false;//Flag that can be set to force reconnection to WiFi AP
char ap_ssid[32] = "";
char ap_pass[64] = "";
bool manIpConf = false;
IPAddress clientIpAddr = IPAddress(192,168,1,142);
IPAddress gatewayIpAddr = IPAddress(192,168,1,1);
IPAddress subnetAddr = IPAddress(255,255,255,0);

void setup(){
  pinMode(CONFIG_BUTTON_PIN, INPUT);
  Serial.begin(115200);
  if (!SPIFFS.begin()){
    while(true){
      Serial.println("Filesystem error!");
      delay(2000);
    }
  }
  loadSettingsFromFile();

  //Init WiFi
  WiFi.begin();
  WiFi.mode(WIFI_STA);

//  WiFi.config(clientIpAddr, gatewayIpAddr, subnetAddr);
  WiFi.begin("YOURSSID", "YOURWIFIPASS");
  while (WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(1000);
  }
  Serial.print(F("Local IP: "));
  Serial.println(WiFi.localIP());

//  server.on("/resetconfig", HTTP_POST, onResetConfig);
  server.on("/resetconfig", HTTP_POST, 
      [](AsyncWebServerRequest *request){},NULL,
      onResetConfig);

  //Static files (htm/css/js/img)
  //server.serveStatic("/", SPIFFS, "/");
  server.on("/", HTTP_GET, onRoot);

  //Serve 404 for all other requests
  server.onNotFound([](AsyncWebServerRequest *request){request->send(404);});
  
  server.begin();
}

void loadSettingsFromFile(){
  if (SPIFFS.exists("/config.json")){
    File file = SPIFFS.open("/config.json", "r");
    size_t size = file.size();
    Serial.print("Config file found. Size is ");
    Serial.print(size);
    Serial.println("b");
    if (size>0) {
      DynamicJsonBuffer jsonBuffer;
      JsonObject& root = jsonBuffer.parseObject(file);
      if (!root.success()) {
        Serial.println("Could not read config file!");
        file.close();
      } else {
        Serial.println("Config file parsed");
        root.prettyPrintTo(Serial);
        Serial.println("");
        //Parse wifi config
        const char* ssid = root["network"]["ssid"];
        if(ssid)
          strlcpy(ap_ssid, ssid, sizeof(ap_ssid));
        const char* pass = root["network"]["pass"];
        if(pass)
          strlcpy(ap_pass, pass, sizeof(ap_pass));
        Serial.println(ap_ssid);
        Serial.println(ap_pass);
        manIpConf = root["network"]["manIpConf"].as<bool>();
        clientIpAddr.fromString(root["network"]["clientIp"].asString());
        gatewayIpAddr.fromString(root["network"]["gatewayIpAddr"].asString());
        subnet.fromString(root["network"]["subnet"].asString());
        file.close();
        return;
      }
    } else {
      Serial.println("Config file empty!");
    }
  } else {
    Serial.println("Config file not found");
  }
  Serial.println("Loading defaults");
  //Removed rest of code
}

void loop(){
  yield();
}
```

webhandles.h
=========
```arduino
#ifndef webhandles_h
#define webhandles_h

#include "ESPAsyncWebServer.h"

void onRoot(AsyncWebServerRequest *request);

void onResetConfig(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total);

#endif
```

webhandles.cpp
==========
```arduino
#include "webhandles.h"
#include "ArduinoJson.h"

const char index_page[] PROGMEM = R"=====(
<html>
  <head>
    <title>Test</title>
  </head>
  <body>

    <input id="clickMe" type="button" value="do POST" onclick="doPost();" />
  
  <script>
    
    function doPost(){
      var xhr = new XMLHttpRequest();
      var url = "/resetconfig";
      xhr.open("POST", url, true);
      //xhr.setRequestHeader("Content-type", "application/json");
      var json = {"reset":true, "type":1};
      console.log(JSON.stringify(json));
      xhr.send(JSON.stringify(json));
      xhr.onloadend = function (){};
    }
  
  </script>
  </body>
</html>
)=====";

void onRoot(AsyncWebServerRequest *request){
  request->send_P(200, "text/html", index_page);
}

void onResetConfig(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
  Serial.println("1");
  if (len>0){
    Serial.println("2");
    Serial.println((char*)data);
    Serial.println("3");
    DynamicJsonBuffer locJsonBuffer;
    Serial.println("4");
    JsonObject& root = locJsonBuffer.parseObject(data);
    Serial.println("5");
    if (root.containsKey("reset")){
      Serial.println("6");
      Serial.println("Found reset key");
      //Load default configuration here
      Serial.println("Settings reset!");
      request->send(202); return;
    }
  }
  //No post data specified
  request->send(405);
}
```

bblanchon, 2017-12-11:
Hi @klaasdc 

Please apologize for the days; I was finishing the [book](https://leanpub.com/arduinojson/).

I looked at your code carefully, and it seems all right.

There is only one thing that bothers me: is `data` null terminated?
Can you check?

I can execute your program on my ESP8266, but please confirm that you didn't make any progress.

Regards,
Benoit

klaasdc, 2017-12-11:
Hello @bblanchon 

I supposed that when printing it to Serial would show a missing null-terminator. How can I check for sure?
As a workaround I put all the handlers in the main file, a bit impractical but it works.

bblanchon, 2017-12-11:
We could simply add an assertion to check that `data[len-1]` or `data[len]` is 0.
Indeed, you would probably see it in Serial.

Is this the complete code for the project, can I use it on my ESP8266?
Is there a GitHub repo that I could clone?

klaasdc, 2017-12-11:
It's not the complete project. That has exceeded a few thousand lines of code and SPIFFS files. But this is a minimalistic version which showed the same behavior and is still practical to set up.

Thank you for your persistence to help :)

klaasdc, 2017-12-11:
I forgot to add the config.json file that is used inside loadSettingsFromFile:

config.json
=======
```
{
  "system": {
    "name": "Controller1"
  },
  "network": {
    "ssid": "SSID",
    "pass": "PASS",
    "manIpConf": false,
    "clientIp": "0.0.0.0",
    "gatewayIpAddr": "0.0.0.0",
    "subnet": "255.255.255.0"
  },
  "channcodes": [
    {
      "index": 0,
      "name": "Channel 0",
      "activtime": 2000,
      "stop": [
        1
      ],
      "down": [
        2
      ],
      "up": [
        3
      ]
    },
    {
      "index": 1,
      "name": "Channel 1",
      "activtime": 2000,
      "stop": [
        1
      ],
      "down": [
        2
      ],
      "up": [
        3
      ]
    },
    {
      "index": 2,
      "name": "Channel 2",
      "activtime": 2000,
      "stop": [
        1
      ],
      "down": [
        2
      ],
      "up": [
        3
      ]
    },
    {
      "index": 3,
      "name": "Channel 3",
      "activtime": 2000,
      "stop": [
        1
      ],
      "down": [
        2
      ],
      "up": [
        3
      ]
    }
  ]
}
```

bblanchon, 2017-12-12:
Hi @klaasdc,

I took me about 3 hours, but I finally found the explanation!

`mainTest.ino` includes ArduinoJson like that:

```c++
#define ARDUINOJSON_USE_LONG_LONG 1 
#include <ArduinoJson.h>
```

but `webhandles.cpp` includes it like that:

```c++
#include <ArduinoJson.h>
```

The two compilation units have different versions of `JsonVariant`.
So the executable contains some function tailored for a big `JsonVariant` and others for a small one.
Hence, the unpredictable behavior.

Solution: define `ARDUINOJSON_USE_LONG_LONG` in `webhandles.cpp` or remove it from `mainTest.ino`.

Regards,
Benoit

klaasdc, 2017-12-12:
Wow, I would have never thought of that.
I always believed that the compiler/linker treats all imports equal. So a single ARDUINOJSON_USE_LONG_LONG before the first import would be enough...

I suppose that it would be different if the code were in a .ino and not a .cpp, since the .ino is joined together first by the Arduino preprocessing?

Thank you very much!

closed 2017-12-12 by klaasdc