bblanchon/ArduinoJson #621: Trailing commas crash my esp

ferchinas, 2017-11-28:
Hi @bblanchon:
Is there any reason why this library does not support trailing commas?
```
{
   "Array": [1,2,3,]
}
```
when I forget to remove the last comma of an array this happens in my esp8266

```
Exception (28):
epc1 = 0x4021f96e epc2 = 0x00000000 epc3 = 0x00000000 excvaddr = 0x00000000 depc = 0x00000000

ctx: cont
sp: 3fff26a0 end: 3fff3170 offset: 01a0

>>> stack >>>
3fff2840: 3fff2f04 3fff5f24 3fff1c10 40217f3f
3fff2850: 3ffe940c 3fff31b4 3fff31b4 3fff5f24
3fff2860: 3fff1a30 3fff5f24 3fff1c10 402140c1
```

It would be preferable to take it as an empty object or a parsing error.

bblanchon, 2017-11-28:
Hi @ferchinas,

It works fine on wandbox:
https://wandbox.org/permlink/b4KUzZZS6espZtYl

Can you spot the line causing the crash?

ferchinas, 2017-11-28:
Hi @bblanchon,
Thanks for answer, I found that the error was caused by my code because `JsonArray :: size ()` counts 4 objects in the previous example.

closed 2017-11-28 by ferchinas