bblanchon/ArduinoJson #622: refto: https://github.com/bblanchon/ArduinoJson/issues/10

r-anwar, 2017-11-28:
@bblanchon  Hi! I have the same problem like ersinpw.

I used the builder pattern with no success. The last entry overrides all elements in 'users' field.

class JSONBuilder
{
DynamicJsonBuffer _buffer;
JsonObject& _root;

public:
JSONBuilder()
: _root(_buffer.createObject()) {
}

void getUsersResult(User users[], int size) {

    JsonArray& usersJSON = _root.createNestedArray("users");

    for(int i = 0; i < size; i++) {
        User user = users[i];

        usersJSON.add(user.toJson(_buffer.createObject()));
    }
}

void dumpTo(Print& destination) const {
    _root.prettyPrintTo(destination);
}

String dumpTo(String destination) const {
    _root.prettyPrintTo(destination);
    Serial.println(destination);
    return destination;
}

private:
};

Please, may you can help me?

Best regards

bblanchon, 2017-11-29:
Hi @r-anwar,

> The last entry overrides all elements in 'users' field.

This usually happens when strings are added in a loop.
I suspect that the following block makes a temporary copy in the `user` variable.

```c++
for(int i = 0; i < size; i++) {
    User user = users[i];

    usersJSON.add(user.toJson(_buffer.createObject()));
}
```

Please try to replace with:

```c++
for(int i = 0; i < size; i++) {
    User& user = users[i];

    usersJSON.add(user.toJson(_buffer.createObject()));
}
```

To prevent the copy.

See also:

* [FAQ: Using a loop, why are values all the same?](http://arduinojson.org/faq/using-a-loop-why-are-values-all-the-same/)

r-anwar, 2017-11-29:
@bblanchon thank you so much! This fixed the problem!

closed 2017-11-29 by r-anwar