bblanchon/ArduinoJson #623: UTF8 Escape sequence filtered

shurillu, 2017-11-28:
Hi!
I'm using your excellent library for integrating Telegram bots inside a ESP8266 and i'm very happy with it.
It work pretty well but when I send some text that contain extended chatacters (like "€" or "à" for example)  to my telegram bot, Telegram send the text coded in UTF8.
For example, if I send the euro character "€" to my telegram bot, this is what I receive (raw string):

{"ok":true,"result":[{"update_id":0000000,
"message":{"message_id":159,"from":{"id":0000000,"is_bot":false,"first_name":"XXXXX","username":"XXXXX","language_code":"it"},"chat":{"id":0000000,"first_name":"XXXXX","username":"XXXXX","type":"private"},"date":1511886142,"text":"\u20ac"}}]}

(I masked some fields)

The "text" field contain the message sent: in that case the message is "\u20ac" that is the "€" character coded in UTF8

If I parse the stream using your libray, I got:

{"ok":true,"result":[{"update_id":0000000,
"message":{"message_id":159,"from":{"id":0000000,"is_bot":false,"first_name":"XXXXX","username":"XXXXX","language_code":"it"},"chat":{"id":0000000,"first_name":"XXXXX","username":"XXXXX","type":"private"},"date":1511886142,"text":"u20ac"}}]}

the backslash "\" disappear (I guess you filter it inside the library)

There is a way to solve that? I know that you are not supporting the UTF8 code, but maybe there is a walkaround...

Thanks, 

Stefano

P.S.: Telegram use the UTF8 standard for coding all messages (input and output)

bblanchon, 2017-11-28:
Hi Stefano,

From your message, I understand that there are some confusions in your mind about UTF-8.
I'll try to clarify.

`\u20ac` is not UTF-8, it is an escape sequence containing the UTF-16 surrogate pair to represent the euro sign.
JSON was standardized that way before UTF-8 was dominant.

ArduinoJson, as well as many other parsers, is not able to translate UTF-16 surrogate pairs because it's very complicated and (almost) nobody uses them.

If you were using UTF-8, the euro sign would be the sequence of byte `0xE2 0x82 0xAC`, which ArduinoJson would accept without any problem.

I'm not familiar with the Telegram API, but I suspect it's possible to switch to UTF-8.
For example, can you try to include `Accept-Charset: utf-8` in your request headers, or something similar.

shurillu, 2017-11-29:
YES! It's the same issue that I have.
I'm planning to release a class (based on your library  - man, with arduinoJSON is a piece of cake!) to easily use the Telegram Bot API.
I'm a founder of an Arduino User Group in Cagliari (my home city) and the Telegram Library is a part of a project. 
Until the feature will be released, I'll try to find a workaround (or simply, we will not use extended charset)
Thanks Benoit for all your effort!

Regards,

Stefano

bblanchon, 2017-12-03:
You're welcome.
Congratulations on your project, it sounds very promising.

Can I close this issue?
Or do you think there is something we can do (except from translating UTF-16 escape sequences to UTF-8 characters)?

shurillu, 2017-12-05:
Hi Benoit,

 

   yes, you can close the issue. I made the Unicode to UTF8 (yes, it’s a UNICODE like coding) translating procedure and it work like a charm.

 

Thanks a lot, regards

 

Stefano

closed 2017-12-05 by bblanchon