bblanchon/ArduinoJson #624: Parsing JSON data with unknown structure/size

azmike, 2017-11-28:
Hi,

I am using an ESP8266 to pull JSON data from a webpage and I am running into a few issues. I am able to successfully pull the data and store it locally.

The JSON file varies in size and data structure/contents returned, and I am unsure how to parse the JSON data or allocate size without knowing what data to expect.

How do you suggest I utilize AruduinoJson to best handle my data?

Also, the data I pulled does not have quotes delimited with slashes; will this affect how the parseObject function oeprates?

bblanchon, 2017-11-29:
Hi @azmike,

> The JSON file varies in size and data structure/contents returned, and I am unsure how to parse the JSON data or allocate size without knowing what data to expect.

What are you going to do with a JSON document that you know nothing about?
If you just need to list the values, use [JsonObject::begin() / JsonObject::end()](http://arduinojson.org/api/jsonobject/begin_end/).
If you don't know the size of the `JsonBuffer`, use a `DynamicJsonBuffer` as it's able to grow.

> Also, the data I pulled does not have quotes delimited with slashes; will this affect how the parseObject function oeprates?

This should not be a problem.

Regards,
Benoit

azmike, 2017-11-29:
@bblanchon , thanks that worked great!

Just as an FYI, I'm pulling varying sets of data from a server and wouldn't be able to hardcode key/value pairs in my device code.

closed 2017-11-29 by azmike