bblanchon/ArduinoJson #626: Soft WDT Reset?

beicnet, 2017-11-29:
Hi @bblanchon,

For some reason, after 6th request from Weather Underground API (every 10 mins) I got "Soft WDT Reset", I tought it's because I'm using the older version of your library v5.6.7, now I updated it from the Git to v5.11.2, but same issue continues.

```
Soft WDT reset

ctx: cont 
sp: 3ffef470 end: 3ffef770 offset: 01b0

>>>stack>>>
3ffef620:  3ffe8a08 3ffef680 3ffef6f0 40204b8c  
3ffef630:  3ffe8a08 000006b1 3fff1acd 40205b48  
3ffef640:  3ffee6fc 00000143 00000143 3ffee71c  
3ffef650:  3ffee604 00000000 3ffee71c 40205b79  
3ffef660:  3fffdad0 00000000 3ffee604 40204671  
3ffef670:  3fffdad0 00000000 3ffee610 402028c1  
3ffef680:  3fff05b4 3ffefff4 3fff0914 0000001f  
3ffef690:  00000014 3f000050 3f001388 3fff093c  
3ffef6a0:  0000003f 00000037 3fff0024 0000000f  
3ffef6b0:  00000004 3fff003c 0000000f 00000000  
3ffef6c0:  3fff05ec 0000001f 00000011 3fff0614  
3ffef6d0:  0000000f 00000000 00000000 00000000  
3ffef6e0:  000000c8 00000a1f 3ffe8500 00000000  
3ffef6f0:  00000000 00000000 00000000 3ffee70a  
3ffef700:  3ffe8550 0000000a 00000000 00000100  
3ffef710:  00000000 00000000 00000000 00000000  
3ffef720:  00000000 00000000 3ffee71c 40204654  
3ffef730:  3fff007c 40202d30 3ffee71c 3ffee748  
3ffef740:  3fffdad0 00000000 3ffee610 40202a80  
3ffef750:  3fffdad0 00000000 3ffee740 402056e4  
3ffef760:  feefeffe feefeffe 3ffee750 40100718  
<<<stack<<<
```
Any thoughts about it?

Regards,
Viktor

bblanchon, 2017-11-30:
Hi Viktor,

Are you reusing the `JsonBuffer`?
If so it's probably filling up until it cannot parse the input anymore.

See:

* [Why does my device crash or reboot?](https://arduinojson.org/faq/why-does-my-device-crash-or-reboot/) (updated a minute ago)
* [How to reuse a `JsonBuffer`?](http://arduinojson.org/faq/how-to-reuse-a-jsonbuffer/)

Regards,
Benoit

beicnet, 2017-11-30:
Hi @bblanchon,

I just using `DynamicJsonBuffer jsonBuffer;` and calling my function every 10 minutes (made with Millis) from loop.

Also can't use `jsonBuffer.clear();`, because it gives me compiler error (your latest Git version).

The second what I testing now, I commented out `Serial.println(JsonData);` in my sketch, so it will not output the parsed data to the Serial monitor and it seems that the "Soft WDT reset" gone for now.

Is it possible that the Json Serial Output did those WDT resets?!

Also I just found your **ArduinoJson Assistant** and it's just great!!!

Kind regards,
Viktor

bblanchon, 2017-12-03:
Yes, `Serial.println(JsonData);` can make the program crash if `JsonData` is dangling or not null-terminated.

beicnet, 2017-12-03:
Oh, okay! Thank you! ;)

closed 2017-12-11 by bblanchon