bblanchon/ArduinoJson #628: JsonArray::get<float>() returns NaN for null field

mill1000, 2017-11-30:
Given a JSON containing an array of nulls, calling get<float>() returns NaN instead of 0.0 as documented [here](http://arduinojson.org/api/jsonarray/get/) 
> In the case of an error (index out of range or incompatible type), the default value for the type T is returned:
    0 for an integer
    0.0 for a double
    NULL for a char*
    etc.

Input JSON: { "offset" : [null,null,null] }

```
JsonArray& offset = root.get<JsonVariant>("offset");
float value = offset.get<float>(0); // Will return NaN not 0.0
```

A work around I've found is to us the | operator to specify a default value of 0.0

bblanchon, 2017-12-01:
Indeed, to reduce the size of the code, the parser only checks the first letter:

* `N` or `n` is NaN
* `i` or `I` is Infinite

I realize that this is a defect because the code doesn't behave as advertised; however, I'm not sure it's worth fixing it, as the code would become bigger.

What do you think?

beicnet, 2017-12-01:
How much bigger in size?!

bblanchon, 2017-12-01:
`NaN`, `Inf` and `Infinite` are already 17 bytes, next i need to add the case-insensitive comparison.

Apparently, you have no idea how crazy I am about code size 😉 
Every byte in ArduinoJson carries its own value, if we make it bigger it will not fit in small MCUs.

beicnet, 2017-12-01:
> it will not fit in small MCUs

Agree with that, but it's not like we talking about 1kB or 10kB range!

I think a few bytes will not hurt anyone! ;)

Kind regards,
Viktor

p.s. Please if you can, give me some reasonable answer or explanation for #626! Thank you!

mill1000, 2017-12-01:
I would be okay with leaving the existing behavior if the documentation were updated to reflect the actual operation.

bblanchon, 2018-07-12:
Fix is available in [v6.2.0-beta](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.2.0-beta)

closed 2018-07-12 by bblanchon